tagyoureit/nodejs-poolController #1: Control from Google Home?

arrmo, 2017-01-19:
OK, have to ask - as the wife just bought one of these ... :).

Any chance we can interface to / control this from Google Home?

Feel free to close if this is nuts!

tagyoureit, 2017-01-19:
If she wants to buy me one, too, I'll be happy to get it going!  lol.  The answer is yes, I would love to have this integration, but it's a bit beyond my reach now.  

We'll leave it open and I flagged it 'help wanted'.

triorbonk, 2019-09-21:
I do much of my GH integration through my ISY.  

But to relay information about the pool IE: Temp, clor, etc.  You would need to following this tutorial. 
https://codelabs.developers.google.com/codelabs/actions-1/index.html?index=..%2F..index#0

tagyoureit, 2019-09-21:
I still don't have a Google Home (nor an ISY).  Would be great to have some community help on this one.  :)

triorbonk, 2019-09-27:
I have started working on this.  Will post info as I figure it out.

Eric

From: tagyoureit 
Sent: Friday, September 20, 2019 10:42 PM
To: tagyoureit/nodejs-poolController 
Cc: Eric Repec ; Comment 
Subject: Re: [tagyoureit/nodejs-poolController] Control from Google Home? (#1)

I still don't have a Google Home (nor an ISY). Would be great to have some community help on this one. :)

closed 2020-09-17 by tagyoureit