tagyoureit/nodejs-poolController #2: npm install/update

arrmo, 2017-01-19:
OK, new repository (new fork) -> warnings / errors in npm,
npm WARN prefer global coffee-script@1.10.0 should be installed with -g

And error in net,
├── auto-updater@1.0.2
├── bootstrap@3.3.7
├── bottlejs@1.5.0
├── chai@3.5.0
├── chai-as-promised@6.0.0
├── dateformat@1.0.12
├── dequeue@1.0.5
├── express@4.14.0
├── grunt@1.0.1
├── grunt-banner@0.6.0
├── http-auth@3.1.1
├── istanbul@0.4.5
├── jquery-ui-dist@1.12.1
├── lodash@4.17.4
├── mocha@3.2.0
├── nanotimer@0.3.14
├── net@1.0.2  invalid
├── request@2.79.0
├── request-promise@4.1.1
├── serialport@4.0.7
├── sinon@1.17.7
├── sinon-chai@2.8.0
├── socket.io@1.7.2
├── socket.io-client@1.7.2
├── underscore@1.8.3
└── winston@2.3.0

tagyoureit, 2017-01-19:
A bunch of those don't need to be installed (mocha, sinon*, chai*, grunt) as they are Dev dependencies. 

I should probably update the install instructions.

`npm install --only=prod` will only install the Production dependencies.  `npm install` will install everything.  I have a lot of stuff installed to get my test suites running (and some of it probably isn't needed).  

I'm not even sure which module is installing coffee-script.  And what is the net error?

arrmo, 2017-01-19:
FYI, erased node_modules, installed only prod => errors out! So some may still be missing. I did a full reinstall of modules, then it runs again.

tagyoureit, 2017-02-21:
Can you pull the latest (3.1.10) and see if you still get errors here?  And, I was wrong above, you need to use `npm install --production` to install on production dependencies.  Quick test of production only, and then test/production, both seem to be working now.

tagyoureit, 2017-02-21:
Yup, just blow out the node_modules directory.  Perfectly safe to do it!

arrmo, 2017-02-22:
OK, that list isn't quite complete ... ;-). Blew it away, installed prod only, and now,
Error: Cannot find module 'promised-io/fs'

tagyoureit, 2017-02-22:
I have the `promised-io` library in the package.json under production dependencies.  Not sure why it isn't being picked up on your end...?  https://github.com/tagyoureit/nodejs-poolController/blob/3.1.x-DEV/package.json#L42

arrmo, 2017-02-22:
My fault, sorry! Needed to pull git again - now OK, thanks!

closed 2017-02-22 by arrmo