tagyoureit/nodejs-poolController #5: Default hideAux to false for new installs?

jeffleus, 2017-01-19:
My EasyTouch controller was setup with spa light, pool light, and blower on AUX1, AUX2, and AUX3.  During initial sfwr setup I was only presented w/ options for "Feature" circuits.  Upon changing the variable to false, the AUX circuits appeared and I could toggle them and control the items.  I then changed the names of the items in the controller and reset the bit back to true to hide the un-used AUX circuits.

https://github.com/tagyoureit/nodejs-poolController/blob/master/bootstrap/main.js#L178

Should this variable default to false during setup?  Or, should there be a note in the readme to help out others in this situation?  Thoughts?

arrmo, 2017-01-22:
Updated - added the variable to control this to clientConfig.json. In PR #8.

Try it out, let me know. Thanks!

arrmo, 2017-01-22:
This should be corrected now - please test when you can. Thanks.

arrmo, 2017-02-07:
Closing this, it's been in place for 2 weeks now. Re-open if it's not working.

closed 2017-02-07 by arrmo