tagyoureit/nodejs-poolController #10: Lighting?

tagyoureit, 2017-01-22:
moved from: https://github.com/tagyoureit/nodejs-Pentair/issues/55

@rayzorben
   
rayzorben commented 25 days ago
Got my RS485 USB port, hooked it up to my RPI and got it working today, excellent job!

Is there any support for the lighting? What I am looking for is the ability to set the spa and pool lights to different colors.

If I use the 'Color Set' functionality I can do this through the Intellilogic software. There is also 'Intellibright' that allows me to set 1 of 5 colors, but it is set for both the pool and the spa to the same color.

The problem that I found out this holiday season is that although 'Intellibright' supports a Green and Red, 'Color Set' does not have a Red color. The bulbs can actually output a red, but I can't set the lights to Red and Green, so I am hoping to direct this manually.

Any ideas?

[see rest of thread in original issue]

rayzorben, 2017-02-27:
I want to help out here anyway I can, do you need anything from me that can help with adding lighting to the feature list?

tagyoureit, 2017-02-27:
Awesome!  I've started putting in the wiring to decode the circuits.  I'll push that up with the next Dev release (it will have a lot of changes, likely be a 4.0-beta).  First step will be to make sure we are decoding the packets, then controlling them through API calls, and finally add them to the UI.  I'm ~75% of the way done with the decoding part.  Once that is finished, I'll push it up and we can test if we see all the packets we think we should see.  So give me another week or so (I'm traveling this week) and hopefully next week we can begin testing.  ☺️

rayzorben, 2017-02-27:
wow, awesome. I'll be on the watch for it and test it out once you push it.

tagyoureit, 2017-03-13:
@rayzorben 1st pass at the Intellibrite lights is posted on the Dev branch https://github.com/tagyoureit/nodejs-poolController/tree/4.x-DEV.

Here's what I've done so far.
* Decoded the assignment of lights to groups/positions (I think I've seen both names... should it be one or the other?  I've been going with groups).  This is packet 39 (broadcast) and also 167 (set).  They are both identical as far as I can tell.

* Decoded the changing of light groups.  This is Packet 96.

* Started to change the circuit object to have this light info.  You can browse to `localhost:3000/circuit` and the status should change.

* If you set `logConfigMessages: 1` and `logLevel: verbose` (in config.json) you should see the messages in the console logs.  You should see messages for adding/changing/deleting positions/groups, and also changing colors.

Example:
```
21:57:32.015 VERBOSE Msg# 58  Light group added or changed for circuit SPA LIGHT (7): {
  "group": 1,
  "colorStr": "off",
  "color": -1
}
```

FYI, it wasn't enough to just assign a light to `Intellibrite`.  In order for it to show up in a group (despite the fact you can set it to a group), it needs to be added to the "Special Lights..." group.  I'm not sure if this is by design or a bug with the ScreenLogic software, but why be able to assign the group/position if it won't do anything without this additional step???)
[image]

* When I start to actually select the colors for Intellibrite, I see some interesting messages...
```
(after a Magenta color)
21:58:21.194 VERBOSE Msg# 168   Controller packet is known to be a Get Pump Status packet, but is NOT DECODED: 165,16,16,34,215,1,1,1,192
21:58:21.323 VERBOSE Msg# 169   Controller packet is known to be a Pump Status packet, but is NOT DECODED: 165,16,15,16,23,16,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,128,1,124

(and after a Romance color)
21:59:11.349 VERBOSE Msg# 327   Controller packet is known to be a Get Pump Status packet, but is NOT DECODED: 165,16,16,34,215,1,2,1,193
21:59:11.454 VERBOSE Msg# 328   Controller packet is known to be a Pump Status packet, but is NOT DECODED: 165,16,15,16,23,16,2,4,0,0,0,0,0,0,0,0,0,0,0,0,0,128,1,129
```

I/we previously thought they might be pump messages but the #'s corresponding to the group/position assignments are 0/16/32/48 (for 1, 2, 3, 4).  So this 16,1,0,0,0... might be part of the light assignment process?

* I default my group/positions to -1 (which means they aren't set) and add them to every circuit.  The settings are group/position assignments 1/2/3/4, but I've also seen commands for group/position 0.  Need to understand this better, but I think 0=all (e.g. all on/all off).  (Circuits that aren't lights will always have group=-1.)

* I'm storing the information for the lights as 
```
 {
  "group": 1,
  "colorStr": "off",
  "color": -1
}
```

group is the group/position.  Color is the # associated with the packet and colorStr is the name of the color.

* I have not tried any other light settings yet (SAMl, Color Wheel, etc etc)

Check it out and let me know what you think/find.  Thanks for the patience in letting me get the basic guts of this working.

bsileo, 2018-04-20:
Started looking at this today. I get this error when I change to a set color.
```Unhandled rejection TypeError: Cannot read property '194' of undefined
    at Object.setControllerLightColor (/home/pi/poolController/nodejs-poolController/src/lib/equipment/circuit.js:527:145)
    at Object.process (/home/pi/poolController/nodejs-poolController/src/lib/comms/inbound/controller/96.js:31:27)
    at Object.processControllerPacket (/home/pi/poolController/nodejs-poolController/src/lib/comms/inbound/process-controller.js:125:45)
    at decode (/home/pi/poolController/nodejs-poolController/src/lib/comms/inbound/decode-helper.js:263:51)
    at Object.processChecksum (/home/pi/poolController/nodejs-poolController/src/lib/comms/inbound/decode-helper.js:301:13)
    at Object.iterateOverArrayOfArrays (/home/pi/poolController/nodejs-poolController/src/lib/comms/inbound/receive-buffer.js:117:44)
    at Object.push (/home/pi/poolController/nodejs-poolController/src/lib/comms/inbound/packet-buffer.js:35:35)
    at SerialPort.<anonymous> (/home/pi/poolController/nodejs-poolController/src/lib/comms/sp-helper.js:74:44)
    at emitNone (events.js:106:13)
    at SerialPort.emit (events.js:208:7)
    at emitReadable_ (_stream_readable.js:513:10)
    at emitReadable (_stream_readable.js:507:7)
    at addChunk (_stream_readable.js:274:7)
    at readableAddChunk (_stream_readable.js:250:11)
    at SerialPort.Readable.push (_stream_readable.js:208:10)
    at binding.read.then (/home/pi/poolController/nodejs-poolController/node_modules/serialport/lib/serialport.js:374:10)
```

tagyoureit, 2018-04-21:
What command are you using?  And my new pool should have these lights so I’ll be able to really dial-in this code in a few weeks anyway.

bsileo, 2018-04-23:
I set it to Party Mode using the SreenLogic App. But I suspect you are
looking for s more technical answer. I may need to learn better how to get
you debugging details / specific command strings from the 485 bus to help
evolve this.

>

tagyoureit, 2018-04-23:
No, actually that's good enough.  I can do the same on my end and debug it.

bsileo, 2018-04-24:
Found the fix to this problem - just missing keys in the config container:

```
    return {
        packetFields: packetFields,
        controllerStatusPacketFields: controllerStatusPacketFields,
        chlorinatorPacketFields: chlorinatorPacketFields,
        pumpPacketFields: pumpPacketFields,
        pumpType: pumpType,
        pumpTypeStr: pumpTypeStr,
        pumpConfigFieldsCommon: pumpConfigFieldsCommon,
        pumpConfigFieldsVS: pumpConfigFieldsVS,
        pumpConfigFieldsVF: pumpConfigFieldsVF,
        pumpConfigFieldsVSF: pumpConfigFieldsVSF,
        namePacketFields: namePacketFields,
        pumpAction: pumpAction,
        strCircuitName: strCircuitName,
        strCircuitFunction: strCircuitFunction,
        strPumpActions: strPumpActions,
        strChlorinatorActions: strChlorinatorActions,
        strControllerActions: strControllerActions,
        strRunMode: strRunMode,
        strValves: strValves,
        heatModeStr: heatModeStr,
        heatMode: heatMode,
        ctrl: ctrl,
        ctrlString: ctrlString,
        schedulePacketBytes: schedulePacketBytes,
        intellichemPacketFields: intellichemPacketFields,
        strIntellibriteModes: strIntellibriteModes,
        intellibriteModes: intellibriteModes

    }
```

tagyoureit, 2018-04-25:
Ok, added the last two lines and pushed it up.  Thanks for checking digging into this one.

bsileo, 2018-04-25:
Any thoughts on the ability to send the light modes at this point? Presently looks like I will want to look at defining an API to send these packets if I want to put it in place. I may take a look at that but let me know if you are already planning to tackle when your new lights get installed.

tagyoureit, 2018-06-02:
I added support for Intellibrite lights with 5.1.

@rayzorben I haven't played around with modifying the code to see if we can force other colors besides "color set" based colors.  Best suggestion I have right now is to set a 25 second delay between your pool and spa.

[image]

It looks like the red/green line up twice (depending on which color you want where).  Then turn on the lights with "color swim" and it should respect the delay.  Once it gets to the right colors, issue the `save` command.  Intellitouch will now store the states of the pool.  You can go to any other modes and when you want to come back to the holiday colors you can use the `recall` feature.  

I still need to add the "assign to all on/off group" so will leave this thread open for a bit until that's completed.  

Let me know if either of you get a chance to check this out.

closed 2018-06-23 by tagyoureit