tagyoureit/nodejs-poolController #11: Add pump control to Bootstrap

tagyoureit, 2017-01-22:
moved from https://github.com/tagyoureit/nodejs-Pentair/issues/49

@arrmo Can you add the pump control to Bootstrap? You can see the initial iteration at https://github.com/tagyoureit/nodejs-Pentair/blob/master/public/pump.html.

@tagyoureit Ya... possibly. But not everyone needs direct control over the pump. Is it possible to do a "nested" container? And that can be hidden in the config file for those that have controllers? Or possibly just a separate one... I'm ok with either until we get some feedback from others that need the pump control. Personally, I'll keep it hidden.

arrmo, 2017-01-24:
Hi,

Can't get to the link above any more - do you still have this code? And is there an API for this?

Thanks!

tagyoureit, 2017-01-24:
It's in the `/public` directory of the current version.  I am working on this a bit tonight, so maybe start to look at it, but I likely need to make some changes to the Socket and `pump.html` file.

arrmo, 2017-01-27:
Let me know when there is some code to look at, will go from there. Thanks!

arrmo, 2017-03-27:
Hi,

OK, I looked at what is in /public - is all of this what you're thinking needs to be in a modal in Bootstrap? I thought you were looking for some sort of "fuzzy" logic - or has that changed? Want to start working on it, but I admit ... a bit confused by what is desired right now ... :-(.

Thanks!

tagyoureit, 2017-03-27:
Ya, we've had lots of different conversations on this.  The current `pump.html` is just for testing the API's .  It's not a useful interface.  As much as I'd really like to see the natural language interface implemented, it isn't a match for the rest of the UI.  If you have something in mind that would match what we have now, then go with that.  If you don't, then let's revisit the natural language path and make that work.  What do you think you'd be most comfortable doing?

tagyoureit, 2017-12-12:
Included with the new 4.x-DEV release.

closed 2017-12-12 by tagyoureit