tagyoureit/nodejs-poolController #13: Better understanding of heater logic

tagyoureit, 2017-01-22:
moved from: https://github.com/tagyoureit/nodejs-Pentair/issues/32
There are two places in the status packet (from Intellitouch) that relate to heater status. I don't think they are being decoded correctly, at this time.

rflemming, 2017-07-07:
Hey look I'm back :)  A couple things since I can't edit the wiki on the new project:

The broadcast message has the wrong byte marked as being pool/spa heat mode.  You have it as byte 21, when really it's byte 27.  This is of course correct in the code, just not on the wiki.  Byte 21 does appear to have something to do with the heater, but the best I can come up with so far is that it changes from 32 to 0 (on my system of course) when the spa mode is set to solar pref or solar.  No other mode changes seem to make a difference.

Regarding the heater logic I mentioned this a long time ago, but byte 15 changes for me when the solar turns on.  Unfortunately this could also just a valve indicator, but here's what I've seen:

3 - Solar was not running
51 - Solar was running
15 - When pool was set to heater and was running (I don't have a heater)

tagyoureit, 2017-07-07:
Welcome back!  I updated the wiki.  And I do think those numbers are the valves.  I've also seen the value of 48.  These may be dependent on the configuration of each pool and what the valves mean for that system... more investigation needed here.

rerouted, 2019-05-07:
Trying to get this heater logic issue closed.  There is a red LED on my IntelliTouch for when the heater is on or off so I do think there is a way to know in the status messages.
I played around with this tonight on my system as I really want to track how long my heater runs for to find the sweet spot to set it at for the average temperature.

In an **Equipment Status** message
*When my heater was on byte 15 was 12.*
Msg#: 340280 Data: [165,1,15,16,2,29,23,27,44,0,0,0,0,0,0,0,`12`,0,0,4,65,65,0,0,49,0,0,0,1,0,0,110,32,3,13,2,164]

*When I turned the set-point below the pool temp the byte 15 changed back to 0.*
Msg#: 340282 Data: [165,1,15,16,2,29,23,27,44,0,0,0,0,0,0,0,`0`,0,0,4,65,65,0,0,49,0,0,0,1,0,0,123,30,3,13,2,163]

I was wondering if anyone else could check there system and see if they see similar results?

tagyoureit, 2019-06-01:
I'm seeing interesting results, which is why I wasn't able to nail this down previously.
On my system, with solar and a heater...
dec (binary)
3 (0000011) = anytime solar/heater is on but setpoint < current temp
15 (0011111) = heater is on, setpoint > current temp
51 (0110011) = solar on, setpoint > current temp

In your scenario, 
12 (0001100) = heater on
0 (0000000) = heater off

It makes sense that bits, given positions (1234567) 
(____56_) 56 == `11` is heater on, and 
(_23______) 23 == `11` = solar on.

I'm not sure why I would have (_____67) 67=`11` bits set and you don't.  I will implement the masks for the 23/56 bits to be solar/heater on and see if we can go from there, but this is still a bit of a mystery to me.

tagyoureit, 2019-06-01:
Ok, I think we've got this one... thanks @rerouted.

Solar on, setpoint > temp.
[image]

Solar on, setpoint < temp.
[image]

Pics look the same for heater.  The only thing still left to check is if the solar setpoint is > current temp but it is cloudy out (eg solar is off even though it would otherwise be on).

tagyoureit, 2019-06-02:
Closing this out... seems to work as expected with no sun and the solar set points above/below the current temp.

closed 2019-06-02 by tagyoureit