tagyoureit/nodejs-poolController #15: Add ability to create/edit schedules

tagyoureit, 2017-01-22:
moved from: https://github.com/tagyoureit/nodejs-Pentair/issues/18

@arrmo Add this to the npm packages? Will be needed for time selections (later),
https://www.npmjs.com/package/timepicker

jQuery UI has datepicker, but not timepicker (odd).

tagyoureit, 2017-02-17:
I added this in 3.1.9 (dev branch).  @arrmo can you test it out when you make your way through the ever-growing list?  👍 

Socket:
| To app | <code>setSchedule(id, circuit, starthh, startmm, endhh, endmm, dow*) | set the schedule on the controller for the particular schedule ID.  dow= day of week as expressed as [0=Sunday, 1=Monday, 2=Tuesday, 4=Wednesday, 8=Thursday, 16=Friday, 32=Saturday] or a combination thereof [3=Monday+Tuesday].  To set a schedule set a valid start and end time (hh:mm).  To set an egg timer, set the start time to 25:00 and the endtime to the duration (hh:mm) you want the egg timer to run.

Or REST API (example):
http://localhost:3000/schedule/set/12/5/13/20/13/40/3

Syntax is very similar to setTime, but for days of the week you can add them together for multiple values.  

Maybe the easiest place to start is by clicking on a single day in the schedule to add/remove that day from the schedule.  I don't think we can (yet) add/remove an entire schedule so that will be next.

Also, I noticed that when I send back "None" for the days bootstrap says it is undefined.  I thought we fixed that a while back so maybe I changed something on my end?

arrmo, 2017-02-19:
Not forgotten - and I think this one is next, agreed? But what is bothering me - the time picker has some bugs, like if you select time (and it's PM), press escape -> it seems to incorrectly return a value, the same time but AM ... :-(.

What to do ... do we fork the time picker and correct it, or find another? Thoughts? If we fork it ... how to get it to npm, or rather then just add it in? It's no longer maintained.

tagyoureit, 2017-02-21:
This comment was from before the new time picker?  Right?  I'm losing track of time... lol.  New one seems to be working great.

arrmo, 2017-02-21:
Yep, correct! Now that the time picker is cleaned up, will get to the schedules (right now, trying to hide panels).

tagyoureit, 2017-10-25:
Hey @arrmo, I got all of this added... ability to turn on/off individual days and delete & add a schedule/egg timer.  Only thing missing is how to change the heat set point on the schedule.  This one is escaping me.  Check out the 4.x-Dev branch.

arrmo, 2017-10-29:
Will do! Sorry for the delay, was out of the country - but back now, will take a look ... :-).

Thanks!

tagyoureit, 2017-10-30:
My code is pretty messy!  I didn't know that [insert cell](https://www.w3schools.com/jsref/met_tablerow_insertcell.asp) and [insert row](https://www.w3schools.com/jsref/met_table_insertrow.asp) existed until just now.  There are lots of opportunities to refactor my code to make it more readable.

arrmo, 2017-11-03:
OK, trying to get this going, but ... updated to the latest code (4.x-DEV), and I get this in my bootstrap log,
21:58:27.724 INFO Network connected to: atticServer:9801
21:58:27.738 INFO Queueing messages to retrieve configuration from Intellitouch
21:58:32.279 INFO Queueing messages to retrieve configuration from Intellitouch
21:58:34.245 INFO Queueing messages to retrieve configuration from Intellitouch

My equipment never seems to show up, so then I refresh, and get,
The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later.

Thoughts?

tagyoureit, 2017-11-03:
`The server is temporarily unable to service your request due to maintenance downtime or capacity problems. Please try again later.` isn't a message that comes from the poolController app.

There was a problem with the `package.json`, but I pushed up a new one last night.  Maybe you grabbed the code with a bad file?  

Try upping the logging (verbose) and see if it tells you more.  I have not been using Socat recently as it wasn't reliable for a reason I could never track down.

arrmo, 2017-11-03:
I can see the "shutdown" issue ... it's from the sequence below. It seems that the latest version is hammering the serial connection ... up and down? Then socat stops responding (hence the message above). Thoughts?

```
Nov  3 14:23:30 atticServer poolTTY: 2017/11/03 14:23:30 socat[4763] E write(7, 0x558a72b5a770, 4): Broken pipe
Nov  3 14:23:30 atticServer poolTTY: poolTTY: client (pid 4763) exited with 1 status
Nov  3 14:23:31 atticServer poolTTY: 2017/11/03 14:23:31 socat[4764] E write(7, 0x5615df37c770, 4): Broken pipe
Nov  3 14:23:31 atticServer poolTTY: poolTTY: client (pid 4764) exited with 1 status
Nov  3 14:23:33 atticServer poolTTY: 2017/11/03 14:23:33 socat[4881] E write(7, 0x55870bcf6770, 4): Broken pipe
Nov  3 14:23:33 atticServer poolTTY: poolTTY: client (pid 4881) exited with 1 status
Nov  3 14:23:35 atticServer poolTTY: 2017/11/03 14:23:35 socat[4882] E write(7, 0x5612da25f770, 4): Broken pipe
Nov  3 14:23:35 atticServer poolTTY: poolTTY: client (pid 4882) exited with 1 status
Nov  3 14:23:37 atticServer poolTTY: 2017/11/03 14:23:37 socat[4999] E write(7, 0x556d11160770, 4): Broken pipe
Nov  3 14:23:37 atticServer poolTTY: poolTTY: client (pid 4999) exited with 1 status
Nov  3 14:23:37 atticServer poolTTY: poolTTY: terminating too quickly, waiting 300 seconds

```

arrmo, 2017-11-03:
FYI, reset back to my "old" code - all works fine, so it is something in the latest code. What log do you want?

tagyoureit, 2017-11-03:
Hmmm, I downloaded from Github, ran `npm install`, changed the `config.json` settings (netConnect) and it started right up on my remote machine.  Have you tried a reboot?  Or at least wipe the old `node_modules` and reinstall?

arrmo, 2017-11-04:
OK, got it working - had to do a git (hard) reset, then update my local config settings. FYI, one message here from npm ... thoughts?
`npm WARN chai-as-promised@6.0.0 requires a peer of chai@>= 2.1.2 < 4 but none was installed.
`

Up and running, but the schedule tab only shows 1 item now (not all of them). Do you see this?

tagyoureit, 2017-11-04:
The `chai-as-promised` is only used for testing code.  It isn't used for the main app so no worries there.

Regarding the schedules, when you browse to `http://atticServer:3000/schedule` do you get only one schedule or all of them?  This will tell us if I messed up the HTML/JS code or it's something else.

arrmo, 2017-11-04:
I get them all, thanks! The format must have changed, let me dig into the (js) code.

arrmo, 2017-11-05:
OK, I found part of the problem - some script and stylesheets are being blocked (flagged as unsafe). Need to figure this one out first.

arrmo, 2017-11-05:
OK, that was the issue - the links. Copied the two files locally, and changed the links ... life is good ... :-).

Let me dig into the issue noted above now!

arrmo, 2017-11-05:
Here is what I think is going on - but I may be wrong! For the clockPicker (for selecting the times), the code below is called with the correct ID (matching to what you want to click on),
`bindClockPicker('#schTime' + currSchedule.ID + 'EndTime', 'right')
`

But, for the row below, the id of the editable items is not set / not matching, so clicking on them doesn't trigger anything. Again, I think ... ;-).
`bindSelectPickerScheduleCircuit('#schTime' + currSchedule.ID + 'Circuit', currSchedule.friendlyName.capitalizeFirstLetter())
`

Thoughts?

tagyoureit, 2017-11-05:
Hmmm...  it's working on mine.  I'm in Chrome.  See if it's a browser thing.  

The clockPicker thing should work because it's referencing the exact element by name.  Each ID and start/end time make a unique combo for the element name.  

Is it just not showing up, or are you getting an error in the console?

tagyoureit, 2017-11-05:
I don't know if this will help you at all, but I completely re-wrote the new dynamic tables (add/edit/delete egg timer and schedules).  It isn't less complicated, but I moved from writing the HTML code directly (`str="<div id='" + blah blah`) to (`el.append(($('<div/>',{id:name,etc etc}))`).  It should be a little easier to read/debug and I'm no longer getting errors because I messed up the single of double quotes 1000x like I did before.

arrmo, 2017-11-19:
So is the editing fully working for you? I was trying to address the item you noted above.

FYI, still having issues getting it running at all here (on 4.x-DEV). Manually trying to run it I get the following error (and crash / exit),
```
fs.js:642
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: ENOENT: no such file or directory, open '/mnt/ProgSSD/nodejs-poolController/src/package.json'
    at Error (native)
    at Object.fs.openSync (fs.js:642:18)
    at Object.fs.readFileSync (fs.js:510:33)
    at Object.<anonymous> (/mnt/ProgSSD/nodejs-poolController/src/etc/settings.js:27:33)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
```

Suggestions?

Thanks!

tagyoureit, 2017-11-19:
Hmmm... how are you starting the app?  You should be starting it as `npm start {config.json}` (where config.json is an optional path/file to load.  This should be run from the `nodejs-poolController` directory. 

The error is correct in that there is no `package.json` in the `src` directory.

arrmo, 2017-11-19:
OK, my fault, sorry! Seems this has changed - and I missed the fact that it changed.

To your questions / issues above ... still need me to dig into setting the schedule(s)?

tagyoureit, 2017-11-19:
I think it's all good!  Test it out and let me know.  We can close this one out and move onto the next.  :)

arrmo, 2017-11-20:
FYI, this is not forgotten! Trying to get through a few hurdles, the first one is the fact that something has broken my proxy now (worked before the recent updates ... :(). I get this now, trying to resolve it,

`Error during SSL Handshake with remote server`

tagyoureit, 2017-11-20:
did you put back the data directory with the proper security files?

arrmo, 2017-11-21:
OK, you may be on to something! I didn't change anything there, but my certs have updated (not real recently, but this still may be it). I admit, I need to go back and look, I can't recall how the files in that directory got generated (or really, what needs to go there).

Still digging ... :).

arrmo, 2017-11-21:
OK, just found a way to check the cert - and it just expired. Man, when it rains it pours. Need to figure out how I generated them before, do that again ... :).

Thanks!

tagyoureit, 2017-11-21:
https://stackoverflow.com/a/43860138/7386278 -- Good instructions for created a Self Signed Cert.

arrmo, 2017-11-21:
Thanks! I admit, pulling my hair out a bit though ... ;). Tried this one, then npm won't even start => had to rename the files, as follows. Perhaps we need to capture this somewhere?
```
mv cert.key server.key
mv cert.pem server.crt
```

Still doesn't fix my Apache proxy issue though - will debug that, get back to you.

Thanks again.

arrmo, 2017-11-21:
FYI, will work this one in parallel - with some workarounds I'm limping along ... so a couple other tickets created to address items I see ... ;)

arrmo, 2017-11-22:
This one seems to work now! Having other issues, but captured those items in other tickets. So closing this one - but re-open if you disagree.

Thanks!

closed 2017-11-22 by arrmo