tagyoureit/nodejs-poolController #9: Reconnect to Socat

tagyoureit, 2017-01-22:
Moved from https://github.com/tagyoureit/nodejs-Pentair/issues/56
[@arrmo]
Hi,

One item I have found - if the socat machine / server is rebooted (goes away for a bit), the Node.js server has to be restarted (i.e. restart nodejs-Pentair) once socat is available again. Perhaps automatically reconnect somehow?
Or also - allow for a reconnect request?
Thanks!

tagyoureit, 2017-01-30:
Per Gitter, check the latest push.  We can trigger a reload.  For the automatic part, probably makes most sense to do it on the server side.  But we need to balance those people who don't have controllers (and therefor the serial bus will theoretically be quiet) with those people that should be hearing/seeing traffic but aren't.  Any thoughts?

arrmo, 2017-03-23:
Not seeing this automatically reconnect - just tested it. Should I request a reconnect from the client, or handle this on the server side?

tagyoureit, 2017-03-25:
So... tell me more about this.  In the logs below, I started the app with Socat.  Then turned off my wireless, and then turned it back on.  It did reconnect.

How did you test this?  What do you expect to happen?  

Calling the reload should also try to re-initialize the serial port.  Did you try this to see what happens?

```
23:46:37.205 INFO 
  Custom Circuit Names retrieved from configuration: 
	{"0":"WtrFall 1","1":"WtrFall 1.5","2":"WtrFall 2","3":"WtrFall 3","4":"Pool Low2","5":"USERNAME-06","6":"USERNAME-07","7":"USERNAME-08","8":"USERNAME-09","9":"USERNAME-10"}
23:46:37.370 DEBUG iOAOA: Packet being analyzed: 255,255,255,255,255,255,255,255,0,255,165,33,15,16,11,5,1,1,72,0,0,1,63  ******START OF NEW PACKET******
23:46:37.370 DEBUG Msg# 22  Found incoming controller packet: 165,33,15,16,11,5,1,1,72,0,0,1,63
23:46:37.371 DEBUG Msg# 22  Msg received: 165,33,15,16,11,5,1,1,72,0,0,1,63 
                           Msg written: 255,0,255,165,33,16,33,203,1,1,1,196 
                           Match?: 1
23:46:37.540 DEBUG iOAOA: Packet being analyzed: 255,255,255,255,255,255,255,255,0,255,165,33,15,16,11,5,2,0,46,0,0,1,37  ******START OF NEW PACKET******
23:46:37.540 DEBUG Msg# 23  Found incoming controller packet: 165,33,15,16,11,5,2,0,46,0,0,1,37
23:46:37.540 DEBUG Msg# 23  Msg received: 165,33,15,16,11,5,2,0,46,0,0,1,37 
                           Msg written: 255,0,255,165,33,16,33,203,1,2,1,197 
                           Match?: 1
23:46:40.037 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,3,1,198
23:46:42.452 ERROR Aborting controller packet 'Get Circuit Name/Function'.  Tried 10 times to write 255,0,255,165,33,16,33,203,1,3,1,198
23:46:45.051 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,4,1,199
23:46:47.467 ERROR Aborting controller packet 'Get Circuit Name/Function'.  Tried 10 times to write 255,0,255,165,33,16,33,203,1,4,1,199
23:46:50.067 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,5,1,200
23:46:52.492 ERROR Aborting controller packet 'Get Circuit Name/Function'.  Tried 10 times to write 255,0,255,165,33,16,33,203,1,5,1,200
23:46:55.096 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,6,1,201
23:46:57.527 ERROR Aborting controller packet 'Get Circuit Name/Function'.  Tried 10 times to write 255,0,255,165,33,16,33,203,1,6,1,201
23:47:00.135 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,7,1,202
23:47:02.549 ERROR Aborting controller packet 'Get Circuit Name/Function'.  Tried 10 times to write 255,0,255,165,33,16,33,203,1,7,1,202
23:47:05.157 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,8,1,203
23:47:07.574 ERROR Aborting controller packet 'Get Circuit Name/Function'.  Tried 10 times to write 255,0,255,165,33,16,33,203,1,8,1,203
23:47:10.184 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,9,1,204
23:47:12.603 ERROR Aborting controller packet 'Get Circuit Name/Function'.  Tried 10 times to write 255,0,255,165,33,16,33,203,1,9,1,204
23:47:15.203 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,10,1,205
23:47:17.629 ERROR Aborting controller packet 'Get Circuit Name/Function'.  Tried 10 times to write 255,0,255,165,33,16,33,203,1,10,1,205
23:47:20.222 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,11,1,206
23:47:22.654 ERROR Aborting controller packet 'Get Circuit Name/Function'.  Tried 10 times to write 255,0,255,165,33,16,33,203,1,11,1,206
23:47:25.258 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,12,1,207
23:47:27.686 ERROR Aborting controller packet 'Get Circuit Name/Function'.  Tried 10 times to write 255,0,255,165,33,16,33,203,1,12,1,207
23:47:30.283 WARN Error writing controller packet 'Get Circuit Name/Function' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,203,1,13,1,208
23:47:31.895 ERROR Error with port: read ETIMEDOUT.  Will retry in 10 seconds
23:47:32.228 ERROR Error with port: This socket is closed.  Will retry in 10 seconds
23:47:32.229 ERROR Error writing packet (255,0,255,165,33,16,33,203,1,13,1,208): This socket is closed
23:47:41.904 ERROR Error with port: getaddrinfo ENOTFOUND raspberrypi raspberrypi:9801.  Will retry in 10 seconds
23:47:42.231 ERROR Error with port: getaddrinfo ENOTFOUND raspberrypi raspberrypi:9801.  Will retry in 10 seconds
23:47:51.908 ERROR Error with port: getaddrinfo ENOTFOUND raspberrypi raspberrypi:9801.  Will retry in 10 seconds
23:47:52.233 ERROR Error with port: getaddrinfo ENOTFOUND raspberrypi raspberrypi:9801.  Will retry in 10 seconds
23:48:01.912 ERROR Error with port: getaddrinfo ENOTFOUND raspberrypi raspberrypi:9801.  Will retry in 10 seconds
23:48:02.241 ERROR Error with port: getaddrinfo ENOTFOUND raspberrypi raspberrypi:9801.  Will retry in 10 seconds
[... cut ...]
23:48:52.264 INFO Network connected to: raspberrypi:9801
23:48:53.455 DEBUG iOAOA: Packet being analyzed: 32,0,53,51,0,0,4,0,0,122,206,0,13,4,42  ******START OF NEW PACKET******
23:48:55.576 DEBUG iOAOA: Packet being analyzed: 255,0,255,165,33,16,34,217,1,0,1,210  ******START OF NEW PACKET******
23:48:55.576 DEBUG Msg# 24  Found incoming controller packet: 165,33,16,34,217,1,0,1,210
23:48:55.577 DEBUG Msg# 24  Msg received: 165,33,16,34,217,1,0,1,210 
                           Msg written: 255,0,255,165,33,16,33,203,1,13,1,208 
```

arrmo, 2017-03-27:
I tested it by power cycling the machine the USB device is connected to (remotely). I may be confused ... should it auto-reconnect, or do I need to request a reconnect?

Thanks!

tagyoureit, 2017-03-27:
So, you have:
1. "Machine A" - broadcasting the Socat messages with the USB Serial Bus
1. "Machine B" - Running nodejs-PoolController
1. "Machine C" - Browser connected to B (could be same as B)

And you reboot A, and B does not automatically reconnect?  Is that correct?

arrmo, 2017-03-30:
Yep, correct! So I set the socketLogLevel to debug, turned of the messages below that (as logs were filled with traffic), and rebooted A. Then waited (11 min, Last Update timer) ... nothing at all in the log, and no updates.

Tried a page refresh - nope ... nothing in the log, and the data is stale (time is 11 min old still).

Restarted nodejs-poolController - yep! Then data is current, and reconnect happens. But this should happen automatically, no?

tagyoureit, 2017-03-30:
Ok, I think I was able to replicate the problem.  Try this... Reboot A.  Then when it is back up try to turn on (or off) a circuit.  It should reconnect at this time (after 10 seconds).

I think the problem is that when there are no packets to write to the bus, the app doesn't know the difference between a quiet (silent) socat not sending any messages and one that is down/been rebooted.  Once the app tries to write something to the bus, it says "oh crap, I can't connect... let's retry."

If I'm right and you can reconnect in this manner, we need to figure out an appropriate mechanism for "testing" the connection with a write command.

arrmo, 2017-03-31:
Yep, that's part of it! See the log below,
17:33:22.654 ERROR Error with port: This socket has been ended by the other party. Will retry in 10 seconds
17:33:22.655 INFO { 
"text": "User request to toggle SPA LIGHT to on", 
"status": "on", 
"value": 1 }
17:33:22.655 ERROR Error writing packet (255,0,255,165,1,16,33,134,2,4,1,1,100): This socket has been ended by the other party
17:33:32.665 INFO Network connected to: atticServer:9801

But ... even after this, commands fail (i.e. the toggle above doesn't work). Restart nodejs ... and then yes, it works again. So 2 things,
1) As you note, we need to detect when the remote end goes away. Perhaps like I do ... have a watchdog, that restarts whenever there is traffic. When the watchdog expires (configurable, but default to 60 sec?), then initiate a reconnect?
2) The reconnect, as above is not quite working yet either ... thoughts?

Thanks!

tagyoureit, 2017-11-25:
I just implemented a reconnect on the client (aka browser) socket connection.  It tries to reconnect after 20 seconds of inactivity (or disconnect).  It goes by the Last Update time so doesn't really matter how the socket is disconnected.  I don't think this is actually this issue... but related in that we can do something similar on the server side.  Are you still having this type of issue?  (This will get pushed up with next set of code...)

tagyoureit, 2017-11-25:
Ok, fully implemented this for the next push up... added a new, configurable, inactivity timeout.  If this timeout is met, serialport/net connect will attempt to re-open the connection.  This should take care of the issues.

```
        "network": {
            "rs485Port": "/dev/ttyUSB0",
            "netConnect": 1,
            "netHost": "11.11.11.170",
            "netPort": 9801,
            "inactivityRetry": 10
        }
```
(Again... next code push will have this.)

arrmo, 2017-12-02:
Is this in place now? If so I'll try to test it out.

Thanks!

tagyoureit, 2017-12-02:
Yup... check it out.

tagyoureit, 2017-12-19:
Please reopen if this doesn't work on your end.

arrmo, 2017-12-20:
My apologies - I did test this last night, just hadn't had a chance yet to post the results. Was sidetracked with dual servers this week, and ST (which I'm really starting to like!). 

Unfortunately this didn't quite work for me. I shut down the socat server, and saw the following in the logs,
```
15:44:37.807 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
15:44:42.687 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
15:44:47.807 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
15:44:52.688 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
15:44:57.808 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
15:45:02.689 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
15:45:07.809 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
15:45:17.811 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
```

This is perfect, as this stopped once I started the remote socat server. But ... even though no more errors, and nodejs-poolController seems to be connected, it didn't communicate / work. Rather, I had to restart npm, then it all worked again.

Would some other logs help?

Thanks!

tagyoureit, 2017-12-20:
Strange... works here.  :)

```
21:23:45.078 VERBOSE Msg# 42   Main --> Pump 2: Provide status: [165,0,97,16,7,0,1,29]
21:23:45.116 DEBUG Msg# 43  Incoming pump packet: 165,0,16,97,7,15,4,0,0,0,0,0,0,0,0,0,0,0,0,22,26,1,96
21:23:45.116 DEBUG Msg# 43  Msg received: 165,0,16,97,7,15,4,0,0,0,0,0,0,0,0,0,0,0,0,22,26,1,96 
                           Msg written: 255,0,255,165,33,16,33,203,1,14,1,209 
                           Match?: false
21:23:55.120 ERROR Net connect (socat) connection lost.  Will retry every 10 seconds to reconnect.
21:23:55.144 ERROR Error with port: connect ECONNREFUSED 11.11.11.170:9801.  Will retry in 10 seconds
21:24:05.140 ERROR Error with port: connect ECONNREFUSED 11.11.11.170:9801.  Will retry in 10 seconds
21:24:15.134 VERBOSE Net connect (socat) not receiving data.  Will retry connection...
21:24:17.009 DEBUG Msg# 44  Incoming controller packet: 165,33,15,16,2,29,22,26,0,64,0,0,0,0,0,0,3,0,64,4,50,50,255,255,255,255,255,255,255,255,0,255,165,33,15,16,2
```

Are you sure that there just weren't any messages to log?  Maybe try with verbose/debug level like I have above.  If the Error with port messages ceased, then communications must have been re-established, yes?  

(And I may have some logging backwards, the line with "Net connect (socat) not receiving data.  Will retry connection..." should be earlier... will have to look at that.)

arrmo, 2017-12-20:
Of course it does (work there) ... ;-). Just teasing - have to have some fun!

OK, tried it here, with more verbose logs. Disabled socat (remotely), got the following ...
```
07:13:06.112 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
07:13:16.116 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
```

Then re-enabled socat (again, remote), logs as below - and cannot seem to communicate / do anything,
```
07:13:26.123 VERBOSE Net connect (socat) not receiving data.  Will retry connection...
07:13:26.329 VERBOSE smartthings Sent temp'{"temperature":{"poolTemp":52,"spaTemp":52,"airTemp":54,"solarTemp":0,"freeze":0,"poolSetPoint":82,"poolHeatMode":0,"poolHeatModeStr":"OFF","spaSetPoint":102,"spaManualHeatMode":"Off","spaHeatMode":1,"spaHeatModeStr":"Heater","heaterActive":0}}'
07:13:30.263 VERBOSE smartthings Sent temp'{"temperature":{"poolTemp":53,"spaTemp":53,"airTemp":54,"solarTemp":0,"freeze":0,"poolSetPoint":82,"poolHeatMode":0,"poolHeatModeStr":"OFF","spaSetPoint":102,"spaManualHeatMode":"Off","spaHeatMode":1,"spaHeatModeStr":"Heater","heaterActive":0}}'
07:13:40.092 VERBOSE smartthings Sent temp'{"temperature":{"poolTemp":52,"spaTemp":52,"airTemp":54,"solarTemp":0,"freeze":0,"poolSetPoint":82,"poolHeatMode":0,"poolHeatModeStr":"OFF","spaSetPoint":102,"spaManualHeatMode":"Off","spaHeatMode":1,"spaHeatModeStr":"Heater","heaterActive":0}}'
07:13:49.923 VERBOSE smartthings Sent temp'{"temperature":{"poolTemp":53,"spaTemp":53,"airTemp":54,"solarTemp":0,"freeze":0,"poolSetPoint":82,"poolHeatMode":0,"poolHeatModeStr":"OFF","spaSetPoint":102,"spaManualHeatMode":"Off","spaHeatMode":1,"spaHeatModeStr":"Heater","heaterActive":0}}'
```

Restarted nodejs-poolController => then it's all working again!

Thoughts?

tagyoureit, 2017-12-20:
Couple of questions...
1. when you say it cannot communicate, how is it sending different pool/spa temps to smart things after re-enabling socat?
2. how are you stopping/starting socat?  Maybe you are doing something different than me?
I am stopping/starting it by running `/usr/bin/socat TCP-LISTEN:9801,fork,reuseaddr FILE:/dev/ttyUSB0,b9600,raw`, then killing (CTRL-C) the process, then restarting it up again.  Can you try that way?  I don't have the daemon running because it messes with running the app on the same box.  How are you stopping/starting the Socat service?

arrmo, 2017-12-20:
OK, I may have an idea what is going on here. I tried it twice, killing my remote service and restarting ... it worked fine (as below) - but I made sure to not send any commands when the connection was down (which can happen, right?).
```
17:04:18.668 ERROR Net connect (socat) connection lost.  Will retry every 10 seconds to reconnect.
17:04:18.670 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
17:04:28.672 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
17:04:38.675 INFO Net connect (socat) recovering from lost connection.
17:04:45.508 INFO 
	{"text":"User request to toggle SPA LIGHT to on","status":"on","value":1}
17:04:47.803 INFO 
	{"text":"User request to toggle SPA LIGHT to off","status":"off","value":0}
17:05:01.870 ERROR Net connect (socat) connection lost.  Will retry every 10 seconds to reconnect.
17:05:01.872 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
17:05:11.876 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
17:05:21.878 INFO Net connect (socat) recovering from lost connection.
17:05:30.519 INFO 
	{"text":"User request to toggle SPA LIGHT to on","status":"on","value":1}
17:05:32.320 INFO 
	{"text":"User request to toggle SPA LIGHT to off","status":"off","value":0}
```

Next, I did send a command when the connection was down -> you can see that failure below (socket is closed). But then, on recovery things don't work (like they did above) - meaning that the spa light does not turn on. So the problem is if a command is sent while the link is down? And I did have to restart the remote connection as well in this case, even reconnect didn't succeed the first time.
```
17:05:46.386 ERROR Net connect (socat) connection lost.  Will retry every 10 seconds to reconnect.
17:05:46.389 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
17:05:49.421 ERROR Error with port: This socket is closed.  Will retry in 10 seconds
17:05:49.421 INFO 
	{"text":"User request to toggle SPA LIGHT to on","status":"on","value":1}
17:05:49.422 ERROR Error writing packet (255,0,255,165,1,16,33,134,2,4,1,1,100): This socket is closed
17:05:56.391 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
17:05:56.392 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
17:06:06.394 INFO Net connect (socat) connected to: atticServer:9801
17:06:06.394 ERROR Error with port: connect ECONNRESET 192.168.2.65:9801.  Will retry in 10 seconds
17:06:16.396 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
17:06:26.396 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
17:06:36.399 INFO Net connect (socat) recovering from lost connection.
17:06:49.496 INFO 
	{"text":"User request to toggle SPA LIGHT to on","status":"on","value":1}
```

After I get in this state, I have to kill and restart nodejs-poolController (and refresh the web page sometimes it seems) to get it all working again ... all without anything happening on the remote end.

Make sense? Can you try with the send attempt while the link is down?

tagyoureit, 2017-12-21:
Ah, great catch!  I never did test that.  And what was happening was the packets were being queued up in the write queue and then getting stuck there.  So every time you tried to do an action it was "frozen" because it was waiting for a response to a command it tried to send when the link was done.

This probably isn't the most elegant approach, but my fix was to wipe out the write queue at the time we recover from the error.  I'll keep an eye on it, but this should suffice for a quick fix for now.

let me know...

arrmo, 2017-12-22:
Yep, thought it may be something like that. Updated to the latest code, but no joy - it behaves the same ... recovers find if nothing attempted to be sent, but if I try to send, then no recovery. And just restart nodejs-poolController, and all is working fine (so the remote end is OK). Here are the logs for this case,

```
19:31:49.511 INFO 
	{"text":"User request to toggle SPA LIGHT to on","status":"on","value":1}
19:31:49.512 ERROR Error writing packet (255,0,255,165,1,16,33,134,2,4,1,1,100): This socket is closed
19:31:52.956 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
19:31:52.957 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
19:32:02.961 INFO Net connect (socat) connected to: atticServer:9801
19:32:02.962 ERROR Error with port: connect ECONNRESET 192.168.2.65:9801.  Will retry in 10 seconds
19:32:04.398 VERBOSE smartthings Sent temp'{"temperature":{"poolTemp":54,"spaTemp":54,"airTemp":63,"solarTemp":0,"freeze":0,"poolSetPoint":82,"poolHeatMode":0,"poolHeatModeStr":"OFF","spaSetPoint":102,"spaManualHeatMode":"Off","spaHeatMode":1,"spaHeatModeStr":"Heater","heaterActive":0}}'
19:32:09.456 WARN Setting logging level to info
19:32:12.962 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
19:32:17.367 INFO 
	{"text":"User request to toggle SPA LIGHT to on","status":"on","value":1}
19:32:22.963 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
19:32:32.964 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
```

You can see, after "recovery" it's still looking for the remote connection.

tagyoureit, 2017-12-22:
Can you make sure that you have the two init statements @ https://github.com/tagyoureit/nodejs-poolController/blob/4.x-DEV/src/lib/comms/sp-helper.js#L83?  This should be where the queue is flushed.  If you can also include this interruption/failed recovery with the logs per #71, I should be able to get a better idea of what's going on.

arrmo, 2017-12-22:
Yep, from (Visual Studio) code on my machine,
[image]

tagyoureit, 2017-12-22:
Ok, interesting... in your logs I see:
```
06:11:22.757 ERROR Net connect (socat) connection lost.  Will retry every 10 seconds to reconnect.
06:11:22.757 SILLY Outputting socket  outputLog
06:11:22.761 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
06:11:22.761 SILLY Outputting socket  outputLog
06:11:32.763 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
06:11:32.764 SILLY Outputting socket  outputLog
06:11:42.468 ERROR Error with port: This socket is closed.  Will retry in 10 seconds
06:11:42.468 SILLY Outputting socket  outputLog
06:11:42.469 INFO 
	{"text":"User request to toggle SPA LIGHT to on","status":"on","value":1}
06:11:42.469 SILLY Outputting socket  outputLog
06:11:42.470 ERROR Error writing packet (255,0,255,165,1,16,33,134,2,4,1,1,100): This socket is closed
06:11:42.470 SILLY Outputting socket  outputLog
06:11:42.765 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
06:11:42.766 SILLY Outputting socket  outputLog
06:11:42.767 ERROR Error with port: connect ECONNREFUSED 192.168.2.65:9801.  Will retry in 10 seconds
06:11:42.767 SILLY Outputting socket  outputLog
06:11:52.264 WARN Setting logging level to info
06:11:52.264 SILLY Outputting socket  outputLog
*06:11:52.768 INFO Net connect (socat) connected to: atticServer:9801*
06:11:52.769 SILLY Outputting socket  outputLog
**06:11:52.770 ERROR Error with port: connect ECONNRESET 192.168.2.65:9801.  Will retry in 10 seconds**
06:11:52.770 SILLY Outputting socket  outputLog
06:11:52.776 SILLY pBTA: bufferToProcess length>0;  bufferArrayOfArrays>0.  CONCAT AoA to BTP
```
The line that is *italics* above is when you reconnect, but the **bold** line seems to throw an error right after that.  I think this was throwing off my logic of flushing the write queues.  I made a small change and moved the commands to flush the queue outside of the if/else statement.  Now, no matter when the serial/net port connects/reconnects, it will flush the queue.  

Can you try it out.  It should be version 4.1.3 now.  👍

arrmo, 2017-12-22:
OK, pulled down 4.1.3 ... :-). It behaves the same though, still showing that error after reconnect.

tagyoureit, 2017-12-22:
Eh?  No change?  I would expect the error message to be the same, but it should be flushing the queues now.  Very strange...   Let me think on this.  Can you send me a log file with *all* the logging enabled?  Maybe I'll see something that I haven't thought of before.

arrmo, 2017-12-23:
Sure, NP! Here is the full log. BTW, this time it recovered once - but not the second time. The only slight difference was that I tried to send a command after restarting socat (but not detected yet by nodejs-poolController?).
[output.log](https://github.com/tagyoureit/nodejs-poolController/files/1583798/output.log)

tagyoureit, 2017-12-24:
Um, please kindly stop being so good at finding bugs.  🤕    

I think I see what happened with your second attempt.  The way the code is written if I detect an error, I set a timer for 10 seconds to try to reconnect (reopen) the port/net connection.  When we do successfully reopen the port, I'm (now) clearing the queue.  However, in this case you re-connected before the 10 seconds and (I think) the port somehow reconnected itself or at the very least we don't see a successful open event so the code to clear the queue was never triggered.  

Regardless, I see where an error is properly detected when writing a packet, but I do nothing with it for both [net connect](https://github.com/tagyoureit/nodejs-poolController/blob/75625ad09b9430eacc1ff340b09e3fc3165987ae/src/lib/comms/outbound/write-packet.js#L183) and serial port, just above it.

I think the best thing to do at this point in the code is just discard the packet.  Reason being if you request to turn on the spa at 4pm, but somehow the socat connection is down from 3:59pm until 2am you wouldn't want the spa turning on (or any other commands running that were previously queued).  Let me know if you have other thoughts, but I'll test out this approach.

tagyoureit, 2017-12-24:
I just pushed this up... it's not perfect (eg if you disconnect socat before the configuration is done, it will just cancel it without retrying... maybe fix that later), but it catches errors upon write, and also discards packets that are queued when the port isn't open.  (<10 seconds will fire an error on the port and close it, basically).

test it out!

arrmo, 2017-12-24:
LMAO - too funny ... ;). Unfortunately I'm even better at introducing bugs ... :(.

Will give it a try, thanks!

arrmo, 2017-12-24:
This version seems to work great - or at least I can't break it ... ;). Closing this now, reopen if you disagree, but looks solid to me.

Thanks!!!

And have a great holiday break!

arrmo, 2018-09-05:
Dang it - OK, breaking it now ... :-(. I seem to have to restart nodejs-poolController for this to reconnect. Is it just me?

Thanks!

arrmo, 2018-09-08:
Absolutely! Trying it ...

tagyoureit, 2018-11-16:
Let's start a new case if there are still issues here.

closed 2018-11-16 by tagyoureit