tagyoureit/nodejs-poolController #17: Bootstrap min errors?

tagyoureit, 2017-01-25:
This morning in loading the app in Safari (Mac) 10.0.2, I had the following errors in the web console: 
[image]

Doing a quick search, I found this error. [stack overflow]) http://stackoverflow.com/questions/35298917/unexpected-css-token-bootstrap-gradient-carousel)
and also [this](https://github.com/FortAwesome/Font-Awesome/issues/5043#issuecomment-148901292) thread in an unrelated project.

It didn't seem to affect anything, but I don't remember seeing it before.

arrmo, 2017-02-20:
I haven't seen any such errors - are you still seeing them? But I admit - close this, as it's in the Bootstrap code itself (that we really don't want to start messing with, agreed?)?

tagyoureit, 2017-02-20:
yup... I'll close it.  Still seeing them, but not bothering anything...

closed 2017-02-20 by tagyoureit