tagyoureit/nodejs-poolController #18: Freeze Protection

arrmo, 2017-01-27:
Not working right on my Intellitouch ... :-(.

arrmo, 2017-01-27:
FYI, 
[image]

tagyoureit, 2017-01-27:
Yup, I haven't push it yet.  It will be in with the next push.  If you want to test it on your end, change the end of line 40 in `lib/comm/inbound/controller/2.js` to `(data[c.controllerStatusPacketFields.UOM] & 8) >> 3`.

arrmo, 2017-01-27:
NP, will wait for the push - then test it here (supposed to be cold again tomorrow morning ... :-().

tagyoureit, 2017-02-06:
I think we are good on this.  Let me know if we need to reopen.

closed 2017-02-06 by tagyoureit

arrmo, 2017-02-06:
Will do - 81F here today, no freeze protection on ... LOL