tagyoureit/nodejs-poolController #19: Bootstrap fixed top menu

tagyoureit, 2017-02-11:
I was playing around with the app on my mobile site and found it a tad bit difficult to find items because there was so much vertical scrolling.  I looked through the Bootstrap documentation and implemented a nav-bar that is responsive to the device size.

This changes the top menu to a "fixed" style so it is always on the top.  To me, the advantage of 2-quick touches to get to any menu is faster than scrolling.  And, bonus, even with panels rearranged it works perfectly.  The debug timer/switch also gets formatted to the bottom of the menu in smaller windows.

[Archive.zip](https://github.com/tagyoureit/nodejs-poolController/files/768242/Archive.zip)

Only thing I didn't explore is do we hide the menu items if they are hidden in `clientConfig.json`?  I'd say yes, but didn't look into how to do this.  Hopefully not much harder then hiding an individual panel.

Here are the main elements (and full files attached).

1. added `navbar-fixed-top` to navbar class.
2. added `body {padding-top: 70px;}` to css to move all content down 70px.
3. added menu items per documentation (https://getbootstrap.com/examples/navbar-fixed-top/)
4. added anchor links just below each div with pnlSystem class : `<a name="systeminformation" class="anchor"></a>  `
5. the `class="anchor"` is needed to also adjust each link down 70px so it is below the menu:
```
.anchor::before {
  content:"";
  display:block;
  height:70px; /* fixed header height*/
  margin:-70px 0 0; /* negative fixed header height */
}
```
6. added the following to main.js to make the menu item active and the collapse the navbar (mobile size) when you select an item.
```
	//set active menu item
		$(".nav li").on("click", function() {
			$(".nav li").removeClass("active");
			$(this).addClass("active");
		});
 //and collapse navbar when selecting
 $('.navbar-collapse a').click(function(){
    $(".navbar-collapse").collapse('hide');
});
```

arrmo, 2017-02-20:
Sorry, trying to understand this one. Do you mean just the single one at the top (overall), or for each panel?

tagyoureit, 2017-02-20:
No, this is the overall header.  It will make the menu persistent at the top so when scrolling (especially on mobile) it is only 2 clicks to get to any section.

tagyoureit, 2017-02-20:
If you want me to make these changes and push it up for your review, let me know.

arrmo, 2017-02-25:
A few questions, 
- Use a dropdown, rather than "buttons" across the top? I assume so, to avoid the horizontal space needed
- Move some of the top information to the bottom, but then also use a sticky footer (see http://getbootstrap.com/examples/sticky-footer-navbar/)

Thoughts?

tagyoureit, 2017-02-26:
The great thing about the CSS I added is that when it is on a large screen, they are buttons across the top; and when it is on a small scree the buttons do become a dropdown.  And the other elements that are at the top (debug/update(s)) also fall to the bottom of the new dropdown menu.  I'd rather not take up space at the top _and_ bottom if at all possible.

arrmo, 2017-02-26:
OK, will fiddle with this. But really thinking the buttons on the top right should go at the bottom ... OK with you? Or perhaps - keep the version warning there (if visible), but move the time update and debug button to the bottom right ... OK? They are not used much - don't need to be "front and center".

tagyoureit, 2017-07-04:
I implemented this... although I couldn't figure out how to get the 'last update' to always float:right.  The navbar seems to shrink to the left when the screen size is `lg` or larger.

closed 2017-07-04 by tagyoureit

arrmo, 2017-07-10:
Thanks! Sorry, have been a bit sidetracked lately ... :-(. Is this in the latest DEV (to check it out)?

tagyoureit, 2017-07-10:
No worries!  Yes, this is the latest in the Dev branch.  
[image]
[image]

You can see in the two screenshots when the width is narrow the timer floats on the right next to the menu.  When the menu disappears the timer floats on the left.

One thing I also didn't think about... if the order of the panels are dragged around, the list of menu items won't change.  Shouldn't be hard to do, but now it's on the list.