tagyoureit/nodejs-poolController #20: Move Web Directories

arrmo, 2017-02-11:
Move /public and /bootstrap under www (new directory), to allow either to be accessed.

arrmo, 2017-02-12:
OK, this seemed easy ... ;-). And it's the first step - dang it!

This can be done by just changing expressDir, to /www in config.json. Works ... sort of - but then libraries, etc. seem to be missing. Is it because expressDir is used for more than just the client code?

tagyoureit, 2017-02-12:
Assuming you have `/www/bootstrap` (and `/www/public/`), you need to change the script references.

```
	<link type="text/css" rel="stylesheet" href="/bootstrap/css/bootstrap.min.css">
	<link type="text/css" rel="stylesheet" href="style.css">
	<script src="/jquery/external/jquery/jquery.js"></script>
	<script src="/jquery/jquery-ui.min.js"></script>
	<script src="/bootstrap/js/bootstrap.min.js"></script>
	<script src="/socket.io/socket.io.js"></script>
```

Otherwise, browsers will be looking for the libraries in `/www/bootstrap` when they are in `/`.  But is the bigger question how will the user access the UI?  Are we moving from `http://localhost:3000` (chosen by the expressDir variable) to `http://localhost:3000/bootstrap` and `http://localhost:3000/public`?  Or are you going to have it automatically redirect to `/bootstrap` if the user doesn't specify the directory?

arrmo, 2017-02-12:
Yep, agreed! Fighting with that now .. :-).

Was thinking as you're saying - access at http://localhost:3000/bootstrap and/or http://localhost:3000/public/, default to bootstrap (if not entered) ... allows other UI's later, and more importantly, not having to reset the server every time switching between the two.

OK with you? But also need to figure out the references - as you say, they are above /www (directory) now, not sure how to get there ... :-(.

arrmo, 2017-02-12:
OK, fixed the bootstrap ones easily, trying to figure out how to get to jquery and socket.io (as they are "published" by nodejs, right?).

tagyoureit, 2017-02-12:
They are, but their directories haven't change.  I used the references I put above and it works fine for me.  Previously, we had `<script src="socket.io/socket.io.js"></script>` so socket.io was being called from the `/` directory (because we made the physical `/bootstrap` map to `/`).  Now that we mapped `/www/bootstrap` to `/bootstrap` the references aren't valid anymore.  Changing it to `<script src="/socket.io/socket.io.js"></script>` makes the reference valid again.  What errors are you getting in the browser error console?

arrmo, 2017-02-12:
Got it worked out! ... :-). Trying to get the Date Setting button working right now - once I do, will push so you can check it out also ... OK?

Will need one thing from you though - it's on the server side that will need to default to /www/bootstrap if not set, agreed? I admit, with the refactoring of the code (which is excellent!), I'm just not sure where to add the logic, and don't want to break anything. Can you take this part on, once I push my code?

Thanks!

tagyoureit, 2017-02-12:
I could do that... but if someone wanted to change it for their local purposes it would be easier for them to do it in a HTML file.  If you put an `index.html` in the `www/` file with a meta-redirect
```
<meta http-equiv="refresh" content="0; URL='bootstrap/index.html'" />
```
then it will redirect to bootstrap by default.  We could look at doing this server side, but I'm hard pressed to think of any advantages right now.

tagyoureit, 2017-02-12:
BTW, this is where the jquery and bootstrap files are mapped: https://github.com/tagyoureit/nodejs-poolController/blob/master/src/lib/comms/server.js#L60

arrmo, 2017-02-12:
Yep, that works! Will add in that html file.

Code working, including DatePicker => just need it to call the API, make sure that works ... then code for you to test!

arrmo, 2017-02-20:
Done, and in the code now - closing (but reopen if needed).

closed 2017-02-20 by arrmo