tagyoureit/nodejs-poolController #21: Change button layout for mobile devices

tagyoureit, 2017-02-11:
The buttons are really small on mobile devices.  One suggestion would be to just make them bigger.  Another would be to eliminate the labels and place them on the buttons themselves.  We could then dual-column the panel, like in the attached pic.

[image]

Of course, adding some space between the buttons... but I think this just means adding a couple of extra classes (eg `<div class="col-xs-6 col-md-6">`).  I used DIV's to mark this up, bit might work withy our existing table layout as well.

arrmo, 2017-02-20:
Thinking about this - not quite sure yet what to do about it. I'm not sure just making everything a button works, and to be honest then it gets a bit messy on a PC, and is also difficult to follow. We could change themes ... less white space, larger font, etc.

Thoughts?

tagyoureit, 2017-02-20:
The bootstrap code allows for responsive layouts,right?  So in theory we should be able to keep the layout we have now for PC/big screen, and add code so that when the screen is more narrow it changes to the more button-like layout.  Have you played with that at all?

arrmo, 2017-02-25:
Check out PR #33, Pool (only) ... better? I can fix this throughout if it's better.

tagyoureit, 2017-02-25:
Ya, on iPhone it is much better.  On iPad it is also better, but we need to adjust one of the widths because the buttons wrap.  See pics.

[image]
[image]

arrmo, 2017-02-26:
Yep, agreed. So make them all larger like this, and fix the wrapping - good with you?

tagyoureit, 2017-02-26:
Yup.  👍

arrmo, 2017-02-26:
Perfect - will take care of it. I also want the Parameter (on the left) centered vertically - need to figure that out. I know, minor, but I'm fussy about that ... :)

arrmo, 2017-02-26:
Addressed - buttons updated, alignment (css) fixed, and wrapping as well (also css). Will close, reopen if needed.

BTW, this is PR #36.

closed 2017-02-26 by arrmo