tagyoureit/nodejs-poolController #22: Create backend Socket call/code to modify clientConfig.json programatically.

tagyoureit, 2017-02-11:
Tag to enable this, then pass to @arrmo for bootstrap integration.

tagyoureit, 2017-02-11:
Ok, done on my end.  :-)

There is a new socket you can call...
`socket.emit('setConfigClient', a,b,c,d)`

a,b,c,d are the keys in the configClient.json file.  So if you want to change system startup state to hidden you would call:
`socket.emit('setConfigClient', 'panelState', 'system', 'state', 'hidden')`

for a property like hideAUX that only has 2 levels, just call:
`socket.emit('setConfigClient', 'generalParams', 'hideAUX', null, true)`
Notice the true can be passed as a boolean and not a string.

So the question on your end is...
a) do you want to have a separate place in the UI to save/configure?  Or should it permanently write the changes to the file if you minimize.  
b) building on the above, maybe just add a remove button next to minimize and it will set the panel to 'hidden'?
c) and building on that, then we would need a place to make the panel visible again.

Let me know if this makes sense or I should change anything around.  This should 100% eliminate the need to touch the .json file itself (except for upgrading to a new version of the code?? all defaults would be restored... will need to think about that).

Tag

arrmo, 2017-02-20:
Yep, agree with you! I'm thinking minimize does it automatically (as you note) ... it's just the hidden part that's a bit more difficult. Or perhaps have a single button at the top (or bottom) to reset (remove all hidden settings)? I'm not thinking this changes a lot once customized ... agreed?

tagyoureit, 2017-02-20:
sure, let's go with a single reset button... that works for me.

arrmo, 2017-02-20:
Will do. First off, try the debug log - to make sure this is all working. Hmmm, here is what I have now, and it is being called => but not setting the state. Thoughts?

	// Button Handling: Debug Log => On/Off
	$('#debugEnable').click(function () {
		if ($('#debug').is(":visible") === true) {
			$('#debug').hide();
			setStatusButton($('#debugEnable'), 0);
			socket.emit('setConfigClient', 'panelState', 'debug', 'state', 'hidden')
		} else {
			$('#debug').show();
			setStatusButton($('#debugEnable'), 1);
			socket.emit('setConfigClient', 'panelState', 'debug', 'state', 'visible')
		}
	});

tagyoureit, 2017-02-20:
Had a small error on my side... yup, it happens even with testing.  Pull down the version I just pushed (modified a path in /src/lib/helpers/bootstrap-config-editor.js) and should be good now.  Working on my side!

arrmo, 2017-02-20:
No worries - it does happen!

Did you push to master for DEV? Pulled down DEV, not working here yet .. :-(.

tagyoureit, 2017-02-20:
Yes, pushed to Dev.  Are you getting any errors?

arrmo, 2017-02-20:
Hmmm ... very odd. Do you send back a debug message (INFO?) when a setting is configured? I'm not seeing any message, and no errors ... just goes in to a black hole ... ;-).

tagyoureit, 2017-02-20:
I have this in my logs `11:50:58.746 VERBOSE Updated configClient.json`.   If you set the extLogLevel = verbose than it should show up in the debug log as well.

arrmo, 2017-02-21:
OK, I think it was a bonehead on my part ... :-(. I had the extLogLevel set to debug, changed it to verbose, and then restarted the service (I have the node.JS side of things as a service). It wasn't the log level change, but rather the restart ... didn't do that after pulling down your update.

Sorry! Obviously, working fine here now ... ;-). So closing - it works, use it for other features now. 

Next up ... add an X to hide, and a reset button to clear.

arrmo, 2017-02-21:
OK, also added in the ability to hide panels now ... check it out, let me know. Not setting the config file yet (that's real easy, but want the reset button first ... :-))

arrmo, 2017-02-25:
See PR #35 - adds storage of panel hide and reset ... but, sometimes I see in the client side log,
16:31:29.701 WARN Error updating bootstrap configClient.json: 

Check this out on the server side perhaps? Reopening until we have this rock solid. Perhaps a delay betwen messages (or buffer on the server side)?

tagyoureit, 2017-02-26:
I pulled #35.  For a few times, it worked to hide the panels (and I watched it modify the configClient.json.  Woot!  

Then it stopped working after I made some changes.  I had this error:
> main.js:388ReferenceError: Can't find variable: generalParams

I think this might just be a race condition where we are not guaranteed to finish reading the configClient.json before the socket calls are received??  Not 100% sure.

And 2nd thing was that the reset button did unhide the panels in the UI, but the changes weren't being written to the file.  This could have to do with the above error I was seeing.  Looks like when the error happens it shuts down the socket but can still do local actions like hiding the panels.

tagyoureit, 2017-02-26:
I wonder if something is wrong on my side.  I'm seeing (in the console logs):
```
17:05:23.341 WARN Error updating bootstrap configClient.json: 
	{"message":"Unexpected token }","stack":"SyntaxError: Unexpected token }\n    at Object.parse (native)\n    at /Users/rgoldin/Documents/programming/nodejs-poolController/src/lib/helpers/bootstrap-config-editor.js:39:33\n    at notify (/Users/rgoldin/Documents/programming/nodejs-poolController/node_modules/promised-io/promise.js:201:22)\n    at notifyAll (/Users/rgoldin/Documents/programming/nodejs-poolController/node_modules/promised-io/promise.js:180:5)\n    at Deferred.resolve.callback.emitSuccess (/Users/rgoldin/Documents/programming/nodejs-poolController/node_modules/promised-io/promise.js:223:3)\n    at arguments.(anonymous function) (/Users/rgoldin/Documents/programming/nodejs-poolController/node_modules/promised-io/promise.js:705:15)\n    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:380:3)"}
```

Basically, there is an extra `}` at the end of the file.  So it isn't loading (on your end) or parsing on my end correctly.... 

After a little basic testing, I can expand/collapse/hide the panels with no problem.  The issue seems to be when I press the reset button.  I have to reload the page and then see the error in the configClient.json.

tagyoureit, 2017-02-26:
A thought... I'm using Async code on the backend to write the configClient.json.  I wonder if there is an issue when you send the reset (that it actually sends 9 back-to-back socket calls) that they are being received faster than I can write the file (and are possibly reading/writing it at the same time?).

I can try to figure out something on my end... or, we could make a separate Socket call (e.g. "resetConfigClient") and I just take care of setting everything to "visible" in one pass on my end.

arrmo, 2017-02-26:
Yep, that makes sense - and I was thinking the same thing ... too fast (socket calls). I can emit a new call if you want, let you cycle through to reset. Just let me know, I can update to do that.

tagyoureit, 2017-02-26:
Sure... Let's go with the `resetConfigClient`.  Should it do anything with the `hideAux` variable?  Or just leave it alone?

tagyoureit, 2017-02-26:
Why don't you push this change up and I'll accept the PR?  Problem on my end is I'm waaaaay deep in making the new config.json changes and other stuff and (of course), did create a new branch.  So for me to push stuff up now it will break lots of things.  (The next push will have lots of changes.)  So easier for me to accept your small change now and likely my next push will be a 3.2.0 beta candidate.

arrmo, 2017-02-26:
Are we mixing two items here? Or may just be me ... :-(.

This is a new message, and then on the server side reset the configuration ... right? The config.json change is for the version notification (different issue).

arrmo, 2017-03-23:
I believe this is working now (changing parameters) - close it?

Reopen if I'm wrong, or missing something. Thanks!

closed 2017-03-23 by arrmo