tagyoureit/nodejs-poolController #23: Include Heater State in 'temp' message?

arrmo, 2017-02-12:
Hi,

Why have the heater state in it's own message - rather include it in 'temp' (like Freeze Protection state)? Then it can be set to off also, when the 'temp' message is sent (and cleans up code a bit).

Thoughts?

tagyoureit, 2017-02-20:
Just pushed up a change so that `/temperatures` now returns a combined heat and temperatures object like:
```
{"poolTemp":0,"spaTemp":0,"airTemp":0,"solarTemp":0,"freeze":0,"poolSetPoint":82,"poolHeatMode":0,"poolHeatModeStr":"OFF","spaSetPoint":100,"spaHeatMode":1,"spaHeatModeStr":"Heater","heaterActive":0}
```

Reopen this if it breaks anything.  :-)

arrmo, 2017-02-20:
So is the old 'config' message removed (so not duplicated)? Just so I know how to adapt here.

tagyoureit, 2017-02-20:
I left the `heat` message the exact same as it was.  It can be removed in the future (I'll combine them).  Do use the new `temperatures` one now.  There wasn't a `config` message, though...?

arrmo, 2017-02-21:
I think it was 'config', no? This was actually what you added .. :-). Don't mean that negative, it works,

	socket.on('config', function(data) {
		if (data !== null) {
			$('#stateHeater').html(data.HEATER_ACTIVE);
		}
		lastUpdate(true);
	});

So remove this, and set this based on 'heaterActive' inside 'temp' ... right?

tagyoureit, 2017-02-21:
Hmmm, maybe that's why it isn't working.  I commented out that code a while ago!  Lol.  Try just changing that (or combining it with) `temperatures`.

tagyoureit, 2017-02-21:
Oh, and it is `temp` in the socket but `temperatures` in the URI.  I should fix that somewhere...

Here is where the code is disabled, fwiw.  https://github.com/tagyoureit/nodejs-poolController/blob/master/src/lib/comms/socketio-helper.js#L49

arrmo, 2017-02-21:
No worries! So I'm OK to blow away 'config', just use 'temp' - right?

Will update it tonight, pull request so you can check it out.

arrmo, 2017-02-21:
OK, PR is there (PR #27), including sync to your latest changes.

tagyoureit, 2017-02-21:
Per my comment in the PR... now that we have this all "fixed", it doesn't really work well because I'm not exactly capturing the right packets.  Status is never going to be updated right until I understand the messages better.  So maybe best to just hide it for now?

arrmo, 2017-02-21:
Yep, will do - PR coming tonight.

arrmo, 2017-02-25:
This one is in place now (and hidden for the time being). Closing, reopen if needed!

closed 2017-02-25 by arrmo