tagyoureit/nodejs-poolController #30: Add notification to UI if there is as newer release available

tagyoureit, 2017-02-23:
Will add code to the main app and create a socket & console notification.  Then hand off to @arrmo to include in bootstrap UI.

tagyoureit, 2017-02-24:
When you get to it... I added a socket to let the UI know if the running version is older, newer or equal to the _latest published release_.  (It right now does not take into account Dev branches, but I can add that...)

socket name:  `updateAvail`
Results look like: 
```
{"local":{"version":"3.1.11"},"remote":{"tag_name":"v3.1.9","version":"3.1.9"},"result":"newer","resultStr":"You are running a newer release (3.1.11) than the published release (3.1.9)"}
```

You can either just display the `resultStr` or use the `result` (newer, older, equal) to display something.  (local is what is running the app; remote is what is available on Github)

I'm thinking we use the alerts (http://getbootstrap.com/components/#alerts-examples)
1. Equal - don't show anything
2. Newer - (using a dev build) - don't show anything
3. Older - Show the alert-warning type message.  And maybe be able to set this to `hidden` in case someone doesn't want to update.  (We would need to ask ourselves if this should permanently hidden, or hidden until the next release is available.)

arrmo, 2017-02-25:
Hmmm ... added this, but not seeing any traffic coming across. Are you sure it's sending?

arrmo, 2017-02-25:
Check out PR #33.

tagyoureit, 2017-02-25:
Whoops!  

This should be: `socket name: updateAvailable`

I changed it on my end and now see the "newer" tag now, so it is receiving data.

arrmo, 2017-02-25:
Works now ... :-).

Add PR #34, for color coding - seems to work here, check it out (and reopen if issues!).

tagyoureit, 2017-02-25:
Works great in its current form.  Just a quick thought to maybe be able to dismiss the message by clicking on it?  Both the yellow and red versions stick out like a sore thumb, and the green isn't as bad but really is superfluous.  And if we do that, thoughts about making it persistent (we could add the state, and the remote version to the JSON and then only display it again if it is a new version or we reset the entire set of panels?

arrmo, 2017-02-26:
Works for me - I can make it hidden, only display when I receive an message, hide on click. Work for you?

tagyoureit, 2017-02-26:
> only display when I receive an message

yes, but does "every time" mean each time the page is loaded or browser refreshed?  Or are you thinking of adding it to `panelState` as a variable?  (And maybe the remoteVersion in generalParams?  Or keep them together somewhere?)

arrmo, 2017-02-26:
Sounds good to me - so here are the steps, agreed?
1) Add variable to config.json -> a flag, say something like, versWarning ... true = display, hide = false.
2) I'll look at the variable, and based on it display or not. 
2a) If the button is pressed, the variable is set to false (hide).
2b) If a new state is received, different than the current / old one, then reset to true (display)

For 2b), does that mean we need another variable - the last state (say versLastState)?

arrmo, 2017-02-26:
The more I think about it, I think I need to add two variables - versLastState, and versWarning (as above). OK with you? If so, I'll add them to config.json.

To the "Misc" section?

arrmo, 2017-03-22:
OK, added to code - seems to be working. It's in PR #42. Please include, and reopen this if you see issues.

Thanks!

closed 2017-03-22 by arrmo