tagyoureit/nodejs-poolController #32: Proposed changes to config.json

tagyoureit, 2017-02-25:
I want to modify the `Equipment` section of the `config.json` to:

```
    "Equipment": {
        "controller": {
            "intellicom": 0,
            "intellitouch": 1,
            "appAddress": 33
        },
        "chlorinator": {
            "installed": 1,
            "standalone": 0,
            "desiredOutput": 0
        },
        "pump": {
            "standalone": 0,
            "numberOfPumps": 0,
            "pump1": {
                "program1rpm": 0,
                "program2rpm": 0,
                "program3rpm": 0,
                "program4rpm": 0
            },
            "pump2": {
                "program1rpm": 0,
                "program2rpm": 0,
                "program3rpm": 0,
                "program4rpm": 0
            }
        }
    },
```

It was not very clear before.  Hopefully, this helps get us in the right direction

- Pumps

I moved `pumpOnly` and `numberOfPumps` under the new `pump` section.  Also added a place for saving the program speeds. 

- Chlorinator
Now you can specify if it is installed, and if it is standalone.  Previously, it was confusing to set `chlorinator=0` if you had a chlorinator, but it was hooked up to a controller.  Also have `desiredOutput` to save the output % setting.

- controllers
Put Intellitouch and Intellicom und this heading, as well as appAddress.

Any comments?  I think we can more easily expand these sections now (like possibly saving a duration with the pump programs?  
Tag

arrmo, 2017-02-25:
No issues here.

tagyoureit, 2017-02-26:
[config.PROPOSED.txt](https://github.com/tagyoureit/nodejs-poolController/files/801662/config.PROPOSED.txt)

Ok, latest iteration attached...

tagyoureit, 2017-02-27:
More changes... A "virtual" controller for both pump(s) & chlorinator control.  A single place to describe the controller in use.  Moved the descriptions into an `id` object.  

[config.PROPOSEDv2.txt](https://github.com/tagyoureit/nodejs-poolController/files/805308/config.PROPOSEDv2.txt)

tagyoureit, 2017-10-05:
Promoted 4.0.  No more changes for now.

closed 2017-10-05 by tagyoureit