tagyoureit/nodejs-poolController #41: setHeatSetPoint, poolsetpoint, and spasetpoint change setpoints relative to current

leftyfl1p, 2017-03-08:
setHeatSetPoint, poolsetpoint, and spasetpoint socket.io endpoints currently change the setpoint of the equipment relative to the current setpoint.

Currently: poolsetpoint(80) adds 80º to the current setpoint, so if the current is 70 the result will be 150º (!).

What it should do: set the current setpoint to 80º.

tagyoureit, 2017-03-21:
@arrmo I'm thinking that it might be a good idea to have both relative and absolute calls for the heat set points.

1st challenge... we have both `setHeatSetPoint` which takes (equip, change) as variables; and also `spasetpoint` and `poolsetpoint`.  Should we rename these to be more consistent/descriptive?  Which

2nd challenge... it isn't (wasn't) clear if the values passed should be the delta in the change or the new desired value.

What do you think about:
* anything that starts with *set* (setHeatSetPoint, setPoolSetPoint (new), setSpaSetPoint (new)) would change to the new desired value?
* we add a new set of API's that start with *increment* or *decrement* (aka incrementHeatSetPoint) that would change the relative value.  If we do this, should it take a parameter?  Would the parameter default to 1 degree change if nothing is passed, or change by a # of degrees if a number is passed?

Thoughts?

arrmo, 2017-03-22:
Yep, I like that! And the naming as well. On my side, I can use either - just let me know when you change it, and I'll adapt right away.

tagyoureit, 2017-03-23:
Lol... I have a nice comment in my code https://github.com/tagyoureit/nodejs-poolController/blob/master/src/lib/equipment/heat.js#L143

```
/TODO: There should be a function for a relative (+1, -1, etc) change as well as direct (98 degrees) method
```

I just pushed this up.  Did some manual testing, but no automated testing yet.

Socket
Removed: changespasetpoint, changeheatsetpoint
Added: setPoolSetPoint, incrementPoolSetPoint, decrementPoolSetPoint
setSpaSetPoint, incrementSpaSetPoint, decrementSpaSetPoint

API
Removed: change...
Added poolheat/increment/:val, poolheat/increment (defaults to 1)
and variations for decrement and spaheat

arrmo, 2017-03-27:
OK, will give this a go. I assume you prefer I use increment and decrement (by 1) ... right?

tagyoureit, 2017-03-27:
Yes, I think that would be the most direct replacement.  If you want to go crazy, maybe if the end user clicks on the temperature itself (and not the +1/-1 buttons) it could bring up the option to type in a new temp (like the chlorinator setting).

arrmo, 2017-03-30:
OK, went to work on this, but ran in to an error along the way - with your push, did you also change the updateAvailable message? Seems the data / object is different that it was before.

Thanks!

tagyoureit, 2017-03-30:
Hmmm, I don't think so, but maybe.  I'm not getting an error on my side with the current code, but here is the output that I'm seeing:
```
{ local: { version: '4.0.0' },
  remote: { tag_name: 'v3.1.13', version: '3.1.13' },
  result: 'newer',
  resultStr: 'You are running a newer release (4.0.0) than the published release (3.1.13)' }
```

What error are you getting?

arrmo, 2017-03-31:
Hmmm ... here is what I'm receiving - seem to be missing result and resultStr?
[image]

arrmo, 2017-03-31:
Yep, reconfirmed ... I'm only receiving local and remote, not the other info any more. You?

leftyfl1p, 2017-07-08:
Fixed at https://github.com/tagyoureit/nodejs-poolController/commit/5640906834222ecbcb4ba0e42f2ef6fb16f7f8ef !

closed 2017-07-08 by leftyfl1p