tagyoureit/nodejs-poolController #46: REST api not working

dhop90, 2017-05-21:
I've gotten nodes-poolController working.  I can access the bootstrap interface and it gets updated fine.  But when I try to access the REST api it fails.

This is what I get when I try to access the api:
# curl http://pi-pool:3000/status
<!DOCTYPE html>
<html lang="en">
<head>
<meta charset="utf-8">
<title>Error</title>
</head>
<body>
<pre>Cannot GET /status</pre>
</body>
</html>

# curl http://pi-pool:3000/pool/status
<!DOCTYPE html>
<html lang="en">
<head>
<meta charset="utf-8">
<title>Error</title>
</head>
<body>
<pre>Cannot GET /pool/status</pre>
</body>
</html>

Is there a different URI I should be accessing to get to the REST api?

Thanks.

tagyoureit, 2017-05-21:
I took status out because it was a combination of other URI's that didn't make sense.  I need to remove it from the docs.  Everything else should work (circuit, chlorinator, pumps, etc) that is listed at https://github.com/tagyoureit/nodejs-poolController#rest-interface--socketio.

tagyoureit, 2017-05-22:
Removed from the 4.x (dev branch) docs.  Will push it up with the next set of changes.

closed 2017-05-22 by tagyoureit