tagyoureit/nodejs-poolController #47: Connecting to Hayward Aquarite Chlorinator

hunjuice, 2017-05-21:
I am trying to connect to a hayward Aquarite chlorinator. The Aquarite has a 4 pin connector labeled GRN (GND), YEL (485B) , BLK (485A), RED (-10v). I am not sure which wires to connect. I have the JBtek USB to RS485 but I cannot seem to get it working. Do I need a different connector?

thanks

tagyoureit, 2017-05-21:
You'll connect the JBTek to the 485A/485B terminals.  You'll need to provide power from another source if it isn't already hooked up to a power source.

hunjuice, 2017-05-21:
So I need to connect a 10v power source to the Aquarite on pins 1(-10v)  & 4 (Gnd) ?

tagyoureit, 2017-05-21:
Yup.  Typically this is powered either by the pool controller or a standalone chlorinator controller (e.g. https://www.hayward-pool.com/pdf/manuals/AquaRite-AQR-Owner.pdf).

hunjuice, 2017-05-21:
Awesome. I got it working. Is there a way to get the pool temperature from the aquarite? I do not have a heater so I would like to get the temp from the chlorinator. 

thanks for the help.

tagyoureit, 2017-05-21:
Unfortunately not. There is some internal temp sensor on these but the values aren't exposed through the serial bus. @johnny2678 (in Gitter forums) hooked up standalone temp sensors to his RasPi so that's likely the easiest option.

hunjuice, 2017-05-21:
Thanks, One more question, Should I be seeing constant log/status messages from the Aquarite or is the program just polling the when needed?

thanks again.

tagyoureit, 2017-05-21:
The program will just be listening unless you set it to be active. See https://github.com/tagyoureit/nodejs-poolController/blob/master/README.md#chlorinator-standalon.   

Despite the instructions you can set the level through the UI now.

tagyoureit, 2017-05-22:
BTW, feel free to pick up the latest Dev build (https://github.com/tagyoureit/nodejs-poolController/tree/4.x-DEV#virtual).  The logic has been re-worked on the back end for the standalone controllers (including chlorinator) and quite a bit more.

How did you end up with a salt cell and no controller, anyway?

hunjuice, 2017-05-22:
It works. I can access the salt status and set the output with no problems. I think all the code is over kill for my setup. I only have the Aquarite Salt generator and just need access to the salt level and the setting the output. I am alos adding  a ORP & PH probe as well. Based on the ORP & PH values, I will adjust the chlorinator output (increase/decrease). 

thanks again for the help.

closed 2017-06-03 by tagyoureit