tagyoureit/nodejs-poolController #51: I have the strings for the intellibrite control

netmagi, 2017-07-07:
Intellibrite LED's (last two columns are chksum):

Color Swim:             ff 00 ff a5 01 10 20 60 02 90 ff 02 c7
Party Mode:             ff 00 ff a5 01 10 20 60 02 b1 ff 02 e8
Romance Mode:           ff 00 ff a5 01 10 20 60 02 b2 ff 02 e9
Caribbean Mode:         ff 00 ff a5 01 10 20 60 02 b3 ff 02 ea
American Mode:          ff 00 ff a5 01 10 20 60 02 b4 ff 02 eb
California Sunset Mode: ff 00 ff a5 01 10 20 60 02 b5 ff 02 ec
Royal Mode:             ff 00 ff a5 01 10 20 60 02 b6 ff 02 ed
Blue:                   ff 00 ff a5 01 10 20 60 02 c1 ff 02 f8
Green:                  ff 00 ff a5 01 10 20 60 02 c2 ff 02 f9
Red:                    ff 00 ff a5 01 10 20 60 02 c3 ff 02 fa
White:                  ff 00 ff a5 01 10 20 60 02 c4 ff 02 fb
Magenta:                ff 00 ff a5 01 10 20 60 02 c5 ff 02 fc

tagyoureit, 2017-07-07:
Awesome... thanks.  Those should line up with what I have (https://github.com/tagyoureit/nodejs-poolController/blob/4.x-DEV/src/etc/constants.js#L426).  If you grab the 4.x branch I have some limited code in there to decode the lights.  I only have old-fashioned incandescent so am very happy to have help decoding/testing.

tagyoureit, 2017-07-10:
I should have mentioned... I'm in the middle of trying to code these in.  If you navigate to [circuit](http://11.11.11.170:3000/circuit) you'll see something like: 
```
"1": {
"number": 1,
"numberStr": "circuit1",
"name": "SPA",
"circuitFunction": "Spa",
"status": 0,
"freeze": 0,
"macro": 0,
"light": {
"group": -1,
"colorStr": "off",
"color": -1
},
"friendlyName": "SPA"
},
```

The light section *should* show the right values.  Let me know if this works for you.  If so, we can figure out how to expose it through the UI for display, and then implement the control to change it.

netmagi, 2017-07-14:
> in. If you navigate to circuit [1] you'll see something like:
> thread [3].
> Links:

Hi,

 No worries.  I don't actually use your code, but may check it out 
 further in the future as it looks pretty complete.

 I have one cobbled together myself that I built a yr ago that does what 
 I needed, I had just never implemented setting color on the lights.  
 When I searched to see if someone already had the RS-485 pieces I 
 needed, I came across your project and saw it was on your list of needs 
 on one of the pages.  I couldn't find it anywhere else either, so put my 
 script back into debug mode and just selected them all one-bye-on on the 
 panel till I had each logged.  Figured I would send them your way as 
 well to save someone else the trouble.  Turns out it looks like you had 
 them, it just wasn't pushed out yet.

 You can see my hack-job below  :)

 http://pool.netmagi.com/cgi-bin/index

 From the LAN, the charts are all clickable, and I can view 365 days of 
 history, but over WAN, they're just static images updated every 60 secs.

 -Rich H.

tagyoureit, 2017-10-05:
Thanks for sharing the link.  Way cool... Closing this for now.

closed 2017-10-05 by tagyoureit