tagyoureit/nodejs-poolController #52: Delay indicator for Cleaner

rflemming, 2017-07-07:
When the cleaner is turned on there is a 5 minute delay where the system pump kicks up to the cleaner RPMs and runs before turning on the cleaner booster pump.  On the intellitouch remote during this time the cleaner indicator light is blinking, once the 5 minutes have passed the light goes solid.  I need to double check, but I seem to recall the status packets showing the cleaner as off during the 5 minute delay.  I need to see if some other bit is set during that period so that the delay can properly be accounted for.

tagyoureit, 2017-07-07:
Yup, I was thinking the same thing.  There is also a 'delay cancel' that should be findable.  Just need some time/help to investigate this.

tagyoureit, 2017-07-11:
Ok... I just pushed up some code that you can test out.  I had to make some big assumptions on the logic, namely:
* I only have one circuit on delay, so it was hard to test if this will work for more than one circuit at once (if that's even possible), or on different circuits
* Not sure if this will work with other delays like heater cool down or valves

[image]

That is what it should look like.  A circuit in delay mode will be yellow.  If you click on it, it will cancel the delay and immediately start the circuit.

Additionally, you can control this via
 * socket = socket.emit('cancelDelay')
 * url = `http://<server>:3000/cancelDelay`

rflemming, 2017-07-11:
I'll check it when I get a chance.  One interesting thing I noticed this weekend was that if the cleaner and the solar were running at the same time and the solar shut off, the cleaner would go back in delay mode for 5 minutes.  I'm assuming that's because there is a change in pump RPMs for the two scenarios, but it may even do it when they are the same, I'm not sure.  It was just odd to see it stop running for 5 minutes after it had already been started.

The experience on the controller when pressing the cleaner button again while in delay is that it is turned off and the delay is canceled.  I assume that turning the circuit off while in delay via the API would produce the same result.

tagyoureit, 2017-07-11:
I found these two little bits in the Intellitouch manual:

> [In Delay Cancel section]: Automatic Pool Cleaner-SOLAR Delay: Allows Cleaner Pump to run even though solar delay has shut it off for 5 minutes.

> [In Assign Circuit Functions]: The cleaner pump would automatically shut off whenever the spa and solar is switched on with the cleaner pump (after a delay of five (5) minutes) if energy from the solar is required.

So maybe expected behavior?

> The experience on the controller when pressing the cleaner button again while in delay is that it is turned off and the delay is canceled. I assume that turning the circuit off while in delay via the API would produce the same result.

I haven't seen this... I know I've pressed the "On" button numerous times through my app (with no immediate effect), but haven't tried to turn it on twice from the panel.  I typically just press go to the Delay Cancel menu item.  I'll have to try this.

rflemming, 2017-08-15:
Sorry I've been really slow on following up here.  I'm watching this morning when the cleaner automatically turns on and I'm not seeing the delay reflected in the UI or in the API.  Once the 5 minutes were up both reflected the correct state for the cleaner.

The log messages seem to indicate that you noticed the change.

14:05:31.140 INFO CHECKING DELAY!  32
14:06:28.298 INFO CHECKING DELAY!  36
14:06:34.652 INFO Msg# 1295347   Pump 1 watts changed >5%: 0 --> 1121 
.
.
.
14:11:24.507 INFO CHECKING DELAY!  36
14:11:26.532 INFO CHECKING DELAY!  32

tagyoureit, 2017-08-18:
Hmmm, this was working.  Ironically, I tried to turn on my cleaner just now and it turned on without any delay so I need to see if I accidentally changed a setting because I can't test it without that.

Can you try to turn on your cleaner (or look when there is a delay in process) and navigate to the circuit status page `http://server:3000/circuit`?  When you see the 32/26 you should see `delay:1` in the corresponding circuits.  EG...

```
"4": {
"number": 4,
"numberStr": "circuit4",
"name": "CLEANER",
"circuitFunction": "Master Cleaner",
"status": 0,
"freeze": 0,
"macro": 0,
"delay": 1,
"light": {
"group": -1,
"colorStr": "off",
"color": -1
},
"friendlyName": "CLEANER"
},
```

If it is showing the delay here, then it is a problem on the HTML/Javascript side.  If not, it's something on the server side.

closed 2020-09-17 by tagyoureit