tagyoureit/nodejs-poolController #53: Add a URL registration system for integrations

rflemming, 2017-07-07:
I wrote an integration for smartthings to stream updates from the socket.io interface back to the smartthings hub.  The problem is that as is I need to hardcode the hub IP address into the integration's configuration.  It would be helpful if an integration could register a URL handler into the express server with an associated callback.  In my case I could then have my smartthings device handler or smartapp hit that URL and the integration code could then obtain the hub's IP that way.  In general though it might be nice if an integration could add its own URLs for receiving information back from somewhere else without spinning up its own web server.

tagyoureit, 2017-07-07:
Have you looked at https://github.com/tagyoureit/nodejs-poolController/wiki/Integrations-in-2.0 at all?  By calling `var server = container.server.getServer()` you would have access to the Express app.  Most integrations I've see call either the sockets or API's to update.  Is there something you want to do that can't be accomplished in this way?

rflemming, 2017-07-08:
I'm completely nodejs ignorant, so part of the problem may be lack of familiarity with how things work. 
 container.server.getServer() isn't going to get me want since what I really want to do in my integration is something like this (from server.js):

app.get('/my-special-thing', function(req, res) {
  myIntegration.magicalStuff()
  res.send('Presto')
}

Trying to access container.express() didn't seem to work inside my integration.  The URL I added couldn't be found.  If I added a getApp() method to server.js and used container.server.getApp() then I was able to do the above, but I have no idea if that's the right thing to do or not.

In short yes I am using the socket.io interface and the existing APIs.  The problem is that I'd also like to add an integration specific API handler and I'm looking for a way to do that.  I could spin up my own server in the integration, but that seems a bit excessive.  Does that make more sense?

tagyoureit, 2017-07-08:
> integration specific API handler

I just want to make sure you actually need something specific and we don't already have a way of doing it.  Can you describe exactly what you are trying to do?  What is the magic stuff?  I think I understand, but also realize I'm making a lot of assumptions in my head.

It sounds like you want this app to be able to tell smartthings what it's ip address is...?

rflemming, 2017-07-08:
Actually the inverse.  I need my integration to figure out the IP of the smartthings hub.  If I can pull the req.ip into my integration somehow then I can get that without needing to hardcode it into the integration's config.  Hence my idea for a custom URL endpoint that I could handle inside my integration. 
 Rather than have the smartthings side poll the API, it's possible to stream updates via the socket.io interface back to the hub.  Polling via smartthings seems to rarely work as well as one would hope and streaming updates make the device state more accurately reflect things in real time.

An alternative idea I explored was writing an SSDP integration that provided UPNP thermostat and switch support, which would be pretty neat if it worked.  I already have the other method working though via a hardcoded IP and it's solving my immediate needs, though it's far from optimal.

johnny2678, 2017-07-08:
Terrible coder here... But as a Smartthings user and a poolController user, I'd be happy to test any integration you come up with.

tagyoureit, 2017-07-08:
Ok, so I did a quick google search and found a few things: https://community.smartthings.com/t/any-way-to-directly-connect-to-the-hub/67140/9.  This seems to indicate they don't have much local control.  A few other threads suggested reserving an IP address on your router so the smart things always has the same IP.  

If you don't know the IP address of smart things, how are you going to find it out via the code in the integrations?  I think that's still the part I'm missing.  If you know the method of finding out the smart things IP, then we wouldn't need an endpoint in express, right???

rflemming, 2017-07-08:
My smartthings device could call http://pool:3000/integrations/smartthings/register (or the like) during initialization which would tie back to a handler inside my integration, which would pull out the req.ip and save that.  Then all future updates could be streamed back to the hub via that IP.  The streaming to the hub bit I have working, I just need to hardcode the hub's IP into the integrations config and I'd like to skip that part if possible.

I have a related request for the data returned via the socket.io interface, but I'll save that for another bug, or two.

tagyoureit, 2017-07-09:
I just pushed up a bit of code for you to test... looking for feedback.  Check out https://github.com/tagyoureit/nodejs-poolController/blob/4.x-DEV/src/integrations/customExpressRoutes.js.  If you call `http://<server>:3000/api/myruntimeroute` it will return the values in the custom route.  

This is based on [this](https://alexanderzeitler.com/articles/expressjs-dynamic-runtime-routing/) approach.

bsileo, 2017-10-22:
@rflemming I am working on a similar project. Would you be willing to share your code? Git or offline is fine. I am happy to share back any additions I make.

rflemming, 2017-10-22:
@bsileo 

I don't have what I did anywhere public, since what I have has a whole lot of hardcoded stuff based upon my setup.  My first goal was to get something functional then refactor and well I never got around to that second part :)  Here's what I have with the associated disclaimers.

I wanted to try and make a device handler that could operate on its own without requiring a smartapp.  That may or may not be the right choice.  I wanted the pool controller to stream updates to smartthings (rather then smartthings needing to poll) so that the state would be better reflected in realtime.

nodejs-poolController Integration:
https://pastebin.com/Y2xNLtRK

This doesn't work in 4.X and I never got around to figuring out why.  The circuit messages aren't sent and/or parsed properly while on 3.X they work fine.  In your config.json you'd need to add a config entry that defines where your smartthings hub is located:

"smartthings": {
  "address": "xxx.xxx.xxx.xxx",
  "port": "39500"
}

This is the thing I wanted to handle via the URL registration mentioned in this bug since having to hard code the location of the hub is pretty lame.

I subscribe to temp and circuit events separately because when doing all events the resulting json doesn't have a top level key to say what it is, so parsing is tricky since you don't know what you're getting.  I think the pool controller should probably always output with that key, ie

{circuits: ...}
{temperatures: ...}

regardless of whether or not you are subscribing to all events or specific events.  This could be corrected in the integration if @tagyoureit didn't want to make that change the default.  For now I danced around this about by adding the event type as a header.

Smartthings Device Handler:
https://pastebin.com/j136zhnX

This is a mess, but at least for me right now it shows the 3 temps I care about and I can turn things on and off.  As a short cut I just hard coded the circuit numbers and such.  It'll at least give you an example of how to parse the events streamed to the hub.  I wanted to re-write it to use child devices so that each circuit could be controlled independently, but I never got around to it.  The idea would be that you could toggle on/off certain components when configuring (ie pool heater, spa heater, circuits, etc) and child devices would be created for each thing you enabled.  I didn't want a device handler that tried to do everything since I don't have a spa and I don't want to see circuit switches for circuits where nothing is configured.  The downside to the child device thing is that when you include "all" child devices the resulting device looks like crap from a UI point of view.

I'm happy to help out, but I haven't had the time to really work on this as much as I wanted to.  Once I got something minimally functional my motivation ran out.  Let me know how it goes or if you have any questions.  The smartthings docs can be seriously lacking at times, so hopefully I've at least provided a starting point for you.  Good luck

bsileo, 2017-10-22:
Thanks for the great write-up...this is right aligned with the project I have been working on. I will integrate into your code and see if I can generalize it as I adapt it to my environment. I also tend to have the problem of running out of time once I get it working in my one-off environment! I'll let you know if I make progress and share the code back.

tagyoureit, 2017-10-23:
Guys - happy to help out and make this happen.  I thought I gave you something that could be workable with the registrations via the custom express route.  Let me know what else we could add on the pool app side to make this easier.

bsileo, 2017-11-23:
@rflemming Just a quick update I am making good progress on this. I need to update to the latest poolController release and adjust parsing for that, and still have a bit of work on the UI and configuration. I have migrated your code over to use composite device son Smartthings which lets all the data and controls be connected to SmartApps, routines, etc. on the platform. I'll share the code for Smartthings once I get it wrapped up.
![screenshot_20171123-161214](

https://user-images.githubusercontent.com/6116016/33188651-7a84687a-d069-11e7-85e3-1206cdd99f43.png

tagyoureit, 2017-11-23:
Nice job. Are you just doing read only? Or are you going to have 2-way control of the pool equipment from smart things? 
Tag

rflemming, 2017-11-25:
@bsileo looks good. I'm glad you found something useful in what I wrote, obviously you've taken it a lot further than I did! I'd be happy to help you test things out when you're ready. With composite devices, it would be great to be able to toggle on/off each of those devices. I'd prefer not to have switches for nonexistent circuits or spa settings since I don't have a spa. Unfortunately that'll probably make the device UI ugly since you lose precise control over the layout when you do that. Keep up the good work.

tagyoureit, 2017-11-25:
I've been trying to work the JSON data such that you can tell what is installed and what isn't... there may still some things missing, but you should be able to tell pool, spa, chlorinator at least.  Solar/heat/which controller are a little harder.

donkarnag, 2017-11-26:
@bsileo - this a great effort.  I would love to jump in and help in any way I can.  I may be able to help to add/test intellichem parts, if needed.  I look forward to seeing the next iteration!

rflemming, 2017-11-28:
@tagyoureit If we can tell the controller what hardware we have then the smartthings app doesn't need that logic since it can just display whatever is presented to it.  Being able to disable the basic major components is probably a good start.  Being able to disable certain circuits would also be nice.

tagyoureit, 2017-11-29:
Here are some ways to do it... but I can also create a separate entry just for the setup/configuration if needed.

* Pumps: First, you can look for the number of keys `pump[1]..., pump[2]...`.  Also, if the `pump[x].type`  is `None` then there is no pump installed there.  
* Chlorinator: `chlorinator.installed` will be 1 (installed) or 0 (not installed)
* Circuits: the name of each circuit, eg `circuit[x].name` will be "Not Used" if that circuit isn't assigned.  "Aux" is another type of circuit that may or may not be used depending on the setup.

Those are the big ones that we can get now.  In the bootstrap UI we do some basic parsing for # of pumps, if the chlorinator is installed, and if circuits are not used (basically what's above)...

tagyoureit, 2017-12-19:
Closing this.  If the custom routes aren't working, we can reopen or readdress.

closed 2017-12-19 by tagyoureit