tagyoureit/nodejs-poolController #54: Addition - Intellichem messages

donkarnag, 2017-10-17:
I am just starting to get this running, but can offer some messages from my Intellichem to the effort.  This is a two-pump intellichem so both pH/acid and ORP/chlorine are sent.  There are two very similar messages - first the intellichem to the controller, and then the controller broadcast.  The message itself is the same except for an extra (apparently) empty first bit.  I haven't figured out the all modes yet (dosing, mixing; chlorine, acid) but will keep studying.  Excuse my messy notes - help me understand where to best put this and I will try to clean it up:

//Status 0x12 (18) - Intellichem Status (length 41)
example:
 1  2  3  4  5  6  7  8  9  10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 
02 E3 02 AF 02 EE 02 BC 00 00 00 02 00 00 00 2A 00 04 00 5C 06 05 18 01 90 00 00 00 96 14 00 51 00 00 65 20 3C 01 00 00 00 

1-4 pH(1-2) / ORP(3-4) reading
02 E3 - pH  2*256 + e3(227) = 739 (=7.39)
02 AF - ORP 2*256 + af(175) = 687

5-6 pH setpoint
D0 = 7.2 (hi/lo bits - 720 = 7.2pH)
DA = 7.3
E4 = 7.4
EE = 7.5
F8 = 7.6

7-8 ORP setpoint
02 BC = 700 (hi/lo bits)

21-22 Tank levels; 21 is acid? 22 is chlorine?
06 and 05 

25-26 CH setpoint
90 is CH (90 = 400; 8b = 395) hi/lo bits

28
00 is CYA (00 = 0; 9 = 9; c9 = 201) (does not appear to have hi/lo - 201 is max

29-30
96 is TA (96 = 150)

34 - Water Flow Alarm (00 is ok; 01 is no flow)
00 flow is OK
01 flow is Alarm on (Water STOPPED)

36 Mode
25 dosing (auto)?
45 dosing acid (manually?)
55 mixing??? 
65 monitoring
02 (12 when mixing) and 04 (27 when mixing) related???

37
20 Nothing
22 Dosing Chlorine(?)

//Broadcast 0x93 (147) - Intellichem Status (length 42)
Note that this is the same as the above with the addition of a 0x00 bit to start.  It comes immediately after the Intellichem (ID 144 in my system) sends the 0x12 message above.
example:
 1  2  3  4  5  6  7  8  9  10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42
00 02 E3 02 AF 02 EE 02 BC 00 00 00 02 00 00 00 2A 00 04 00 5C 06 05 18 01 90 00 00 00 96 14 00 51 00 00 65 20 3C 01 00 00 00

donkarnag, 2017-10-17:
I should add the the request for the 18 message is a 210 message to the intellichem:
165,0,144,16,210,1,210,2,234

tagyoureit, 2017-10-23:
I've got a bit of this done.

CH= chlorine?
TA= total available solids?  total available chlorine?

How is the CH calculated?  You have `90 is CH (90 = 400; 8b = 395) hi/lo bits` but not sure what the math is here.

Also, the 147 (0x93) is the SET packet indicator (controller telling the intellichem what to do), the 210 is get, and 18 is the status as you say. 

I prefer to work in Decimal.  So I've converted everything to decimal on my end.  I should get something pushed up in a day or so that you can test.

donkarnag, 2017-10-24:
Thanks!

CH is Calcium Hardness; TA is Total Alkalinity.  These are settings within the UI (you measure and input) and are used for the Saturation Index (SI = pH + TF + CF + AF - 12.1).  

Page 25 of this link explains more (and see some of the most relevant text below: https://www.pentairpool.com/~/media/websites/pool/downloads/automation/manuals/intellichem_water_chemistry_controller_installation_and_users_guide_english.pdf

The example I gave (90 is CH was missing a bit, sorry).  The actual reading is two bits, i.e 01 90, which are the high/low - 0x01*256 + 0x90 = 1*256 + 144 = 400.  And, to be redundant, this is entered in the ScreenLogic interface / control panel.

I am struggling a little bit with the idea that 147 (0x93) is a SET packet.  The message goes from 16 to 15, so appears as a broadcast, rather than a SET-type command.  I will keep digging to see if I can wrap my head around it.

From Manual for SI:
pH: Enter the current pH value.
Temperature: Enter the current temperature setting.
CH: Enter the current calcium hardness level.
SLT: Enter the current salt level or Total Dissolved Solids (TDS) level if no SCG is used. TA: Enter the current total alkalinity value.
CYA: Enter the current cyanuric acid value.
SI: Displays the last calculated SI value (Corrosive/Scaling/Ideal). Note: If there is no  ow or Flow Delay is enabled, no SI value displayed.

tagyoureit, 2017-10-24:
I just pushed up some code (although I didn't see your post until after I did it).  But it should be a decent first pass and we can adjust as needed.

This [file](https://github.com/tagyoureit/nodejs-poolController/blob/4.x-DEV/src/lib/comms/inbound/controller/18.js) has the logic in it.

Check out the [broadcast](https://github.com/tagyoureit/nodejs-poolController/wiki/Broadcast) messages, and especially the first paragraph below it.  It describes the set/get/status bits.  (Also, feel free to include the headers as my logic always starts with 165 as the 0 bit in the packet.)

tagyoureit, 2017-11-02:
Ok, made another pass (along with a test case to see how far I actually got).  The code is pushed up.  Here is what I'm seeing (my output) based on the packet you gave me:
```
16:00:00.000 INFO Intellichem Saturation Index:
	SI = pH + CHF + AF + TF - TDSF
	-0.31 = 7.39 + 2.2 + 2.2 + 0 - 12.1
json for intellichem:  {
  "readings": {
    "PH": 7.39,
    "ORP": 687,
    "CYA": 0,
    "TOTALALKALINITY": 150,
    "waterFlow": 0,
    "SI": -0.31
  },
  "setpoint": {
    "PH": 7.5,
    "ORP": 700,
    "CALCIUMHARDNESS": 400
  },
  "tankLevels": {
    "1": 6,
    "2": 5
  },
  "mode": {
    "1": 0,
    "2": 101
  }
}
```

What is interesting is that when I click the "Intellichem Present" in the Intellitouch setup, I get multiple packets.
```
19:58:28.674 VERBOSE Msg# 1276   Controller packet is known to be a Get   IntelliChem packet: 165,0,144,16,210,1,210,2,234 
19:58:30.379 VERBOSE Msg# 1278   Controller packet is known to be a Intelli(?)   packet: 165,33,16,34,19,1,0,1,12
19:58:30.556   VERBOSE Msg# 1279     Controller packet is known to be a Set Intelli(?) packet:   165,33,15,16,147,42,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,128,0,0,0,0,0,0,0,2,34

```

This has a 210, 19 packet, and 147 packet (pattern somewhat repeats).  

Here is what I have so far:
```
        18: 'IntelliChem',  //status
        19: 'Intelli(?)',  //status
        146: 'Set IntelliChem',
        147: 'Set Intelli(?)',
        210: 'Get IntelliChem',
        211: 'Get Intelli(?)',
```

I see why you are confused... somehow there are two sets of packets that are being passed around for the Intellichem.  My working theory is that one set (maybe 19/147/211) is specifically for the IntelliPH, or possibly the [commercial](http://pentaircommercial.com/products/chemical-controllers-intellichem-commercial-water-chemistry-controller-488.htm) version of Intellichem?  Point is, I think that Intellitouch is...
1. Sends a "Get" Intellichem packet (210)
1. Receives an Intellichem status packet (19 -- which is weird because I don't have intellichem)
1. Sets the other Intelli??? (PH, commercial?) system with a nearly identical packet.

Sort of like a master/slave system.

tagyoureit, 2018-11-16:
Closing due to lack of activity.  If anyone wants to help develop this further let me know.

closed 2018-11-16 by tagyoureit