tagyoureit/nodejs-poolController #55: First Time Help

donkarnag, 2017-10-17:
If anyone can spare a moment, I could use some first-timer help.  I have successfully installed the package and can run it using "npm start", however this generates a running log after startup (the logs are as expected, and after reading many of these messages in another program I created, I know they are legit).  

However, the web interface at {ip}:3000/index.html doesn't respond, either with the running log or without.  This is what I was really after, and after as much self-help as I could find, I cannot seem to understand where to start.  I have done all of the simple fixes - restarting httpd and computer, uninstalling/reinstalling, etc.

Any help you can offer would be appreciated.

johnny2678, 2017-10-17:
Maybe /bootstrap/index.html ?

tagyoureit, 2017-10-19:
Hi and welcome.
you should be able to navigate to `http://ip:3000` or `http://ip:3000/index.html` and it should redirect you to `http://ip:3000/bootstrap/index.html`.  

Did you enable HTTPS or the port 3000 by chance?  They are settings in the `config.json`.  

I'll take a look at your Intellichem packets.  I can try to code that and we can work on that together.

donkarnag, 2017-10-20:
Thanks to both.  

After install (and a few fresh reinstalls), I can't find anywhere that uses :3000 to direct to this folder.  I was expecting an addition in httpd.conf, or a module .conf, or a virtual, but couldn't find any of those.  In addition, after I created a virtual host for :3000, it didn't work either.  What I did do was create my own module .conf that redirected IP/pool to the install www folder and that started things.  It required a few aliases to pull files from node_modules for socket.io and jquery.  As of yet, the alias I created for the bootstrap css and js files haven't worked, but I was able to get the page up and running by copying the two files over to the documentroot.  Although that may be part of the problem (below), I am looking at other things first.

Now, however, all of the css and js files load, but nothing happens in the page.  The counter counts up appropriately, and other js-enabled features work, but no pool data shows up, including the debug panel.  I can run a command line 'npm start' and that works as expected, but nothing from the web page still.  

the npm.debug.log shows an error that seems to occur the first time the app is run - the config.json file always gets an extra '}' at the end and I have to manually remove it.  After the first time it is fine.

Any thoughts?  I am so close I can just see the data waiting to populate - but it won't.

tagyoureit, 2017-10-22:
You shouldn't need to do any of that.  It has worked for everyone else right out of the "box".  This [section](https://github.com/tagyoureit/nodejs-poolController/blob/master/config.json#L93) in  `config.json` defines the HTTP variables.

You do need to start the pool app before navigating to the page.  `npm start` will start the web server, the pool app, the socket.io connection, etc etc.  

Did you do anything different from the steps [here](https://github.com/tagyoureit/nodejs-poolcontroller#installation-instructions)?

donkarnag, 2017-10-24:
I figured it out!!!  This was a tough one - I had to spend some quality time with node.js and build some exampled scripts to figure this out.  Here is the solution that worked for me.

First, I was trying to access this from a different computer on a local network.  It is a headless server which made this a little harder.  I finally figured out that port access was only allowed locally (using wget, etc.).  I assumed that there is something that was calling `server.listen(3000);`, but this needed to be opened to allow LAN access, like `server.listen(3000, '1.1.2.0');`.  The config.json file allowed `"expressPort": "3000, '1.1.2.0'",` and this took care of it!

I can now see status and messages, and was able to turn lights on and off.

tagyoureit, 2017-10-24:
First off, good sleuthing and glad you got it working.  I access the app from all kinds of devices (phone, tablet, other computers, etc) but don't need to specify an IP address to listen to.  According to [this](https://serverfault.com/questions/271824/node-js-is-not-accessible-from-external-ips-on-ubuntu) you can specify to bind to all IPs with `0.0.0.0`, but I think that's what the default is, anyway.  I'm wondering if there still wasn't anything else (firewall??, routing tables?) interfering with the external communication.

tagyoureit, 2017-11-27:
Closing for now.

closed 2017-11-27 by tagyoureit