tagyoureit/nodejs-poolController #57: Include event keys in result data structure

rflemming, 2017-10-23:
Adding @bsileo  since he might be interested:

The data returned via socketio events could benefit on being standardized.  Regardless of whether you are asking for "all", "circuit", "temp", etc the resulting data structure should contain the appropriate event keys.  ie

"all" would produce: "{ 'circuits': ..., 'temperatures': ...}"
"circuit" would produce: "{ 'circuits': ... }"
"temp" would produce: "{ 'temp': ...}"

As-is the event key is left out when subscribing to the more specific event type.  The issue is, that means the parser needs to know what type of message it's getting in each case.  If you're just sending the data somewhere else you might need to pass along that metadata via an alternate channel.  If the data structure produced had the event key names in place the same parser can be used for every event without additional info or trying to guess.

Make sense?

bsileo, 2017-10-24:
+1 for this approach as I tend to favor being more descriptive with data given the low cost of transmission.  I was just working on @rflemming 's code to handle this earlier today in the colloration we have going with a Smarthings integration.

BTW - got things almost up and running today - working on some enhancements to the code on the Smarthings side. Still pondering if there is a better way to architect the LAN handshake between the two versus a hardcoded IP address/port.

johnny2678, 2017-10-24:
I'm happy to help you test the Smartthings integration if you like...

tagyoureit, 2017-10-24:
I hadn't thought about that.  I was planning that either the `all` socket would be called (more descriptive) or the individual sockets would be called, in which case you would know what they are because you are explicitly listening to them.  

We could certainly include this in the next major (5.0) release, but not sure it is the right thing to do to make this change for a 4.1 release.  

In the meantime, couldn't you just use:
```
socket.on('circuit', function data(){
  data = {'circuit': data}
  send_to_my_parser(data)
}
```

Let's keep this in mind for the next major release.

rflemming, 2017-10-24:
I noted in the other bug where I posted my test code that modifying the data structure like you suggested in the integration itself was also an option.  I haven't touched this in so long that I forgot why I actually did things....  You mentioned the "all" event.  If I recall correctly at the time I tried this it didn't work the way you describe.  I'm fuzzy on the details, but looking back at the code I think it may just emit the individual event data structures rather than a combined data structure with the relevant event keys.  I'd need to go back and test, but reading lib/comms/socketio-helper.js I'm pretty sure that's what happens.

tagyoureit, 2017-10-24:
[image]

Here's a screenshot of `all`.  Let me know where we need to go from here.

rflemming, 2017-10-24:
Hrm, I don't remember.  Also I'm still using 3.1.13 because 4.x didn't work and I never figured out why, so that could be it.  I just tried adding:

    socket.on('all', function(data) { <do stuff> })

to my smartthings integration and didn't see any data being sent, so something is up.  I may try to chase this down again, but I don't want to waste my time if @bsileo is going a different direction.

bsileo, 2017-10-26:
So far I concur with @rflemming findings - I do not get anything through a 'all' integration. I did confirm that the direct API call for /all works and returns the data as expected just that nothing comes from the socket interface in the integrations. However, this may be a case of me not understanding the integrations - what on the system would actually trigger the "All" integration to fire @tagyoureit ?

tagyoureit, 2017-10-26:
Ohhhhh... the way I programmed `all` in sockets was to send all of the commands individually.  I did this because if you load the web page, you won't want to wait for any individual updates.  I thought we were only talking about the API calls so I didn't look at the sockets... now I get it.

You should be able to use this code:
```
    if (outputType === 'all') {
        io.sockets.emit('all', container.helpers.allEquipmentInOneJSON())
    }
```
Put it in the [emit to clients](https://github.com/tagyoureit/nodejs-poolController/blob/4.x-DEV/src/lib/comms/socketio-helper.js#L27) code block.  It will keep the existing functionality, but all now output a socket called `all` which is what you are looking for.  Thanks for catching this.  I'll push it with the next set of changes in 4.x-DEV.  (This will also work in the 4.0 and 3.x latest branches.)

tagyoureit, 2017-11-18:
I just pushed up a new 4.x-DEV build that has this change in it.  It's a BIG change and any code that deals with these API's/sockets should be checked...
1. all structures renamed to the singular form (pumps->pump, etc)
1. all (individual) json data now returned with the qualifier for what type it is.  `{pump: ...}`, `{circuit:...}`

Please test it out and let me know if you have any feedback.

tagyoureit, 2017-12-12:
As this is in the 4.x-DEV code line, I'm closing this.

closed 2017-12-12 by tagyoureit