tagyoureit/nodejs-poolController #59: Integration help needed - Simple http POST

donkarnag, 2017-11-19:
All - I have had some trouble getting an integration to work, and could use some (hopefully quick) advice if someone has some time.  The goal is a simple web post to a web page where I can integrate the data with other home data, and do so in a language I am far more familiar with... 

The integration currently looks like the below.  I borrowed most of this code from a smartthings integration (can't find it right now, thought happy to update with an appropriate attribution), and a number of examples from node.js sites.   

The problem I am having is that I cannot seem to post data into the body.  The data is posted back to this script using a simple php echo "RETURN THIS"; var_dump($_POST);, and this is appropriately sent back, but the $_POST var is empty:
```
16:53:52.704 INFO xxxxxx.com/upload_anypoint.php
16:53:52.925 INFO Status: 200
16:53:52.929 INFO Headers: {"date":"Sun, 19 Nov 2017 00:53:52 GMT","server":"Apache","x-powered-by":"PHP/5.4.45","vary":"Accept-Encoding,User-Agent","connection":"close","transfer-encoding":"chunked","content-type":"text/html"}
16:53:52.932 INFO Body: RETURN THIS: array(0) {
}
```
I would expect that this has to do with `req.write('{"data":"foobar"}'); req.end();` but that seems to be the correct way to do this.

I can post GET variables to pass correctly, but not POST.  If anyone can help debug this, I would greatly appreciate it!

```
module.exports = function(container) {

    var http = require('http');
    var request = require("request");

    var configFile = container.settings.getConfig()

    var address = configFile.webget.address
    var path = configFile.webget.path
    var protocol = configFile.poolController.web.expressTransport

    var uploadTimeCircuit = new Date().getTime();
    var uploadTimeTemp = new Date().getTime();

    var serverURL;
    var secureTransport;

    if (protocol === 'http') {
        serverURL = 'http://localhost:3000'
        secureTransport = false
    } else {
        serverURL = 'https://localhost:3000'
        secureTransport = true
    }

    var io = container.socketClient
    var socket = io.connect(serverURL, {
        secure: secureTransport,
        reconnect: true,
        rejectUnauthorized: false
    });

    function notify(event, data) {
      container.logger.info(address+path);
      //var url = 'http://'+address+path;

      var options = {
        hostname: address,
        path: path,
        method: 'POST',
        headers: {
            'Content-Type': 'application/json',
        }
      };

      var req = http.request(options, function(res) {
        container.logger.info('Status: ' + res.statusCode);
        container.logger.info('Headers: ' + JSON.stringify(res.headers));
        res.setEncoding('utf8');
        res.on('data', function (body) {
          container.logger.info('Body: ' + body);
        });
      });
      req.on('error', function(e) {
        container.logger.info('problem with request: ' + e.message);
      });
      req.write('{"data":"foobar"}');
      req.end();
    }

    socket.on('circuit', function(data) {
      var newTime = new Date().getTime();
      if ((newTime - uploadTimeCircuit) > 60000) {
        container.logger.info((newTime - uploadTimeCircuit)+'millis... Uploading');
        uploadTimeCircuit = newTime;
        notify('circuit', data);
      }
    })

    socket.on('temp', function(data) {
      //notify('temp', data)
    })

    function init() {
        container.logger.verbose('WebGet Loaded.')
    }

    return {
        init: init
    }
}
```

donkarnag, 2017-11-19:
I solved.  First, the `data` element is passed as an object, so needs JSON.stringify to pass in the form.  Then, for some reason, the only place it shows up on the other end is in $GLOBAL['HTTP_RAW_POST_VARS'].

tagyoureit, 2017-11-19:
That's great! I know there are slight differences in the different URL libraries, but flag you figured it out.

closed 2017-11-19 by tagyoureit