tagyoureit/nodejs-poolController #60: Cannot Hide Intellichem Panel

arrmo, 2017-11-21:
Whey I try to hide it, I get (in the shell where npm / nodeJS is running),
16:16:42.012 WARN Error updating bootstrap configClient.json:

Other panels can be hidden, but not this one?

FYI, after a refresh of the browser, the panel is back (not hidden).

tagyoureit, 2017-11-22:
I added the Intellichem [section](https://github.com/tagyoureit/nodejs-poolController/blob/4.x-DEV/src/www/bootstrap/configClient.json#L30) to `configClient.json`.  Is that section of the code in your file?  I have been moving fast with pushing updates...

arrmo, 2017-11-22:
NP - fixed this on my side as well. Very odd, as I merged in your latest changes, but this one didn't merge in. Not sure why, but no biggie - it works. Thanks!

BTW, does reordering the panels work for you? It seems to be broken now - will debug it here, submit PR if I can get it fixed.

tagyoureit, 2017-11-22:
Yes, I can reorder the panels.  However, the panel order isn't saved anywhere.  So it works for the current browser window, but once I reload, it goes back to the default order.

Also, there isn't a way to click/drag on mobile browsers (that I know of)... so this isn't available there.  No big deal, but I did notice it.  We can start a different thread for these issues.

closed 2017-11-22 by tagyoureit

arrmo, 2017-11-22:
Sounds good. It used to save position, so let's open a new ticket for that - I'll dig into it here.

arrmo, 2017-11-27:
OK, I think I found the issue. The code is there still, but the anchor panel is breaking it. Do we really want to anchor the Release Details? Or if we do - should it be outside the draggablePanelList?

Thanks!

tagyoureit, 2017-11-27:
I'll open a separate issue... good call on moving the release notes outside the draggable panels, but I'm still not able to save/persist the order next time I reload the page.