tagyoureit/nodejs-poolController #62: Panel re-order broken

tagyoureit, 2017-11-27:
I can re-order the panels within each browser window.  But when reloading them the next time, they are back to the default order.

arrmo, 2017-11-28:
NP, working on it. Moving files to assets seems to have broken things here though - re-trenching again.

arrmo, 2017-11-29:
OK, got it working here, but a couple questions first,
1) Did you move the systeminformation anchor outside the draggablePanelList? I thought you had, but don't see it in the code I pulled. It's key to this.
2) I seem to have to add ../ in front of all my links to stylesheets and code. Do you see this also?

Thanks!

tagyoureit, 2017-11-29:
1.  No, it's right below it in my (current) code.
2.  I haven't seen this either.  I was messing around with a couple of them, but it didn't appear to make a difference.  Obviously, it does and I missed it.

Can you attach a working version of the file here and I'll try it tomorrow?  I have a bunch of other off-line edits that I'm working on so merging the code might be tricky.

Thanks!

arrmo, 2017-11-30:
For 1), try swapping the order (i.e. as below) - this way the anchor is outside the draggable list.
<a name="systeminformation" class="anchor"></a>
<div id="draggablePanelList">

I think this is all you need to get the order restore working again - please let me know.

On 2), let's revisit after your updates.

Thanks!

PS: I'm not sure if Hide is what we want for the Release Details ... rather, once read, then put them away until new info comes - right? Separate ticket for this?

tagyoureit, 2017-11-30:
See attached for my order.  systeminformation is inside the draggablePanelList... 

[image]

(new thread for release notes created...)

arrmo, 2017-12-01:
Yep, that's what I was trying to say above - change the order, like this,
```
<a name="systeminformation" class="anchor"></a>
<div id="draggablePanelList">
```

Then, does panel restore work for you?

tagyoureit, 2017-12-01:
Ah!  Yes... good to go on this one.  Thanks!

closed 2017-12-01 by tagyoureit

arrmo, 2017-12-01:
Excellent! So you'll include this correction, right? Just want to make sure one of us does ... :-).