tagyoureit/nodejs-poolController #64: Smarthings / Pentair Device Handler

bsileo, 2017-11-30:
I have finished Alpha development on a UI to access the poolController from the SmartThings platform. This UI allows the user to see all the relevant details of the Pool in the ST platform and also exposes individual devices to allow integration of the pool equipment into the overall home automation system. Temperatures, pump controls, setpoints, chlorinator, and lighting / features can be controlled with the Smartthings environment, connected to separate SmartApps, etc. 

Thanks to @rflemming for the initial code and inspiration and @tagyoureit for help along the way.

This is only tested so far on my pool which is currently closed for the winter, so thats pretty limiting! 
@donkarnag have a go if you want to try it out!

Code to support this and install docs are at [(https://github.com/bsileo/SmartThings_Pentair)]

Feedback on the UI, code, docs, etc. all welcome! @tagyoureit can decide if we track via this issue, move to Gitter, etc. 

[image]

johnny2678, 2017-11-30:
yessssss!  Will be loading this shortly

johnny2678, 2017-11-30:
ok, not sure where you want to track issues.  This could totally be on my end, but after installing all Handlers and adding the PC device, Smartthings logging reports:
```
cc5dff39-606a-4111-ac14-22196e0bcd31  2:40:36 PM: error groovy.lang.MissingPropertyException: No such property: auxButton for class: script_dth_28eb9210_61f5_4646_9be9_9b485aa528e5_ver_0_1
cc5dff39-606a-4111-ac14-22196e0bcd31  2:40:36 PM: debug Getting ready to create Aux switch Pentair Pool Controller (Aux 1) Named=circuit1
cc5dff39-606a-4111-ac14-22196e0bcd31  2:40:36 PM: debug Created SpaHeat
cc5dff39-606a-4111-ac14-22196e0bcd31  2:40:36 PM: debug Created PoolHeat
cc5dff39-606a-4111-ac14-22196e0bcd31  2:40:35 PM: error groovy.lang.MissingPropertyException: No such property: auxButton for class: script_dth_28eb9210_61f5_4646_9be9_9b485aa528e5_ver_0_1
cc5dff39-606a-4111-ac14-22196e0bcd31  2:40:35 PM: debug Getting ready to create Aux switch Pentair Pool Controller (Aux 1) Named=circuit1
cc5dff39-606a-4111-ac14-22196e0bcd31  2:40:35 PM: debug Created SpaHeat
cc5dff39-606a-4111-ac14-22196e0bcd31  2:40:35 PM: debug Created PoolHeat
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:29 PM: debug Executing 'sendEthernet' http://192.168.5.31:3000/all
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:29 PM: info Requested a refresh
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:27 PM: debug Executing 'sendEthernet' http://192.168.5.31:3000/all
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:27 PM: info Requested a refresh
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:07 PM: error groovy.lang.MissingPropertyException: No such property: auxButton for class: script_dth_28eb9210_61f5_4646_9be9_9b485aa528e5_ver_0_1
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:07 PM: debug Getting ready to create Aux switch Pentair Pool Controller (Aux 1) Named=circuit1
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:07 PM: debug Created SpaHeat
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:07 PM: debug Created PoolHeat
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:07 PM: error groovy.lang.MissingPropertyException: No such property: auxButton for class: script_dth_28eb9210_61f5_4646_9be9_9b485aa528e5_ver_0_1
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:07 PM: debug Getting ready to create Aux switch Pentair Pool Controller (Aux 1) Named=circuit1
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:07 PM: debug Created SpaHeat
cc5dff39-606a-4111-ac14-22196e0bcd31  2:39:07 PM: debug Created PoolHeat
```
Any ideas?

donkarnag, 2017-11-30:
Same error messages here, and the app is not updating - it seems to try to get a response from the pool controller - I can't tell if it does.  
* the 'sendEthernet' link works from a browser and returns the full 'all' response
* Also, every time I update the settings to try different things, both in the IDE and the app, the parent device creates 3 new Child devices - see johnny's output above.

donkarnag, 2017-11-30:
I am probably getting ahead of myself, but I also get an error when trying to turn the light on - it seems to think that the circuit is 'null'.  I might guess that this will be corrected when it updates with the initial information:
```
2fdfdefc-f265-43ed-996c-fba4fc52edbd  12:02:13 PM: debug Executing 'sendEthernet' http://10.0.1.14:3000/circuit/null/set/1
2fdfdefc-f265-43ed-996c-fba4fc52edbd  12:02:13 PM: debug Executing 'set(null, 1)'
```

bsileo, 2017-11-30:
Thanks let me take a look and see if we can get this to stable. Might have to rebuild my environment to try this out. testing with Smartthings is a bit of a challenge!

bsileo, 2017-11-30:
Ok this is fixed in the Repo now. You just need to refresh the [Pool Controller](https://github.com/bsileo/SmartThings_Pentair/blob/master/devicetypes/bsileo/pentair-pool-controller.src/pentair-pool-controller.groovy) code. I refactored a bit of code since I had reinstalled form scratch in my environment so missed this minor typo.

bsileo, 2017-11-30:
As far as bugs and other issues I think it will be cleaner to open them on the [repository](https://github.com/bsileo/SmartThings_Pentair) to separate them out as specific to the Smarthings UI.

tagyoureit, 2017-12-01:
Nice work!  I'll close this issue for now, and add this integration/repo to the Readme.  👍

closed 2017-12-01 by tagyoureit