tagyoureit/nodejs-poolController #66: UnhandledPromis

donkarnag, 2017-12-04:
@tagyoureit I just updated all of the code today and am getting a new error - see below.  I am not running Promise, which seems to be the root of this.  Is there something new that you added?  And is there a way around it?
```
Error in settings: TypeError: Cannot read property 'enable' of undefined
    at GenericProvider.module.exports [as $get] (/usr/share/node/pool/4.x/nodejs-poolController/src/lib/logger/winston-helper.js:50:35)
    at Object.getService [as logger] (/usr/share/node/pool/4.x/nodejs-poolController/node_modules/bottlejs/dist/bottle.js:368:89)
    at /usr/share/node/pool/4.x/nodejs-poolController/src/lib/helpers/config-editor.js:53:26
    at tryCatcher (/usr/share/node/pool/4.x/nodejs-poolController/node_modules/bluebird/js/release/util.js:16:23)
    at Promise._settlePromiseFromHandler (/usr/share/node/pool/4.x/nodejs-poolController/node_modules/bluebird/js/release/promise.js:512:31)
    at Promise._settlePromise (/usr/share/node/pool/4.x/nodejs-poolController/node_modules/bluebird/js/release/promise.js:569:18)
    at Promise._settlePromise0 (/usr/share/node/pool/4.x/nodejs-poolController/node_modules/bluebird/js/release/promise.js:614:10)
    at Promise._settlePromises (/usr/share/node/pool/4.x/nodejs-poolController/node_modules/bluebird/js/release/promise.js:689:18)
    at Async._drainQueue (/usr/share/node/pool/4.x/nodejs-poolController/node_modules/bluebird/js/release/async.js:133:16)
    at Async._drainQueues (/usr/share/node/pool/4.x/nodejs-poolController/node_modules/bluebird/js/release/async.js:143:10)
    at Immediate.Async.drainQueues (/usr/share/node/pool/4.x/nodejs-poolController/node_modules/bluebird/js/release/async.js:17:14)
    at runCallback (timers.js:672:20)
    at tryOnImmediate (timers.js:645:5)
    at processImmediate [as _immediateCallback] (timers.js:617:5)
(node:9406) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: Cannot read property 'enable' of undefined
Exited nodejs-poolController cleanly
```

tagyoureit, 2017-12-04:
I may have pushed up a bad build... changing a whole lot of things currently (and I use promises internally).  Stay tuned!

donkarnag, 2017-12-04:
Thanks.

Thanks,
Kevin

tagyoureit, 2017-12-06:
Try again now.   When you start up the app, the console should indicate `Version: 4.1.1`.  I haven't bumped the version in a while, but this is now the latest.

donkarnag, 2017-12-06:
This still shows, but does not fail because of this.  Works as needed - ok to close from my perspective.

closed 2017-12-06 by donkarnag

tagyoureit, 2017-12-06:
I think you are missing a part of your `config.json` file.  See if [fileLog](https://github.com/tagyoureit/nodejs-poolController/blob/master/config.json#L117) exists.

donkarnag, 2017-12-06:
I do/did have that code, but now even that error doesn't show up.  Looks like all is good.