tagyoureit/nodejs-poolController #67: Yet Another Smartthings UI announcement

dhop90, 2017-12-15:
Not to be out done by [bsileo](https://github.com/bsileo/SmartThings_Pentair), I created my own smartthings app.

[image]

check it out [here](https://github.com/dhop90/pentair-pool-controller/blob/master/README.md)

tagyoureit, 2017-12-15:
Very cool!  I will really need to get a SmartThings app now.  (BTW, just check your spelling... "What is the Smartthing Penatair Pool Controller").

arrmo, 2017-12-28:
Just a thought, and this definitely @tagyoureit 's call, but ... I think we need to have a consistent and defined interface on the nodejs-poolController side (so it's the same for any and all ST apps). Agreed?

Is this the case here?

tagyoureit, 2017-12-30:
Well, from my end, it's only 1 interface.  :)  But I'm all for consistency.  @dhop90, how is your implementation different from @bsileo?  I have played with his, but have not yet (but would like to) check out yours.

tagyoureit, 2018-11-16:
Added to readme, so removing from the issues list.

closed 2018-11-16 by tagyoureit