tagyoureit/nodejs-poolController #71: Failure on Startup, Extended Pump Config

arrmo, 2017-12-20:
Hi,

I seem to be seeing this pretty consistently,
```
17:14:31.445 WARN Error writing controller packet 'Get Pump Config (Extended)' to serial bus.  Tried 5 times to write 255,0,255,165,1,16,33,219,1,1,1,180
17:14:33.825 ERROR Aborting controller packet Get Pump Config (Extended).  Tried 10 times to write 255,0,255,165,1,16,33,219,1,1,1,180
17:14:33.825 WARN Setting logging level to Debug.  Will revert to previous level in 2 minutes.
17:14:36.382 WARN Error writing controller packet 'Get Pump Config (Extended)' to serial bus.  Tried 5 times to write 255,0,255,165,1,16,33,219,1,2,1,181
17:14:38.763 ERROR Aborting controller packet Get Pump Config (Extended).  Tried 10 times to write 255,0,255,165,1,16,33,219,1,2,1,181
```

I think it makes sense, given my pumps, but perhaps have an option to disable this check?

tagyoureit, 2017-12-21:
How many pumps do you have?  And which controller (Intellitouch, right)?

I might need to see some full logs (debug level) for this one.  I do see this occasionally, but usually when Socat is acting up.  It's coming through on mine currently.

arrmo, 2017-12-22:
2 pumps - regular pool pump, and cleaner. And yes, Intellitouch (old ... :-)). I see this every time on startup, perhaps just because I don't have smart pumps of any sort?

Here is the debug log - does this help? Thanks!

```
/usr/bin/node --nolazy --debug-brk=3774 src/index.js 
Debugger listening on [::]:3774
c
Silly:  { '0': 'Loading settings with config file: ',
  '1': 'config.json' }
Debug:  { '0': 'Starting configEditor.init()' }
Debug:  { '0': 'Finished configEditor.init()' }
Silly:  { '0': 'Finished loading settings.' }
19:16:25.004 INFO Starting up express auth, https (port 3001)
19:16:25.055 INFO Starting up express auth, http (port 3000)
19:16:25.063 INFO initializing logger
19:16:25.219 DEBUG Initialized intellichem module
19:16:25.219 INFO Intro:  
*******************************
 poolController in brief (for full details, see README.md):
 Intellitouch: Configuration is read from your pool.  The application will send the commands to retrieve the custom names and circuit names.
 It will dynamically load as the information is parsed.  
 Intellicom: If you have an IntelliCom, set the Intellicom flag to 1 in the config file.
 Pump controller: default: poolController pump controller will start if intellicom and intellitoch = 0
                  always: poolController pump controller will always start
                  never: poolController pump controller will never start

 Writing: If there is a write error 5 times, there will be a warning message.
 If there is a write error 10 times, the logging will change to debug mode for 2 minutes and.
 it will abort the packet and go to the next one.

 To change the amount of output to the console, change the "logx" flags in lines 45-51 of this app.
 Visit http://_your_machine_name_:3000 for a web interface *******************************

19:16:25.220 INFO Settings:  
 Version: 4.1.2
 Config File: config.json
 
 //-------  EQUIPMENT SETUP -----------
 var intellicom = {
    "installed": 0,
    "friendlyName": ""
}
 var intellitouch = {
    "installed": 1,
    "friendlyName": ""
}
 var virtual = {
    "pumpController": "default",
    "chlorinatorController": "default"
}
 var controller.id = {
    "productName": "",
    "productNumber": "",
    "manufacturer": "",
    "description": ""
}
 var circuitFriendlyNames = {
    "1": "",
    "2": "",
    "3": "",
    "4": "",
    "5": "",
    "6": "",
    "7": "",
    "8": "",
    "9": "",
    "10": "",
    "11": "",
    "12": "",
[... 156 lines cut ...]
19:16:33.895 INFO Msg# 51  Schedules discovered:
ID: 1  CIRCUIT:(6)POOL  MODE:Schedule START_TIME:6:0 END_TIME:10:0 DAYS:Sunday Monday Tuesday Wednesday Thursday Friday Saturday 
ID: 2  CIRCUIT:(5)CLEANER  MODE:Schedule START_TIME:7:0 END_TIME:9:0 DAYS:Sunday Monday Tuesday Wednesday Thursday Friday Saturday 
ID: 3  CIRCUIT:(6)POOL  MODE:Schedule START_TIME:18:0 END_TIME:22:0 DAYS:Sunday Monday Tuesday Wednesday Thursday Friday Saturday 
ID: 4  CIRCUIT:(5)CLEANER  MODE:Schedule START_TIME:19:0 END_TIME:21:0 DAYS:Sunday Monday Tuesday Wednesday Thursday Friday Saturday 
ID: 5  CIRCUIT:(0)NOT USED 
ID: 6  CIRCUIT:(0)NOT USED 
ID: 7  CIRCUIT:(0)NOT USED 
ID: 8  CIRCUIT:(0)NOT USED 
ID: 9  CIRCUIT:(0)NOT USED 
ID:10  CIRCUIT:(0)NOT USED 
ID:11  CIRCUIT:(0)NOT USED 
ID:12  CIRCUIT:(0)NOT USED 

19:16:35.724 VERBOSE smartthings Sent all'{"circuit":{"1":{"number":1,"numberStr":"circuit1","name":"SPA","circuitFunction":"Spa","status":0,"freeze":1,"macro":0,"delay":0,"light":{"group":-1,"colorStr":"off","color":-1},"friendlyName":"SPA"},"2":{"number":2,"numberStr":"circuit2","name":"AIR BLOWER","circuitFunction":"Generic","status":0,"freeze":0,"macro":0,"delay":0,"light":{"group":-1,"colorStr":"off","color":-1},"friendlyName":"AIR BLOWER"},"3":{"number":3,"numberStr":"circuit3","name":"POOL LIGHT","circuitFunction":"Generic","status":0,"freeze":0,"macro":0,"delay":0,"light":{"group":-1,"colorStr":"off","color":-1},"friendlyName":"POOL LIGHT"},"4":{"number":4,"numberStr":"circuit4","name":"SPA LIGHT","circuitFunction":"Generic","status":0,"freeze":0,"macro":0,"delay":0,"light":{"group":-1,"colorStr":"off","color":-1},"friendlyName":"SPA LIGHT"},"5":{"number":5,"numberStr":"circuit5","name":"CLEANER","circuitFunction":"Generic","status":1,"freeze":0,"macro":0,"delay":0,"light":{"group":-1,"colorStr":"off","color":-1},"friendlyName":"CLEANER"},"6":{"number":6,"numberStr":"circuit6","name":"POOL","circuitFunction":"Pool","status":1,"freeze":1,"macro":0,"delay":0,"light":{"group":-1,"colorStr":"off","color":-1},"friendlyName":"POOL"},"7":{"number":7,"numberStr":"circuit7","name":"AUX 5","circuitFunction":"Generic","status":0,"freeze":0,"macro":0,"delay":0,"light":{"group":-1,"colorStr":"off","color":-1},"friendlyName":"AUX 5"},"8":{"number":8,"numberStr":"circuit8","name":"AUX 6","circuitFunction":"Generic","status":0,"freeze":0,"macro":0,"delay":0,"light":{"group":-1,"colorStr":"off","color":-1},"friendlyName":"AUX 6"},"9":{"number":9,"numberStr":"circuit9","name":"NOT USED","circuitFunction":"Generic","status":0,"freeze":0,"macro":0,"delay":0,"light":{"group":-1,"colorStr":"off","color":-1},"friendlyName":"NOT USED"},"10":{"number":10,"numberStr":"circuit10","name":"NOT USED","circuitFunction":"Generic","status":0,"freeze":0,"macro":0,"delay":0,"light":{"group":-1,"colorStr":"off","color":-1},"friendlyName":"NOT USED"},"11":{"number":11,"numberStr":"circuit11","name":"NOT USED","circuitFunction":"Generic","status":0,"freeze":0,"macro":0,"delay":0,"light":{"group":-1,"co [... cut ...]
19:16:41.343 WARN Error writing controller packet 'Get Pump Config (Extended)' to serial bus.  Tried 5 times to write 255,0,255,165,1,16,33,219,1,2,1,181
19:16:43.724 ERROR Aborting controller packet Get Pump Config (Extended).  Tried 10 times to write 255,0,255,165,1,16,33,219,1,2,1,181
19:16:56.129 VERBOSE smartthings Sent temp'{"temperature":{"poolTemp":54,"spaTemp":54,"airTemp":63,"solarTemp":0,"freeze":0,"poolSetPoint":82,"poolHeatMode":0,"poolHeatModeStr":"OFF","spaSetPoint":102,"spaManualHeatMode":"Off","spaHeatMode":1,"spaHeatModeStr":"Heater","heaterActive":0}}'
```

tagyoureit, 2017-12-22:
Can you change this `var fileLog = {"enable":0,"fileLogLevel":"silly","fileName":"output.log"}` in your config file to include `enable:1`, and also change ` var logMessageDecoding = 0` to  `var logMessageDecoding = 1`.  Then attach the file that will be called output.log.  It will be very big, but I need to see the extended log with the message decoding enabled.  This will also help me look at #9.

arrmo, 2017-12-22:
Sure, NP - attached! And to hopefully help, here are the steps I went through,
1) Start up, spa light on and off - works fine
2) Stop socat (remote), then try spa light on - fails (of course)
3) Restart socat (remote), again try spa light on - also fails
=> Then I stopped and restarted nodejs-poolController, and the spa light worked fine.

Hope this helps, yell if I can do anything else!

PS: Where is the version set? I noticed this isn't being updated, but that may be me.
[output.log](https://github.com/tagyoureit/nodejs-poolController/files/1582425/output.log)

tagyoureit, 2017-12-22:
> PS: Where is the version set? I noticed this isn't being updated, but that may be me.
output.log

It's set in `package.json`, but I've been lazy about updating it for each push.  I need a Grunt task or something to help me with that.

arrmo, 2017-12-22:
Makes sense - would be nice if it auto-updated somehow on a push to Github.

tagyoureit, 2017-12-22:
> I see this every time on startup, perhaps just because I don't have smart pumps of any sort?

I think you are probably right here.  So is it a single speed or dual speed pump?  Not a VS, VF or VSF?

I think I should probably use that "failure" to be more smart about how we treat the pumps and what features to turn on/off.

arrmo, 2017-12-22:
Just a dumb single speed pump ... :-).

tagyoureit, 2017-12-27:
Grab the latest push (4.1.33) and in `config.json` set your pump.1.type to "SS".  (SS = single speed; this can really be anything except VS/VF/VSF).  I just do a quick check to see if it is a valid type and if not then skip the extended pump config.  There is probably more to it than this (for controlling a single speed pump) but won't worry about it for now.

Also, if you have a pump.2 in config.json you can either delete it or also set type="SS".  The extended pump config is a per-pump setting.

arrmo, 2017-12-27:
That works - thanks! Closing, feel free to reopen if I'm doing this in error.

closed 2017-12-27 by arrmo