tagyoureit/nodejs-poolController #75: Config endpoint/socket

tagyoureit, 2017-12-27:
Make a basic configuration endpoint that will let consumers/services know what pool equipment is installed.

tagyoureit, 2018-01-17:
Pushed up with 4.1.34 as a beta test... please check it out and provide feedback.

tagyoureit, 2018-01-27:
@arrmo @bsileo What do you think about moving the following `configClient.json`
```
    "generalParams": {
        ...
        "hideAUX": false
    }
```
to the `config.json`?  If you aren't familiar with the former, the `configClient.json` controls the current Web UI and determines whether or not to show the Aux circuits.

However, now there is a need to be able to determine whether to show/hide these types of circuits in other Apps as well (like SmartThings).  

By moving this setting to the `config.json` we can then expose it through this new `\config` endpoint.  We'll be able to automatically hide the intellichem, solar, etc in the Web UI as an added benefit in addition to storing states in `configClient.json`.  

EG, user sets `intellichem:{installed:false}` in `config.json`.  When the Web UI loads up it can read this endpoint and automatically set the intellichem panel to never show.

bsileo, 2018-01-27:
This makes sense to me and would flow right into the SmartThings UI with the upcoming dynamic config interface off the /config endpoint.

tagyoureit, 2018-01-28:
I made a few changes (local) and my `/config` endpoint now looks like this:

```
{
"config": {
    "systemReady": 1,
    "intellicom": 0,
    "intellitouch": 1,
    "circuit": {
       "nonLightCircuit": {
           "1": {
            "circuitName": "SPA"
           },
          "2": {
               "circuitName": "JETS"
           },
           ...
         "20": {
              "circuitName": "AUX EXTRA"
           }
       },
        "lightCircuit": {
             "7": {
               "circuitName": "SPA LIGHT"
              },
             ....
            "9": {
                 "circuitName": "PATH LIGHTS" 
             }
        },
        "hideAux": false
     },
     "chlorinator": 1,
     "pumps": 2,
      "intellichem": 0,
      "spa": 1,
      "solar": 1
    }
}
```

I added the `hideAux` and nested them under the `circuit` key.  

1.  I can still get rid of the circuit names if they are not needed.
1.  Do we want to nest these objects closer to the actual `config.json`?  I just don't know what else we will need/want in the future (like pump type, or something else).  Should we nest all the sections under their respective equipment names?
1.  How should we keep the code in sync?  It definitely helps keeping the integrations code here, but I don't want to push up stuff and then have it break your code @bsileo.

Thoughts?

arrmo, 2018-01-28:
Hi,

This also makes sense to me, no issues here. As for sync ... if you add it to config.json first, the circuits just "go away", so the client will be happy, even with the duplicated definition - agreed? Then "after", we can remove it from the client end.

OK?

bsileo, 2018-01-28:
>I added the hideAux and nested them under the circuit key.

I am not clear on the intent here. Does this imply that all of the circuits should be hidden in a downstream UI?  One idea - maybe this option applies to individual circuits? For example my system comes back with 20 circuits but only 8 of them are installed in the panel and only 3 of those actually electrically connected. With this thinking I was wondering about this being a key that is set circuit by circuit to suppress some of them. 

> I can still get rid of the circuit names if they are not needed.

I would like to have them here as I am using them in ST config.

>Do we want to nest these objects closer to the actual config.json? I just don't know what else we will need/want in the future (like pump type, or something else). Should we nest all the sections under their respective equipment names?

In ST config processing, I am passing all the light details through to underlying processing because I want whatever details I can get form this. For example, I may be trying to figure out the type of lights in the near future to put an appropriate child light controller in place to manage them (color setting, etc).

>How should we keep the code in sync? It definitely helps keeping the integrations code here, but I don't want to push up stuff and then have it break your code @bsileo.

This is non trival! W could say its an immutable API but that is way more rigorous than we want for this level of dev I think!

bsileo, 2018-01-28:
@tagyoureit  Thoughts on including "circuitFunction" in the /config response? I am specifically interested in any info we have on the lights and see in there that we can results like this from All:

```2": {
   "number": 2,
   "numberStr": "circuit2",
   "name": "LIGHTS",
   "circuitFunction": "Intellibrite",
   "status": 0,
   "freeze": 0,
   "macro": 0,
   "delay": 0,
   "light": {
      "group": 1,
      "colorStr": "off",
      "color": -1
      },
   "friendlyName": "LIGHTS"
},
```
 which may help me with configuring the right type of light control functionality. I am looking at enabling color / mode controls for my Intellisync lights.  This line sup with an overall goal of building a Routine like "Party mode" for the back yard that controls pool and other lights etc.

tagyoureit, 2018-01-28:
> I added the hideAux and nested them under the circuit key.
   I am not clear on the intent here. Does this imply that all of the circuits should be hidden in a downstream UI?

Circuits are either "Not Used", they have a dedicated function, or there is a catch-all called "Aux".  Most of the time these Aux circuits are in support of other functions (like the valve for Solar or other functions).  Most people hide these, but some do want them exposed.  In general, we start with any "Aux" circuit hidden and this `hideAux` would apply to any Aux circuit.  The `Not Used` are always hidden.

> Thoughts on including "circuitFunction" in the /config response?

Lumping in most of the other comments here... maybe we flip and go the other way.  `/config` could just include summary information, BUT, we also include the `/config` as part of the `/all` endpoint?  That way we don't need to replicate anything.  In the end, you would just need to listen/subscribe to the `/all` and get everything you could possibly want to know at the summary and detail levels.

Another thought... when I built the Siri/Homebridge integration I split out the lights from the other circuits (like I did in the `/config`).  It was/is really helpful to have a light switch icon for the lights and a switch icon for the cleaner, spa, waterfall, etc.  It would be another device handler in ST, but pretty much duplicated except for the icons/labels.

bsileo, 2018-01-28:
I am on the same path for light controls with the new dynamic config - This will be a in new DeviceHandler which can then be extended with options to support color controls and InteliSync mode selection.

bsileo, 2018-01-28:
> Lumping in most of the other comments here... maybe we flip and go the other way. /config could just include summary information, BUT, we also include the /config as part of the /all endpoint? That [...]

Yes this is a reasonable approach. i have been torn on using /config in my SmartApp vs. just calling and parsing /all or some combination of the two.  Given the challenges in managing the UI in SmartThings, I really need to gather all the details in the SmartApp and have them in place before building out the collection of Devices to try to get the UI to be somewhat consistent and usable.

tagyoureit, 2018-06-02:
Closing this...

closed 2018-06-02 by tagyoureit