tagyoureit/nodejs-poolController #79: Proposed changes to `config.json` and how to apply custom changes.

tagyoureit, 2018-01-14:
Per some frustrations/challenges with an ever-updating `config.json`, I've been playing with an approach that adds a new `defaults.json` file.  The defaults will be compared to the current `config.json` (or custom file) and check for changes.  Questions...

1. Should the output be to `config.json` (or your currently used custom version), or would it be more preferable to have config.json treated more as _only_ the values you want to change, and write actual changes to a third file?
1. Should the changes be destructive (i.e. if I remove something from the `defaults.json` should it also remove it from the target file?  Or  just print out a warning and let you (the user) manually remove them?

Would like some input... but I'll forge ahead with my own thoughts and can alter if there is consensus on another direction.

arrmo, 2018-01-14:
Hi,

This is what I was mentioning a little while back - seems we're finding the same challenge. I'm thinking that the custom file is only the changes (i.e. merges) ... thoughts?

Thanks!

tagyoureit, 2018-01-14:
yup... I'm doing all of this for you.  ;-)

I was starting to lean towards only having two files.  My thought was that the defaults would be never be changed (I think that's a given).  

I think the challenge comes in where it isn't clear what will be updated by the app and what won't.  Maybe this just comes down to a documentation thing, but I want to add (for example) `"spa": {"installed": 0 }`.  You could specify this in a middleman file, but then I might override it in the final file (I could figure out from the circuit array if you have a spa or not).  But I have not yet figured out how to tell if you have an intellichem `"intellichem": {"installed": 0 }`.  Hopefully, the smarter the app gets the less you will need to change. 

The integrations, of course, will always be custom.  But even those could (in the future) be added via a UI.  The notifications, on the other hand, should never need to be touched by the end user.

Maybe the best option would be to only have this middleman include things that are user set (vs set by the app)... but I am still learning what can be discovered from the packets and what can't.

arrmo, 2018-01-14:
LMAO - too funny! I wasn't quite meaning that, rather that we're thinking the same thing here .. ;-).

I admit, you have me a bit confused on the middleman file ... :-(. Isn't the idea to load a safe set of default values, to make sure all items are configured (with at least a baseline), and then just allow customization on top of this? Sorry, I may be missing the point.

Thanks!

tagyoureit, 2018-01-14:
I went as far last night as trying to completely separate out the user changes from the system changes... 4 total files.  It was pretty impossible.  Some settings are controlled by the user, some by the app, and some by both depending on the specific scenario.

Now I'm back to the original 2 or 3 files.  

If we have 2 files:
File 1 = System defaults.  Never touched by the user.  Used to add new keys and warn about existing ones.
File 2 = Config file.  User would change defaults in here.  App would write to the file as well.  
This would be the closest to what we have now, but with the benefit of more flexibility and dynamic changes.

If we have 3 files:
File 1 = System defaults.  Never touched by the user.  Used to add new keys and warn about existing ones.
File 2 = User overrides.  A specific file that would be used to set values in the config file.  This would be the only place that a user would make changes.
File 3 = Config file.  This would combine the System defaults and the User overrides.  The app would be the only one writing to this file.
This is what I was referring to as a middleman file (File 2).

While the latter, 3 files, sounds good, I'm still running into trouble with the actual implementation details.  For example, If the app makes a change to the config file, do we over write that with the user overrides file next time the app starts?  Or, do we only look at this file once per version and then migrate the changes.  If we are only migrating the changes once, but the user wants to change them (logging level, for example), where should they change them?  

I think, for simplicity sake, of going with the 2 files.  It's pretty close to what we have now.  It should also allow for in-place upgrades of the file which is currently the biggest problem with upgrades.  

In essence...
 
> Isn't the idea to load a safe set of default values, to make sure all items are configured (with at least a baseline), and then just allow customization on top of this?

I think the 2 files solves for this, while the 3 files also would be introduces other unneeded complexity.

arrmo, 2018-01-15:
Yep, agreed about 2 vs. 3 files (complexity) ... but why is the 2nd file not real-only also? Meaning they are both input, read on startup. No real need to write out a file, is there?

I may be missing the point of the write yet ... :-(.

tagyoureit, 2018-01-15:
The config file is written to quite often.  It caches info about the pool and app between restarts.

bsileo, 2018-01-15:
I like the idea of a separate user-managed config file for overrides.  It is  like a scaled down version of the real config file where you only list out things you are changing from the "off the shelf" config. If I understand the challenge right, it is around the problem of the user defining a setting and the system detecting one, so then who wins int he final merged runtime system configuration. 

One thought here is to let the user choose. If I set a value in my config file I could mark it as an "hard" config. For example, if I want to specify a name for some circuit in my config that overrides the system setting on the controller, that would always apply, so I might include that in a section for "hard" changes. 
I might also have "soft" changes that can be overridden. I do not know the section by section details well enough to give pertinent examples here, but the concept is to not try to guess the intent but let the configurer specify it.   

This could work in a future state where you have a UI or in-app changes to config happening as well. You need the UI to determine the context of the setting change (one time for this session only, permanent update tot he app config?) and then write the results into the correct place.

tagyoureit, 2018-01-17:
Thanks guys... good thoughts.  I'm moving forward (ok, done with) a 2-file implementation.  I think it should be pretty straightforward but you can give me some feedback when I post it.  (Hopefully tonight!).

@bsileo You have some good thoughts with hard/soft requirements, but it's just a bit too much work on my end.  I have been trying to come up with a good reason why a user would want to override something on the system permanently, but really can't.  (If something is coming from the Intellitouch controller, it could be changed there, for example.)

Anyway, I think you'll like what I have.  Take a look when I push up the release.  It has actually made my testing a bit easier in that I can just take snippets of the config file that I want changed, and automatically build an entire, complete config file from it.

tagyoureit, 2018-01-17:
Pushed up 4.1.34 tonight.  Please check it out.  See the README for new entries/notes.  Make sure to `npm update` as there are new modules.

tagyoureit, 2018-01-29:
check the build I just pushed up.  The `/config` is now included in the `/all`.  It is the equipment section of the `config.json` with the added `config.systemReady` parameter.  Some of the info might be duplicated (like pump info), but I think this might cover everything...

Here's a snapshot of what mine looks like.
```
{
  "config": {
    "systemReady": 1,
    "equipment": {
      "controller": {
        "intellicom": {
          "installed": 0,
          "friendlyName": ""
        },
        "intellitouch": {
          "installed": 1,
          "friendlyName": ""
        },
        "virtual": {},
        "id": {}
      },
      "chlorinator": {
        "installed": 1,
        "desiredOutput": {},
        "friendlyName": "",
        "id": {}
      },
      "pump": {},
      "intellichem": {
        "installed": 0
      },
      "spa": {
        "installed": 1
      },
      "solar": {
        "installed": 1
      },
      "circuit": {
        "friendlyName": {},
        "hideAux": false,
        "nonLightCircuit": {},
        "lightCircuit": {}
      }
    }
  },
  "circuit": {},
  "pump": {},
  "schedule": {},
  "temperature": {},
  "time": {},
  "UOM": {},
  "valve": {},
  "chlorinator": {},
  "intellichem": {}
}
```
(or the picture, with comments...)
[image]

bsileo, 2018-02-05:
@tagyoureit I am not seeing this additional /config content in /all.  It may be an environment issue with me syncing up with Git.  What file can I look in to verify I have this change and/or what commit is it from?

tagyoureit, 2018-02-05:
Ah, looks like I committed on my end, but never pushed up.  Just did that now.  This is where the [new code](https://github.com/tagyoureit/nodejs-poolController/blob/4.x-DEV/src/etc/settings.js#L467) lives, but you should be good with a fetch now.

bsileo, 2018-02-05:
thanks good now

tagyoureit, 2018-05-16:
Closing this as it was included in 5.0.  We can revisit again if we want to remove/make changes.

closed 2018-05-16 by tagyoureit