tagyoureit/nodejs-poolController #81: Error on app launch - Virtual Clorinator controller not started because it is set to default

Bboy486, 2018-01-27:
and another controller (intellitouch/intellicom) is present.

Does this mean the software cant run when this runs?

tagyoureit, 2018-01-27:
Hi,
Most people who have a pool controller let the Pentair controller control the chlorinator, so there isn't a need for this software to control it.  Conversely, if the pool controller AND this app try to be the primary controller for the chlorinator then there will be an issue because the chlorinator will be getting conflicting commands.

To change this behavior, and _only if_ your chlorinator is not directly controlled by you Pentair equipment, change the setting to `virtual:{chlorinatorController:"always"}` as outlined [here](https://github.com/tagyoureit/nodejs-poolcontroller#virtual).

If your chlorinator is directly hooked up to the Pentair controller than this message is expected.  You will still be able to change the setting and view the output, we just change the settings on the Pentair controller instead of sending the message directly to the chlorinator every 4 seconds.

hope that helps,
Tag

Bboy486, 2018-01-28:
Thank you, but I am not sure I follow. I am actually trying to use SmartThings https://community.smartthings.com/t/pentair-screenlogic/71870/10

I need this software to run that SmartApp/Device Handler.

tagyoureit, 2018-01-28:
You need to get this app up and running before you try to integrate it with SmartThings.  Do you have a chlorinator?

Bboy486, 2018-01-28:
No. I thought I just needed this to intercept the commands from the Pentair Screenlogic to Smartthings.

tagyoureit, 2018-01-28:
If you can start this app, and navigate to `http://{yourserver}:3000` and see/control your pool equipment then you are good on this side.  You can ignore any warnings like above.  Once you get to this point, you can install the SmartThings integration.

Bboy486, 2018-01-28:
I did install everything but the CMD keeps retrying to start the app. So i dont think I can finish the ST integration until the app is running. 

the 3000 should be the port number for my pentair control right?

tagyoureit, 2018-01-28:
> I did install everything but the CMD keeps retrying to start the app. 

I'm not sure what you mean by this.

> the 3000 should be the port number for my pentair control right?

3000 is the port where this app runs.  You should get a web UI loaded.

Maybe you can attach some screenshots or logs (please attach logs as a file, not copy/paste, if they are more than a couple dozen lines) so I can see what you are talking about.

Bboy486, 2018-01-28:
When I run npm start it has the error I mentioned earlier and keeps retrying to connect to the app.

Here is a screenshot of the command prompt and config file.
[image]

tagyoureit, 2018-01-28:
The app isn't restarting, it is just trying to open up the port over and over again.

Do you have an RS485 adapter that can be found at the address `/dev/ttyUSB0`?  I *strongly* suggest that you go through the Readme carefully.  It mentions the need for a physical RS485 adapter as the [first line in the install instructions](https://github.com/tagyoureit/nodejs-poolController#installation-instructions) and goes into further detail [here](https://github.com/tagyoureit/nodejs-poolController#rs485-adapter).

You need to make a physical connection between this RS485 adapter and your pool equipment.

Bboy486, 2018-01-29:
I will review when I am home.

Bboy486, 2018-04-01:
Yes i have n RS485 Adapter 

Logs below
0 info it worked if it ends with ok
1 verbose cli [ 'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'start' ]
2 info using npm@5.6.0
3 info using node@v8.9.4
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle nodejs-poolcontroller@4.0.0~prestart: nodejs-poolcontroller@4.0.0
6 info lifecycle nodejs-poolcontroller@4.0.0~start: nodejs-poolcontroller@4.0.0
7 verbose lifecycle nodejs-poolcontroller@4.0.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle nodejs-poolcontroller@4.0.0~start: PATH: C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin;C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0\node_modules\.bin;C:\Users\Bboy\AppData\Roaming\npm;C:\Program Files\nodejs\;C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Common Files\Acronis\SnapAPI\;C:\Program Files (x86)\Overlook Fing 2.2\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\OpenVPN\bin;C:\Program Files (x86)\FileBot\;C:\Program Files (x86)\AOMEI Backupper;C:\Program Files\nodejs\;C:\Users\Bboy\AppData\Local\Programs\Python\Python35\Scripts\;C:\Users\Bboy\AppData\Local\Programs\Python\Python35\;C:\Users\Bboy\AppData\Local\Microsoft\WindowsApps;C:\Users\Bboy\AppData\Roaming\npm
9 verbose lifecycle nodejs-poolcontroller@4.0.0~start: CWD: C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0
10 silly lifecycle nodejs-poolcontroller@4.0.0~start: Args: [ '/d /s /c', 'node src/index.js' ]
11 silly lifecycle nodejs-poolcontroller@4.0.0~start: Returned: code: 1  signal: null
12 info lifecycle nodejs-poolcontroller@4.0.0~start: Failed to exec start script
13 verbose stack Error: nodejs-poolcontroller@4.0.0 start: `node src/index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\index.js:285:16)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at EventEmitter.emit (events.js:214:7)
13 verbose stack     at ChildProcess.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:925:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid nodejs-poolcontroller@4.0.0
15 verbose cwd C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0
16 verbose Windows_NT 10.0.16299
17 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "start"
18 verbose node v8.9.4
19 verbose npm  v5.6.0
20 error code ELIFECYCLE
21 error errno 1
22 error nodejs-poolcontroller@4.0.0 start: `node src/index.js`
22 error Exit status 1
23 error Failed at the nodejs-poolcontroller@4.0.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

tagyoureit, 2018-04-01:
I haven't tested with node v8.9.  I would try with a [LTS 6](https://nodejs.org/en/blog/release/v6.12.2/) version.

Bboy486, 2018-04-02:
Do I need to uninstall to do what your suggesting?

tagyoureit, 2018-04-02:
You can do that.  Or use the [n](https://github.com/tj/n) package to have multiple versions of NodeJS available on your machine.  If you aren't sure, just uninstall and use the v6.

Bboy486, 2018-04-02:
I did that (thanks)

C:\Users\Bboy>cd "C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0"

C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0>npm start

> nodejs-poolcontroller@4.0.0 start C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0
> node src/index.js

Exited nodejs-poolController cleanly
C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0\src\etc\settings.js:86
        throw new Error(err)
        ^

Error: Error: Your configuration file is out of date.  Please update to the latest version.
    at checkForOldConfigFile (C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0\src\etc\settings.js:86:15)
    at Object.exports.load (C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0\src\etc\settings.js:103:5)
    at Object.exports.init (C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0\src\lib\app.js:158:31)
    at Object.<anonymous> (C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0\src\index.js:25:5)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:383:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:496:3

npm ERR! Windows_NT 10.0.16299
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "start"
npm ERR! node v6.12.2
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! nodejs-poolcontroller@4.0.0 start: `node src/index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the nodejs-poolcontroller@4.0.0 start script 'node src/index.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the nodejs-poolcontroller package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node src/index.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs nodejs-poolcontroller
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls nodejs-poolcontroller
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0\npm-debug.log

C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0>

tagyoureit, 2018-04-02:
[LMGTFY](http://lmgtfy.com/?q=Module+version+mismatch.+Expected+48%2C+got+57.).

It means your modules installed with Node v8 and you need to remove/reinstall them with the correct v6 versions.

Bboy486, 2018-04-07:
I had uninstalled node.js restarted and reinstalled. Should be all good there. I also reinstalled npm and your app (to be clean).

Your environment has been set up for using Node.js 8.11.1 (x64) and npm.

C:\>cd C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0

C:\Users\Bboy\Downloads\Pentair and Node\nodejs-poolController-4.0.0>npm start

> nodejs-poolcontroller@4.0.0 start C:\Users\Bboy\Downloads\Pentair and Node\nod
ejs-poolController-4.0.0
> node src/index.js

14:46:20.359 INFO initializing logger
14:46:20.383 INFO Intro:
*******************************
 poolController in brief (for full details, see README.md):
 Intellitouch: Configuration is read from your pool.  The application will send the commands to retrieve the custom names and circuit names.
 It will dynamically load as the information is parsed.
 Intellicom: If you have an IntelliCom, set the Intellicom flag to 1 in the config file.
 Pump controller: default: poolController pump controller will start if intellicom and intellitoch = 0
                  always: poolController pump controller will always start
                  never: poolController pump controller will never start

 Writing: If there is a write error 5 times, there will be a warning message.
 If there is a write error 10 times, the logging will change to debug mode for 2 minutes and.
 it will abort the packet and go to the next one.

 To change the amount of output to the console, change the "logx" flags in lines 45-51 of this app.
 Visit http://_your_machine_name_:3000 for a web interface *******************************

14:46:20.389 WARN Settings:
 Version: 4.0.0 alpha 7
 Config File: config.json

 //-------  EQUIPMENT SETUP -----------
 var intellicom = {
    "installed": 0,
    "friendlyName": ""
}
 var intellitouch = {
    "installed": 1,
    "friendlyName": ""
}
 var virtual = {
    "pumpController": "default",
    "chlorinatorController": "default"
}
 var controller.id = {
    "productName": "",
    "productNumber": "",
    "manufacturer": "",
    "description": ""
}
 var circuitFriendlyNames = {
    "1": "",
    "2": "",
    "3": "",
    "4": "",
    "5": "",
    "6": "",
    "7": "",
    "8": "",
    "9": "",
    "10": "",
    "11": "",
    "12": "",
    "13": "",
    "14": "",
    "15": "",
    "16": "",
    "17": "",
    "18": "",
    "19": "",
    "20": ""
}

 var chlorinator = {
    "installed": 1,
    "desiredOutput": -1,
    "friendlyName": "",
    "id": {
        "productName": "",
        "productNumber": "",
        "manufacturer": "",
        "description": ""
    }
}

 var pump = {
    "1": {
        "type": "VS",
        "externalProgram": {
            "1": -1,
            "2": -1,
            "3": -1,
            "4": -1
        },
        "friendlyName": "",
        "id": {
            "productName": "",
            "productNumber": "",
            "manufacturer": "",
            "description": ""
        }
    },
    "2": {
        "type": "None",
        "externalProgram": {
            "1": -1,
            "2": -1,
            "3": -1,
            "4": -1
        },
        "friendlyName": "",
        "id": {
            "productName": "",
            "productNumber": "",
            "manufacturer": "",
            "description": ""
        }
    }
}
 //-------  END EQUIPMENT SETUP -----------

 //-------  POOLCONTROLLER SETUP -----------
 var appAddress = 33
 //-------  WEB SETUP -----------
 var expressPort = 3000
 var expressTransport = http
 var expressAuth = 0
 var expressAuthFile = /users.htpasswd
 //-------  END MISC SETUP -----------

 //-------  NETWORK SETUP -----------
 // Setup for Network Connection (socat or nc)
 var netConnect = 0
 var rs485Port = /dev/ttyUSB0
 var netHost = raspberrypi
 var netPort = 9801
 //-------  END NETWORK SETUP -----------

 //-------  LOG SETUP -----------
 var logLevel = info
 var socketLogLevel = info
 var fileLog = {"enable":0,"fileLogLevel":"silly","fileName":"output.log"}
 var logPumpMessages = 1
 var logDuplicateMessages = 0
 var logConsoleNotDecoded = 0
 var logConfigMessages = 0
 var logMessageDecoding = 1
 var logChlorinator = 1
 var logPacketWrites = 1
 var logPumpTimers = 0
 var logApi = 0
 //-------  END LOG SETUP -----------

 //-------  DATABASE SETUP -----------
 var influxEnabled = 0
 var influxHost = localhost
 var influxPort = 8086
 var influxDB = pool
 //-------  END DATABASE SETUP -----------

14:46:20.395 INFO Virtual chlorinator controller not starting because it is set to default and another controller (Intellitouch/Intellicom) is present.
14:46:20.405 ERROR Error opening port: Opening /dev/ttyUSB0: Unknown error code 3.  Will retry in 10 seconds
14:46:20.931 INFO Your version (4.0.0) is the same as the latest published release.
14:46:30.428 ERROR Error opening port: Opening /dev/ttyUSB0: Unknown error code 3.  Will retry in 10 seconds
Shutting down open processes
nodejs-poolController services stopped successfully
Exited nodejs-poolController cleanly

tagyoureit, 2018-04-09:
Did you remove the node_modules directory under the app and then run `npm install` again?  Removing node and npm won't affect these files.

Bboy486, 2018-04-14:
Yes I did that and ran it again. [image]

Bboy486, 2018-04-23:
I am not sure how to do that. Above my understanding (and likely the issue here). I do have the adapter (it came with my system). I can hit the localhost:3000 website (but no information is shown).

I am not sure how to check  /dev/ttyUSB0. Sorry. I tried od -x < /dev/ttyUSB0 in command line but that doesnt do anything. 

Again, sorry I cant follow what you just asked.

tagyoureit, 2018-04-23:
> I do have the adapter (it came with my system).

What do you mean by this?  RS485 comes with the pool equipment (it's how they all talk).  But no RS485 adapters come with any computers.  

I wrote up a new [wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details) page that outlines how you can test this.  It has instructions for both Win/Mac and also the most popular FTDI/CH340x drivers.

Bboy486, 2018-04-28:
I see. It makes sense now. pentair called the RS485 an adapter. So I have a protocol and a wireless adapter.[image]

closed 2018-11-16 by tagyoureit