tagyoureit/nodejs-poolController #82: Difficulty with RS485 USB adapter

ljohnston0423, 2018-02-01:
I have installed the application, but I cannot seem to get the computer to recognize the RS485 adapter on my Mac or PC machine. 

The error I get is:

21:37:06.858 ERROR Error opening port: Error: No such file or directory, cannot open /dev/ttyUSB0.  Will retry in 10 seconds

I would really appreciate any help.

tagyoureit, 2018-02-02:
I have a Mac, so I'll provide insight on that platform but you should be able to follow along in Windows as well.
1. Does the RS485 Adapter show up in `/dev`?  In mine, it is missing here.
[image]

But after I plug it in, you can see it listed:
[image]

2. If it doesn't show up when you insert it, you probably need to figure out what chip your USB adapter uses.
Two of the bigger ones are:
CH340x: https://sparks.gogo.co.nz/ch340.html (This is my recommended chipset.)
FTDI: http://www.ftdichip.com/FTDrivers.htm 
Of course, install the drivers, reboot, yadda yadda...

3.  Once you see the driver listed in the `/dev` directory, you can follow these [directions](http://www.ftdichip.com/FTDrivers.htm) to make sure that you have the wires properly setup.  You'll need to run `od -x < /dev/ttyUSB0` that matches your driver name.  In this example, I would use `od -x < /dev/tty.usbserial-AIZ9PCX`.  (ttyUSB0 is the default name for the CH340 chipset, I think.)

4.  Change the line in your `config.json`
```
...
"network": {
            "rs485Port": "/dev/ttyUSB0",
```
to match the port you see above.

If you are successful with steps 1-3, it should work in the app with no issues.
Tag

arrmo, 2018-02-02:
If this is Windows ... check the Device Manager, COM Ports (to get the COM port number).

ljohnston0423, 2018-02-02:
Does the adapter need to be wired to the pentair protocol adapter before it
will show up in the /dev list?

arrmo, 2018-02-02:
I don't believe so - the USB side of things should take care of that. Do you see it in device manager?

ljohnston0423, 2018-02-02:
I don’t have device manager. I am using a Mac and have followed the above instructions, but even after installing the drivers it does not show up in the /dev

arrmo, 2018-02-02:
Sorry, I misunderstood - my apologies!

ljohnston0423, 2018-02-02:
It was the driver version!! Your instructions help a GREAT deal. Thank you

ljohnston0423, 2018-02-02:
I was able to get this all up and running and can even see the dashboard at
port 3000, but after a little while of running just fine the application
crashes and I get the below error

TypeError: Cannot read property 'name' of undefined
    at Object.setCircuitFromController
(/Users/*****/Downloads/nodejs-poolController/src/lib/equipment/circuit.js:296:39)
    at Object.process
(/Users/******/Downloads/nodejs-poolController/src/lib/comms/inbound/controller/11.js:32:27)
    at Object.processControllerPacket
(/Users/*****/Downloads/nodejs-poolController/src/lib/comms/inbound/process-controller.js:64:45)
    at decode
(/Users/*****/Downloads/nodejs-poolController/src/lib/comms/inbound/decode-helper.js:247:51)
    at Object.processChecksum
(/Users/*****/Downloads/nodejs-poolController/src/lib/comms/inbound/decode-helper.js:285:13)
    at Object.iterateOverArrayOfArrays
(/Users/*******/Downloads/nodejs-poolController/src/lib/comms/inbound/receive-buffer.js:114:44)
    at Object.push
(/Users/*******/Downloads/nodejs-poolController/src/lib/comms/inbound/packet-buffer.js:35:35)
    at SerialPort.<anonymous>
(/Users/*******/Downloads/nodejs-poolController/src/lib/comms/sp-helper.js:58:36)
    at emitOne (events.js:116:13)
    at SerialPort.emit (events.js:211:7)
    at SerialPort.raw
(/Users/*****/Downloads/nodejs-poolController/node_modules/serialport/lib/parsers.js:7:13)
    at SerialPort._emitData
(/Users/*****/Downloads/nodejs-poolController/node_modules/serialport/lib/serialport.js:313:18)
    at SerialPort.<anonymous>
(/Users/*****/Downloads/nodejs-poolController/node_modules/serialport/lib/serialport.js:293:14)
    at SerialPort.<anonymous>
(/Users/*******/Downloads/nodejs-poolController/node_modules/serialport/lib/serialport.js:306:7)
    at FSReqWrap.wrapper [as oncomplete] (fs.js:658:17)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! nodejs-poolcontroller@4.0.0 start: `node src/index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the nodejs-poolcontroller@4.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely
additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/*******/.npm/_logs/2018-02-02T20_05_46_877Z-debug.log

tagyoureit, 2018-02-02:
Hmmm, interesting.  I would suggest that you try the [4.1](https://github.com/tagyoureit/nodejs-poolController/tree/4.x-DEV) version.  It's not officially released as I'm actively making changes to it, but I'm so far down the road with it I don't want to start making changes with the 4.0 code again.  Most of the people that are helping me are on it running without issue.

ljohnston0423, 2018-02-02:
I downloaded and installed version 4.1, and it works great, but after a
little while I get this error:

Unhandled rejection TypeError: Cannot read property 'name' of undefined
    at Object.setCircuitFromController
(/Users/**/Downloads/nodejs-poolController-4/src/lib/equipment/circuit.js:342:43)
    at Object.process
(/Users/****/Downloads/nodejs-poolController-4/src/lib/comms/inbound/controller/11.js:32:27)
    at Object.processControllerPacket
(/Users/****/Downloads/nodejs-poolController-4/src/lib/comms/inbound/process-controller.js:64:45)
    at decode
(/Users/****/Downloads/nodejs-poolController-4/src/lib/comms/inbound/decode-helper.js:263:51)
    at Object.processChecksum
(/Users/*******/Downloads/nodejs-poolController-4/src/lib/comms/inbound/decode-helper.js:301:13)
    at Object.iterateOverArrayOfArrays
(/Users/*******/Downloads/nodejs-poolController-4/src/lib/comms/inbound/receive-buffer.js:117:44)
    at Object.push
(/Users/*******/Downloads/nodejs-poolController-4/src/lib/comms/inbound/packet-buffer.js:35:35)
    at SerialPort.<anonymous>
(/Users/*******/Downloads/nodejs-poolController-4/src/lib/comms/sp-helper.js:74:44)
    at emitNone (events.js:106:13)
    at SerialPort.emit (events.js:208:7)
    at emitReadable_ (_stream_readable.js:513:10)
    at emitReadable (_stream_readable.js:507:7)
    at addChunk (_stream_readable.js:274:7)
    at readableAddChunk (_stream_readable.js:250:11)
    at SerialPort.Readable.push (_stream_readable.js:208:10)
    at binding.read.then
(/Users/*******/Downloads/nodejs-poolController-4/node_modules/serialport/lib/serialport.js:378:10)

tagyoureit, 2018-02-04:
Ok, thanks for letting me know it still happens.  I'll re-open this for you.  Can you please turn on the [fileLog](https://github.com/tagyoureit/nodejs-poolController#notifications) option at the `silly` level and attach it here after the error occurs.  That should give me the information to be able to track this down.

ljohnston0423, 2018-02-09:
Here is the output log

ljohnston0423, 2018-02-12:
The best I can figure from my minimal knowledge of this topic is that
something is wrong with the "POOL" object. Anytime I attempt to make a
change to the pool through the bootstrap interface or through the
smartthings app, it "breaks" the app and gives me an error in terminal.
Once I stop and restart npm it works fine until I attempt to change
anything attached to pool again.

tagyoureit, 2018-02-13:
I've been traveling the last week and am not back yet. Will be able to look at this in a few days. Thanks for your patience.

ljohnston0423, 2018-02-13:
No worries.

tagyoureit, 2018-02-14:
Are you able to enable the logging for each of the specific types of circuit?  You’ll need to set logChlorinator=1, logConfig=1, etc etc (the section [log](https://github.com/tagyoureit/nodejs-poolController/tree/4.x-DEV#version) that is just below the version section).  It will make a very large file, but will give me the info I need.  Forgot to mention this earlier.  Also, do send the error message again if it is displayed on the screen and not the log file.

ljohnston0423, 2018-02-15:
The terminal error
[output.log](https://github.com/tagyoureit/nodejs-poolController/files/1728857/output.log)
 and the new log file attached. 

Unhandled rejection TypeError: Cannot read property 'name' of undefined
    at Object.setCircuitFromController (/Users/lukejohnston/Downloads/nodejs-poolController-4/src/lib/equipment/circuit.js:342:43)
    at Object.process (/Users/lukejohnston/Downloads/nodejs-poolController-4/src/lib/comms/inbound/controller/11.js:32:27)
    at Object.processControllerPacket (/Users/lukejohnston/Downloads/nodejs-poolController-4/src/lib/comms/inbound/process-controller.js:64:45)
    at decode (/Users/lukejohnston/Downloads/nodejs-poolController-4/src/lib/comms/inbound/decode-helper.js:263:51)
    at Object.processChecksum (/Users/lukejohnston/Downloads/nodejs-poolController-4/src/lib/comms/inbound/decode-helper.js:301:13)
    at Object.iterateOverArrayOfArrays (/Users/lukejohnston/Downloads/nodejs-poolController-4/src/lib/comms/inbound/receive-buffer.js:117:44)
    at Object.push (/Users/lukejohnston/Downloads/nodejs-poolController-4/src/lib/comms/inbound/packet-buffer.js:35:35)
    at SerialPort.<anonymous> (/Users/lukejohnston/Downloads/nodejs-poolController-4/src/lib/comms/sp-helper.js:74:44)
    at emitNone (events.js:106:13)
    at SerialPort.emit (events.js:208:7)
    at emitReadable_ (_stream_readable.js:513:10)
    at emitReadable (_stream_readable.js:507:7)
    at addChunk (_stream_readable.js:274:7)
    at readableAddChunk (_stream_readable.js:250:11)
    at SerialPort.Readable.push (_stream_readable.js:208:10)
    at binding.read.then (/Users/lukejohnston/Downloads/nodejs-poolController-4/node_modules/serialport/lib/serialport.js:378:10)

ljohnston0423, 2018-02-17:
> Ok, thanks. I see the problem. It's been something that I've had on the
> backlog for quite a while but you are only the 2nd person to ever run into
> this issue. I will give you a manual workaround and hopefully can make this
> dynamic at some point down the road.
>
> It looks like you have more than 20 circuits. One of the components in
> your system is requesting circuit 21 165,1,16,35,203,1,21,1,186 and
> getting a response 165,1,15,16,11,5,21,0,0,0,0,0,234. But the system is
> coded for only 20 circuits.
>
> Here is where you can change the two items that should make this work:
>
>    1. Update the numberOfCircuits
>    variable to match the number you have in your system. (You can have up to
>    50 total expansion ports, I believe.)
>    2. Expand the config.json section of circuit friendly names to match
>    the number you set in the step above this.
>
>         "circuit": {
>             "friendlyName": {
>                 "1": "",
>                 "2": "",
>                 "3": "",
>                 "4": "",
>                 "5": "",
>                 "6": "",
>                 "7": "",
>                 "8": "",
>                 "9": "",
>                 "10": "",
>                 "11": "",
>                 "12": "",
>                 "13": "",
>                 "14": "",
>                 "15": "",
>                 "16": "",
>                 "17": "",
>                 "18": "",
>                 "19": "",
>                 "20": "",  //do not forget the comma after the 20
>                 "21": "",
>                 keep going....
>                 "40": "" //etc... and no comma after the 40 (or last number)
>             },
> ```
>
> On the other hand, it actually looks like you are only using 4 circuits.  If you don't have an expansion board, this might be a configuration issue on your system (but the above "fix" should still work).
>
> Let me know if this works.

I will try this fix this weekend. I only have 4 circuits configured in my
Screenlogic 2/Intellitouch configuration with no expansion pack. Any clue
why it would think I have 20?

tagyoureit, 2018-02-18:
I'm not sure why your system would think it has more circuits than it actually does... great question.  I don't know what tells the controller if there are expansion ports.

ljohnston0423, 2018-02-19:
I have implemented the fix as described and it is still stopping after a
few hours. But I cannot find in the log where you are finding the reference
to circuit 21 to continue to troubleshoot.

tagyoureit, 2018-02-19:
The packets I mentioned above are for the [circuit names](https://github.com/tagyoureit/nodejs-poolController/wiki/Configuration#13-circuit-names):
1. 165,1,16,35,203,1,21,1,186  -- Request for Circuit Name
1. 165,1,15,16,11,5,21,0,0,0,0,0,234  --- Response for Circuit Name

The 165,1 is part of the preamble.  (The 1 may change depending on the address that the controller is using).  The next two numbers are the Destination and Source.  The request is being sent to the main controller (16) from device with ID 35.  I don't know what 35 is.  The response is going to the broadcast address (15) and coming from the main controller (16).  The next numbers (203 and 11) are the request and response for a specific circuit.  Following that are the length of the remaining packets (1 for the request; 5 for the response).  

The 21 in both packets mean that the controller is being asked, and responding to, a request for the 21st circuit.  The 0,0,0,0,0 in the response just means that nothing is assigned to this circuit.  The last two numbers in each packet are just the checksum.

The following are the addresses that I know about:
```
        2: 'Chlorinator',
        15: 'Broadcast',
        16: 'Main',
        32: 'Remote',
        33: 'PoolControllerApp', //default address
        34: 'Wireless',
        96: 'Pump 1',
        97: 'Pump 2',
        98: 'Pump 3',
        99: 'Pump 4',
        100: 'Pump 5',
        101: 'Pump 6',
        102: 'Pump 7',
        103: 'Pump 8',
        104: 'Pump 9',
        105: 'Pump 10',
        106: 'Pump 11',
        107: 'Pump 12',
        108: 'Pump 13',
        109: 'Pump 14',
        110: 'Pump 15',
        111: 'Pump 16',
        144: 'Intellichem',
```

If you can let me know what other controllers/devices you have on your pool maybe we can figure out what is at address 35.  I'm still missing quite a bit of setup information and it is very likely that the main controller is telling all devices about how many circuits are installed and it isn't the attached device blindly asking about a 21st circuit in hopes that it exists.

[image]

You should also be able to go into your Controller menu-->Setup-->Circuit Names and it should give you an idea of how many circuits the system thinks it has.  The `Circuit #: 1/18` means that I have 18 useable circuits (with 1 or 2 not useable, hence the 20 total).

tagyoureit, 2018-03-02:
I just pushed up some code changes with the [4.x-DEV](https://github.com/tagyoureit/nodejs-poolcontroller/tree/4.x-DEV) branch.  It will now output warnings if the equipment wants to request more circuits than the app knows about.

ljohnston0423, 2018-03-03:
It seems to be working perfectly now! It ran all night without crashing.
Great work!

tagyoureit, 2018-04-09:
Awesome... will close for now.

closed 2018-04-09 by tagyoureit