tagyoureit/nodejs-poolController #86: Connect via network

parnic, 2018-04-02:
I saw your "wish list" note about wanting to add support for communicating with a system over the network. I've created [node-screenlogic](https://github.com/parnic/node-screenlogic) for this if you'd like to take a look or use it. It currently only supports "read" queries (no way to set heat mode, light colors, etc. just yet) and only a subset of messages, but I plan on expanding it to add more stuff.

tagyoureit, 2018-04-02:
Fantastic!  I'm really interested in seeing what Configuration information we can get back for more dynamic support of setting up the the right configuration (number of expansion ports, presence of different components, etc).  Also anything about pump status.  And some more.  Am excited to take a look at what you have already.

abovesociety, 2018-04-07:
Can you guys create a Homebridge plug-in for pentair integration with home kit?

tagyoureit, 2018-04-09:
> Can you guys create a Homebridge plug-in for pentair integration with home kit?

This should really not belong to this issue... but having said that, and having done some homebridge programming before I can say this would be a great community investment, but I won't be putting hours into this.  You should check out the SmartThings integrations that we have, though.

christensonautohome, 2018-04-20:
isn't this a plug-in for homebridge?
https://github.com/leftyfl1p/homebridge-poolcontroller

tagyoureit, 2018-04-21:
Oh, yes. I forgot about that. I don’t remember if he made it compatible with 4.0 or the 4.x-DEV branch, though.

closed 2018-11-16 by tagyoureit

dennypage, 2020-10-18:
Was this closed because the network connection feature is no longer desirable? Or because the Homebridge issue?

tagyoureit, 2020-11-03:
This was closed because this app is primarily based on reading the serial port packets and decoding them.  Not everyone has a ScreenLogic.  It would be interesting at some point to read the screenlogic network traffic but it would likely be a heavy lift when you can buy a $7 RS485 adapter and plug it into your system.

While it might be slightly more convenient for those that want to read the network protocol it doesn't but us much in functionality and the effort probably isn't worth it.