tagyoureit/nodejs-poolController #87: Pump St

TaZZaT, 2018-04-11:
### Expected behavior and actual behavior
Should see Pump RPMs, drive state... in WebUI, mode.  they all say "***notset"

### Steps to reproduce the problem
Stop/Start Service
### Specifications

1. What version of the code are you using?
4.0.0
1. All output from the file log (See [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))

1. Any errors that are in the console
I'm getting the following errors on startup.

10:23:40.202 WARN Error writing controller packet 'Get Pump Config (Extended)' to serial bus. Tried 5 times to write 255,0,255,165,33,16,33,219,1,1,1,212
10:23:42.662 ERROR Aborting controller packet 'Get Pump Config (Extended)'. Tried 10 times to write 255,0,255,165,33,16,33,219,1,1,1,212
10:23:42.664 WARN Setting logging level to Debug. Will revert to previous level in 2 minutes.

1. Your config.json or in-use configuration file
{
    "equipment": {
        "controller": {
            "intellicom": {
                "installed": 0,
                "friendlyName": ""
            },
            "intellitouch": {
                "installed": 1,
                "friendlyName": ""
            },
            "virtual": {
                "pumpController": "default",
                "chlorinatorController": "default"
            },
            "id": {
                "productName": "",
                "productNumber": "",
                "manufacturer": "",
                "description": ""
            },
            "circuitFriendlyNames": {
                "1": "",
                "2": "",
                "3": "",
                "4": "",
                "5": "",
                "6": "",
                "7": "",
                "8": "",
                "9": "",
                "10": "",
                "11": "",
                "12": "",
                "13": "",
                "14": "",
                "15": "",
                "16": "",
                "17": "",
                "18": "",
                "19": "",
                "20": ""
            }
        },
        "chlorinator": {
            "installed": 1,
            "desiredOutput": -1,
            "friendlyName": "",
            "id": {
                "productName": "",
                "productNumber": "",
                "manufacturer": "",
                "description": ""
            }
        },
        "pump": {
            "1": {
                "type": "VSF",
                "externalProgram": {
                    "1": -1,
                    "2": -1,
                    "3": -1,
                    "4": -1
                },
                "friendlyName": "",
                "id": {
                    "productName": "",
                    "productNumber": "",
                    "manufacturer": "",
                    "description": ""
                }
            },
            "2": {
                "type": "none",
                "externalProgram": {
                    "1": -1,
                    "2": -1,
                    "3": -1,
                    "4": -1
                },
                "friendlyName": "",
                "id": {
                    "productName": "",
                    "productNumber": "",
                    "manufacturer": "",
                    "description": ""
                }
            }
        }
    },
    "poolController": {
        "appAddress": 33,
        "web": {
            "expressPort": 3000,
            "expressTransport": "http",
            "expressAuth": 0,
            "expressAuthFile": "/users.htpasswd"
        },
        "network": {
            "rs485Port": "/dev/ttyUSB0",
            "netConnect": 0,
            "netHost": "raspberrypi",
            "netPort": 9801
        },
        "notifications": {
            "version": {
                "remote": {
                    "version": "4.0.0",
                    "tag_name": "v4.0.0",
                    "dismissUntilNextRemoteVersionBump": false
                }
            }
        },
        "log": {
            "logLevel": "info",
            "socketLogLevel": "info",
            "fileLog": {
                "enable": 0,
                "fileLogLevel": "silly",
                "fileName": "output.log"
            },
            "logPumpMessages": 1,
            "logDuplicateMessages": 0,
            "logConsoleNotDecoded": 0,
            "logConfigMessages": 1,
            "logMessageDecoding": 1,
            "logChlorinator": 1,
            "logPacketWrites": 0,
            "logPumpTimers": 0,
            "logApi": 0
        },
        "database": {
            "influx": {
                "enabled": 0,
                "host": "localhost",
                "port": 8086,
                "database": "pool"
            }
        }
    },
    "integrations": {
        "socketISY": 0,
        "outputSocketToConsoleExample": 0
    },
    "socketISY": {
        "username": "blank",
        "password": "blank",
        "ipaddr": "127.0.0.1",
        "port": 12345,
        "Variables": {
            "chlorinator": {
                "saltPPM": 16
            },
            "pump": {
                "1": {
                    "watts": 25,
                    "rpm": 24,
                    "currentprogram": 13,
                    "program1rpm": 10,
                    "program2rpm": 11,
                    "program3rpm": 12,
                    "program4rpm": 13,
                    "power": 14,
                    "timer": 15
                }
            }
        }
    },
    "outputSocketToConsoleExample": {
        "level": "warn"
    }
}

[output.log](https://github.com/tagyoureit/nodejs-poolController/files/1899952/output.log)

1. Your pool equipment
intellitouch, intelliclor, intelliflow (Pump 1), Booster pump (pump 2 - on/off only)

tagyoureit, 2018-04-12:
Hi, I would try with the [4.x-DEV](https://github.com/tagyoureit/nodejs-poolController/tree/4.x-DEV) branch.  I may have already fixed some issues in there.

The extended pump config messages you can also safely ignore.  I'll look into it, but I'm actually not doing anything with this information.

You may have a different problem going on with the communications.  Do you see other data from Intellitouch?  Can you check to make sure your [wires are hooked up correctly](https://github.com/tagyoureit/nodejs-poolController#rs485-adapter)?

TaZZaT, 2018-04-12:
Thank you for the response and thank you for all your efforts to make this code available... awesome work.  

I will install the DEV branch to see it that fixes the issue.  I do think my wires are connected correctly as I do see all other information from my intellitouch system - schedules, circuit names... etc and can operate all the the buttons from the webUI, Smartthings, and Alexa.  Given that, I do have an interesting issue with my intelliclor comms where I see all information except salt levels on the intellitouch panel and in poolController.  I've trouble shot this issue many different ways but honestly don't have enough brain power to actual decipher the control messages yet...:-).  

I'll let you know if the DEV branch works... Thanks again.