tagyoureit/nodejs-poolController #91: Uncaught exceptions may stop serialport from working

tagyoureit, 2018-05-22:
### Expected behavior and actual behavior
Errors will not prevent serial port from working.

### What is actually happening (screenshots are helpful)
```
08:16:59.714 DEBUG Msg# 47  Light group/position packet is: 7,16,0,0,8,0,0,0,9,48,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
_temp: {"group":2,"circuit":7}
Unhandled rejection TypeError: Cannot read property 'friendlyName' of undefined
    at Object.setControllerLightGroup (/home/pi/nodejs-poolController/src/lib/equipment/circuit.js:604:136)
    at Object.process (/home/pi/nodejs-poolController/src/lib/comms/inbound/controller/39.js:34:27)
    at Object.processControllerPacket (/home/pi/nodejs-poolController/src/lib/comms/inbound/process-controller.js:115:45)
    at decode (/home/pi/nodejs-poolController/src/lib/comms/inbound/decode-helper.js:263:51)
    at Object.processChecksum (/home/pi/nodejs-poolController/src/lib/comms/inbound/decode-helper.js:301:13)
    at Object.iterateOverArrayOfArrays (/home/pi/nodejs-poolController/src/lib/comms/inbound/receive-buffer.js:117:44)
    at Object.push (/home/pi/nodejs-poolController/src/lib/comms/inbound/packet-buffer.js:35:35)
    at SerialPort.<anonymous> (/home/pi/nodejs-poolController/src/lib/comms/sp-helper.js:74:44)
    at emitNone (events.js:86:13)
    at SerialPort.emit (events.js:185:7)
    at emitReadable_ (_stream_readable.js:432:10)
    at emitReadable (_stream_readable.js:426:7)
    at readableAddChunk (_stream_readable.js:187:13)
    at SerialPort.Readable.push (_stream_readable.js:134:10)
    at binding.read.then (/home/pi/nodejs-poolController/node_modules/serialport/lib/serialport.js:378:10)
    at tryCatcher (/home/pi/nodejs-poolController/node_modules/bluebird/js/release/util.js:16:23)

08:17:02.225 WARN Error writing controller packet 'Get Schedule' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,209,1,1,1,202
08:17:04.655 ERROR Aborting controller packet Get Schedule.  Tried 10 times to write 255,0,255,165,33,16,33,209,1,1,1,202
08:17:07.223 WARN Error writing controller packet 'Get Schedule' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,209,1,2,1,203
08:17:09.609 ERROR Aborting controller packet Get Schedule.  Tried 10 times to write 255,0,255,165,33,16,33,209,1,2,1,203
08:17:12.173 WARN Error writing controller packet 'Get Schedule' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,209,1,3,1,204
08:17:14.560 ERROR Aborting controller packet Get Schedule.  Tried 10 times to write 255,0,255,165,33,16,33,209,1,3,1,204
08:17:17.134 WARN Error writing controller packet 'Get Schedule' to serial bus.  Tried 5 times to write 255,0,255,165,33,16,33,209,1,4,1,205
```

ragaimeena, 2018-05-25:
I wonder if this is related
I followed the instruction and am able to control the pool from web browser ip-address:3000 for a few seconds then the port fails and i get this message
21:41:25.654 ERROR Error opening port: Opening COM6: Access denied.  Will retry in 10 seconds
Unhandled rejection TypeError: Cannot read property 'toJSON' of null
    at Object.push (C:\Users\ragai\pool\src\lib\comms\inbound\packet-buffer.js:30:30)
    at SerialPort.<anonymous> (C:\Users\ragai\pool\src\lib\comms\sp-helper.js:74:44)
    at emitNone (events.js:106:13)
    at SerialPort.emit (events.js:208:7)
    at emitReadable_ (_stream_readable.js:513:10)
    at emitReadable (_stream_readable.js:507:7)
    at addChunk (_stream_readable.js:274:7)
    at readableAddChunk (_stream_readable.js:250:11)
    at SerialPort.Readable.push (_stream_readable.js:208:10)
    at binding.read.then (C:\Users\ragai\pool\node_modules\serialport\lib\serialport.js:374:10)
    at tryCatcher (C:\Users\ragai\pool\node_modules\bluebird\js\release\util.js:16:23)
    at Promise._settlePromiseFromHandler (C:\Users\ragai\pool\node_modules\bluebird\js\release\promise.js:512:31)
    at Promise._settlePromise (C:\Users\ragai\pool\node_modules\bluebird\js\release\promise.js:569:18)
    at Promise._settlePromise0 (C:\Users\ragai\pool\node_modules\bluebird\js\release\promise.js:614:10)
    at Promise._settlePromises (C:\Users\ragai\pool\node_modules\bluebird\js\release\promise.js:693:18)
    at Async._drainQueue (C:\Users\ragai\pool\node_modules\bluebird\js\release\async.js:133:16)

21:41:35.652 ERROR Error opening port: Opening COM6: Access denied.  Will retry in 10 seconds
21:41:45.653 ERROR Error opening port: Opening COM6: Access denied.  Will retry in 10 seconds
21:41:55.653 ERROR Error opening port: Opening COM6: Access denied.  Will retry in 10 seconds
21:42:01.490 WARN queuePacket: Not queueing packet 165,1,16,33,134,2,2,1 because SerialPort/Net connect is not open.
21:42:01.491 INFO
        {"text":"User request to toggle POOL LIGHT to on","status":"on","value":1}
21:42:05.655 ERROR Error opening port: Opening COM6: Access denied.  Will retry in 10 seconds
21:42:15.655 ERROR Error opening port: Opening COM6: Access denied.  Will retry in 10 seconds

please help

tagyoureit, 2018-05-25:
Hi, can you turn on [logging](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution)?  I've seen the same issue before, but haven't been able to track it down.  If I can see what packets are causing the error, I can likely fix the [root] cause of your issue.  I'll still need to debug why serial port shuts down with any error in addition.

ragaimeena, 2018-05-25:
[output.log](https://github.com/tagyoureit/nodejs-poolController/files/2039239/output.log)
[output.log](https://github.com/tagyoureit/nodejs-poolController/files/2039240/output.log)
here is the log file

tagyoureit, 2018-05-25:
I don't see anything in the logs that would throw an error.  It looks like you receive one packet and then try to write back to the serial bus but it isn't successful.  I'm going to throw some additional error catching on the .toJson function you mentioned above and will push it up with the next iteration.  I'm working on a couple things now but will post when it is available.  Maybe a couple of days.

Also, do you have anything else that might be blocking/using the com port?  In your errors above it has `Opening COM6: Access denied.`.  Not sure what the access denied might be other than another app trying to also use the port.

ragaimeena, 2018-05-26:
I figured this issue out. I was connectin via the wireless extender and it needed to be hard wired
I am running into difficulty now getting integration with smartthings to work. I followed the directions but cannot get it to work. here is the error I am gettoing, please help if you can

13:39:00.169 INFO Virtual chlorinator controller not starting because it is set to default and another controller (Intellitouch/Intellicom) is present.
13:39:00.527 ERROR uncaughtException: configurationFileContent is not a function
        {"date":"Sat May 26 2018 13:39:00 GMT-0400 (Eastern Daylight Time)","process":{"pid":4480,"uid":null,"gid":null,"cwd":"C:\\Users\\ragai\\pool","execPath":"C:\\Program Files\\nodejs\\node.exe","version":"v8.9.4","argv":["C:\\Program Files\\nodejs\\node.exe","C:\\Users\\ragai\\pool\\src\\index.js"],"memoryUsage":{"rss":51515392,"heapTotal":36855808,"heapUsed":21107728,"external":966755}},"os":{"loadavg":[0,0,0],"uptime":12733.2218544},"trace":[{"column":40,"file":"C:\\Users\\ragai\\pool\\src\\etc\\settings.js","function":"Object.getConfig","line":611,"method":"getConfig","native":false},{"column":41,"file":"C:\\Users\\ragai\\pool\\src\\integrations\\outputToSmartThings.js","function":"GenericProvider.module.exports [as $get]","line":5,"method":"exports [as $get]","native":false},{"column":89,"file":"C:\\Users\\ragai\\pool\\node_modules\\bottlejs\\dist\\bottle.js","function":"Object.getService [as outputToSmartThings]","line":205,"method":"getService [as outputToSmartThings]","native":false},{"column":33,"file":"C:\\Users\\ragai\\pool\\src\\etc\\integrations.js","function":null,"line":71,"method":null,"native":false},{"column":17,"file":"C:\\Users\\ragai\\pool\\src\\etc\\integrations.js","function":null,"line":45,"method":null,"native":false},{"column":null,"file":null,"function":"Array.forEach","line":null,"method":"forEach","native":false},{"column":15,"file":"C:\\Users\\ragai\\pool\\src\\etc\\integrations.js","function":null,"line":41,"method":null,"native":false},{"column":15,"file":"fs.js","function":"FSReqWrap.oncomplete","line":135,"method":"oncomplete","native":false}],"stack":["TypeError: configurationFileContent is not a function","    at Object.getConfig (C:\\Users\\ragai\\pool\\src\\etc\\settings.js:611:40)","    at GenericProvider.module.exports [as $get] (C:\\Users\\ragai\\pool\\src\\integrations\\outputToSmartThings.js:5:41)","    at Object.getService [as outputToSmartThings] (C:\\Users\\ragai\\pool\\node_modules\\bottlejs\\dist\\bottle.js:205:89)","    at C:\\Users\\ragai\\pool\\src\\etc\\integrations.js:71:33","    at C:\\Users\\ragai\\pool\\src\\etc\\integrations.js:45:17","    at Array.forEach (<anonymous>)","    at C:\\Users\\ragai\\pool\\src\\etc\\integrations.js:41:15","    at FSReqWrap.oncomplete (fs.js:135:15)"]}
nodejs-poolController has closed successfully.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! nodejs-poolcontroller@5.0.0 start: `node src/index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the nodejs-poolcontroller@5.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\ragai\AppData\Roaming\npm-cache\_logs\2018-05-26T17_39_00_699Z-debug.log

C:\Users\ragai\pool>

tagyoureit, 2018-05-27:
HI, I responded to you in the forums.  Feel free to start a separate issue for this but let's not continue this discussion which isn't for this issue here.

tagyoureit, 2018-06-02:
Think I fixed the original issue with 5.1.  Let me know if it comes back.

closed 2018-06-02 by tagyoureit