tagyoureit/nodejs-poolController #98: Hello. I am testing this with just an Intellichem connected via RS485

ArtemioUrbina, 2018-07-10:
I am testing this with the JBtek USB to RS485 and just an Intellichem connected to it. I have no other piece of equipment.

I always get the message: "INFO Serial port recovering from lost connection" And "od -x < /dev/ttyUSB0" is completely silent, and tty config is correct (dmesg gives "ch341-uart converter now attached to ttyUSB0"
)

Checked the lines with a scope, and there is no activity at all (even while testing directly at the DATA+ DATA- terms of the Intellichem). Do I need something else? Do I need to configure or send something to start communications?

Do I need extra hardware in the setup? Config on the Intellichem? Any other tests? 

Thanks for your work and help.

tagyoureit, 2018-07-11:
The code today requires an Intellitouch.  To get this working without an Intellitouch there would need to be code added to emulate the intellitouch.  I've added code to make the chlorinator and the pumps work standalone, but I don't have an Intellichem and it would be rather difficult.  If there is someone who wants to help with this that has both an Intellichem and Intellitouch we might be able to get the correct packets, but I don't have the resources to do it currently.

tagyoureit, 2018-11-16:
Closing due to lack of activity.

closed 2018-11-16 by tagyoureit

ArtemioUrbina, 2020-08-25:
> 
> 
> Thanks to @happyiphone we were able to successfully poll and send/receive commands to the Intellichem without a OCP (Outdoor control panel). If you'd like to test this out, @ArtemioUrbina , please pull the latest `next` branch for this app and also the -webClient (also `next` branch). It will be pushed to `master` in the near future.

This is awesome, thank you very much. I'll set everything up this week and report back.