tagyoureit/nodejs-poolController #99: Unhandled rejection TypeError: Cannot set property 'light' of undefined, with two Intellibrites and logging on

emes, 2018-07-20:
### Expected behavior and actual behavior

When logIntellibrite is on, I'm getting the following error:

Unhandled rejection TypeError: Cannot set property 'light' of undefined

and thereafter everything goes to hell :-) (all packetWrites fail)

### What is actually happening (screenshots are helpful)

The actual exception:

```
Unhandled rejection TypeError: Cannot set property 'light' of undefined
    at Object.assignControllerLightGroup (/home/ms/nodejs-poolController/src/lib/equipment/circuit.js:774:61)
    at Object.process (/home/ms/nodejs-poolController/src/lib/comms/inbound/controller/39.js:34:27)
    at Object.processControllerPacket (/home/ms/nodejs-poolController/src/lib/comms/inbound/process-controller.js:115:45)
    at decode (/home/ms/nodejs-poolController/src/lib/comms/inbound/decode-helper.js:263:51)
    at Object.processChecksum (/home/ms/nodejs-poolController/src/lib/comms/inbound/decode-helper.js:301:13)
    at Object.iterateOverArrayOfArrays (/home/ms/nodejs-poolController/src/lib/comms/inbound/receive-buffer.js:117:44)
    at Object.push (/home/ms/nodejs-poolController/src/lib/comms/inbound/packet-buffer.js:36:37)
    at SerialPort.<anonymous> (/home/ms/nodejs-poolController/src/lib/comms/sp-helper.js:78:44)
    at emitNone (events.js:106:13)
    at SerialPort.emit (events.js:208:7)
    at emitReadable_ (_stream_readable.js:513:10)
    at emitReadable (_stream_readable.js:507:7)
    at addChunk (_stream_readable.js:274:7)
    at readableAddChunk (_stream_readable.js:250:11)
    at SerialPort.Readable.push (_stream_readable.js:208:10)
    at binding.read.then (/home/ms/nodejs-poolController/node_modules/serialport/lib/serialport.js:374:10)
```

Probably related, even without logIntellibrite, I see the following error in the javascript console when using the bootstrap UI.

```
Uncaught TypeError: Cannot read property 'colorSetStr' of undefined
    at Object.<anonymous> (main.js:791)
    at Function.each (jquery.js:360)
    at insertLightEdit (main.js:627)
    at r.<anonymous> (main.js:1051)
    at r.emit (index.js:83)
    at r.onevent (index.js:83)
    at r.onpacket (index.js:83)
    at r.<anonymous> (index.js:83)
    at r.emit (index.js:83)
    at r.ondecoded (index.js:83)
```

### Steps to reproduce the problem

change config.json so log.logIntellibrite is 1

### Specifications

1. What version of the code are you using?

5.1.1

2. All output from the file log

https://gist.github.com/emes/213c2d6d4adb1e1bf31b6c6862d63a9a

3. Any errors that are in the console

see above.

4. Your config.json or in-use configuration file

fresh install using default config.json, see logs for embedded config

5. Your pool equipment

IntelliTouch 5+S (with IntelliChlor disabled), pool-only
IntelliFlow VS
(2) IntelliBrite lights, each on a separate relay
(2) valve actuators, on when Pool Heat on

tagyoureit, 2018-07-22:
Can you replace lines 774-779 with the following:
```
if (currentCircuitArrObj[cir].hasOwnProperty('light')) {
    currentCircuitArrObj[cir].light = new Light(diff1[key].rhs.position, 'off', 0)

    if (logIntellibrite) {
        container.logger.silly('NEW key:', JSON.stringify(diff1[key], null, 2))
        container.logger.verbose('Msg# %s  Light details added for circuit %s (%s):', counter, getFriendlyName(cir), cir, diff1[key].rhs.position)
    }
}
else {
    logger.warn('Trying to add light to circuit %s but it has no light property. \n\t %j', currentCircuitArrObj[cir].number, currentCircuitArrObj[cir])
}
```

Let me know what it says.  I'm also working on a better way to debug these (capture/replay packets), but am waiting on Winston for some unrelated changes.

emes, 2018-07-23:
Thanks.  Did that, here's the output:

```
01:28:03.425 INFO Potential expired/deprecated keys in
        file: config.json
        key: equipment.circuit.friendlyName.21:""
        key: equipment.circuit.friendlyName.22:""
        key: equipment.circuit.friendlyName.23:""
        key: equipment.circuit.friendlyName.24:""
        key: equipment.circuit.friendlyName.25:""
        key: equipment.circuit.friendlyName.26:""
        key: equipment.circuit.friendlyName.27:""
        key: equipment.circuit.friendlyName.28:""
        key: equipment.circuit.friendlyName.29:""
        key: equipment.circuit.friendlyName.30:""
        key: equipment.circuit.friendlyName.31:""
        key: equipment.circuit.friendlyName.32:""
        key: equipment.circuit.friendlyName.33:""
        key: equipment.circuit.friendlyName.34:""
        key: equipment.circuit.friendlyName.35:""
        key: equipment.circuit.friendlyName.36:""
        key: equipment.circuit.friendlyName.37:""
        key: equipment.circuit.friendlyName.38:""
        key: equipment.circuit.friendlyName.39:""
        key: equipment.circuit.friendlyName.40:""
        key: equipment.circuit.friendlyName.41:"Pump Medium"
        key: equipment.circuit.friendlyName.42:"Pump High"
        key: equipment.circuit.friendlyName.43:"Pump Max"
        key: equipment.circuit.friendlyName.44:"Cl Boost"
        key: equipment.circuit.friendlyName.45:"Cl Daily"
        key: equipment.circuit.friendlyName.46:""
        key: equipment.circuit.friendlyName.47:""
        key: equipment.circuit.friendlyName.48:""
        key: equipment.circuit.friendlyName.49:""
        key: equipment.circuit.friendlyName.50:""
01:28:03.565 INFO Starting up express server, https (port 33001)
01:28:03.582 INFO Starting up express server, http (port 33000)
01:28:03.593 INFO Starting up MDNS server
01:28:03.608 INFO initializing logger
01:28:03.685 DEBUG Initialized intellichem module
01:28:03.685 INFO Configuration settings:
01:28:03.685 INFO Settings:

*******************************
 Version: 5.1.0

Configuration file name: config.json
*******************************{
    "version": "5.1.0",
    "equipment": {
        "controller": {
            "intellicom": {
                "installed": 0,
                "friendlyName": ""
            },
            "intellitouch": {
                "installed": 1,
                "friendlyName": ""
            },
            "virtual": {
                "pumpController": "default",
                "chlorinatorController": "default"
            }
        },
        "circuit": {
            "hideAux": false,
            "friendlyName": {
                "1": "",
                "2": "Levolor",
                "3": "Pool Light",
[... 353 lines cut ...]
    at binding.read.then (/home/ms/pooldock/nodejs-poolController/node_modules/serialport/lib/serialport.js:374:10)

01:28:21.166 WARN Error writing controller packet 'Get Schedule' to serial bus.  Tried 5 times to write 255,0,255,165,1,16,33,209,1,1,1,170
01:28:23.553 ERROR Aborting controller packet Get Schedule.  Tried 10 times to write 255,0,255,165,1,16,33,209,1,1,1,170
01:28:26.113 WARN Error writing controller packet 'Get Schedule' to serial bus.  Tried 5 times to write 255,0,255,165,1,16,33,209,1,2,1,171
01:28:28.502 ERROR Aborting controller packet Get Schedule.  Tried 10 times to write 255,0,255,165,1,16,33,209,1,2,1,171
01:28:28.654 ERROR Serial port connection lost.  Will retry every 10 seconds to reconnect.
01:28:28.658 INFO Serial port recovering from lost connection.
01:28:38.658 INFO Serial port recovering from lost connection.
^CShutting down open processes
01:28:39.418 SILLY http has 0 connections in it:
01:28:39.420 DEBUG Stopping Socket IO Server
01:28:39.420 SILLY All sockets removed from connection
01:28:39.421 DEBUG Socket IO Server closed
01:28:39.421 SILLY https has 0 connections in it:
01:28:39.422 DEBUG Stopping Socket IO Server
01:28:39.422 SILLY All sockets removed from connection
01:28:39.422 DEBUG Socket IO Server closed
01:28:39.423 VERBOSE SSDP/uPNP Server closed
01:28:39.424 VERBOSE Express Server http closed
01:28:39.424 VERBOSE Express Server https closed
01:28:39.424 VERBOSE All express + ssdp servers closed
nodejs-poolController services stopped successfully
nodejs-poolController has closed successfully.
```

emes, 2018-07-23:
And here's output.log with all logging on...

[output.log](https://github.com/tagyoureit/nodejs-poolController/files/2218015/output.log)

tagyoureit, 2018-07-24:
There are a few places that the code needs to change.  I think I have most of them identified, which is the good news.  Bad news is that you'll need to wait until I fix them for the Intellibrites work.  

In the mean time, if you want the rest of the code working without it crashing sans Intellibrite, you can comment out the following lines:
* https://github.com/tagyoureit/nodejs-poolController/blob/master/src/lib/comms/inbound/controller/39.js#L34
* https://github.com/tagyoureit/nodejs-poolController/blob/master/src/lib/comms/inbound/controller/96.js#L34

tagyoureit, 2018-07-25:
Thanks for the typo note...
I just pushed up a [5.2 Dev](https://github.com/tagyoureit/nodejs-poolController/tree/5.2-DEV) branch.  Can you try this out?  It should be good for reading the Intellibrite messages.  I took a stab at coding the set/change methods (change color, swim delay, light position) in the same manner.  I'm curious if they will work for you.  If not, I'll need your help with some packet captures.

This branch should also fix the 
> and thereafter everything goes to hell :-) (all packetWrites fail)

Cheers.

emes, 2018-09-28:
Thanks again!

closed 2018-09-28 by emes