tagyoureit/nodejs-poolController #100: add patch number to function deviceXML in helpers.js

dhop90, 2018-07-22:
### Expected behavior and actual behavior
function deviceXML doesn't provide patch number
provide patch number to function output

curl -s http://<user><password>@localhost:3000/device | xq .
{
  "root": {
    "@xmlns": "urn:schemas-upnp-org:PoolController-1-0",
    "specVersion": {
      "major": "5",
      "minor": "1",
      "patch": "1"
    },
    "device": {
      "deviceType": "urn:echo:device:PoolController:1",
      "friendlyName": "NodeJS Pool Controller",
      "manufacturer": "tagyoureit",
      "manufacturerURL": "https://github.com/tagyoureit/nodejs-poolController",
      "modelDescription": "An application to control pool equipment.",
      "serialNumber": "0",
      "UDN": "uuid:806f52f4-1f35-4e33-9299-b827ebbc1405",
      "serviceList": null
    }
  }
}

### What is actually happening (screenshots are helpful)

### Steps to reproduce the problem

### Specifications

1. What version of the code are you using?
2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment

diffs:
227c227,229
<                         XML += "</minor></specVersion><device><deviceType>urn:echo:device:PoolController:1</deviceType><friendlyName>NodeJS Pool Controller</friendlyName><manufacturer>tagyoureit</manufacturer><manufacturerURL>https://github.com/tagyoureit/nodejs-poolController</manufacturerURL><modelDescription>An application to control pool equipment.</modelDescription><serialNumber>0</serialNumber>				<UDN>uuid:806f52f4-1f35-4e33-9299-";
---
> 			XML += "</minor><patch>"
>                         XML += results.local.version.split('.')[2]
>                         XML += "</patch></specVersion><device><deviceType>urn:echo:device:PoolController:1</deviceType><friendlyName>NodeJS Pool Controller</friendlyName><manufacturer>tagyoureit</manufacturer><manufacturerURL>https://github.com/tagyoureit/nodejs-poolController</manufacturerURL><modelDescription>An application to control pool equipment.</modelDescription><serialNumber>0</serialNumber>				<UDN>uuid:806f52f4-1f35-4e33-9299-";

tagyoureit, 2018-07-22:
Added.  Will include in the next release.  Thx.

tagyoureit, 2018-07-27:
Included in 5.2-DEV branch.  Thx.

closed 2018-07-27 by tagyoureit