tagyoureit/nodejs-poolController #102: V5.1.1 believes it's running 5.1.0

mark-hoy, 2018-08-25:
### Expected behavior and actual behavior
Latest should not suggest an update is available.

### What is actually happening (screenshots are helpful)
bottom right of web page says there is an update available

### Steps to reproduce the problem
npm start

### Specifications

1. What version of the code are you using?
v5.1.1

2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
3. Any errors that are in the console
4. Your config.json or in-use configuration file
config.json:    "version": "5.1.0",
sysDefault.json:  "version" : "5.1.0",

mark-hoy, 2018-08-25:
16:36:41.530 WARN Update available!  Version 5.1.1 can be installed.  You have 5.1.0

tagyoureit, 2018-08-26:
Yes, I know about this one.  Thanks.  Forgot to bump the version last push.  Will be fixed with the next code push.

closed 2018-11-16 by tagyoureit