tagyoureit/nodejs-poolController #103: Lots of undefined and can't write to serial bus

csblank, 2018-09-05:
Having problems writing to serial bus.  I have a Waveshare HAT RS485.  I have a Pentair Easy Touch Controller with a heat pump and Intellibright light.

> nodejs-poolcontroller@5.1.0 start /Pubs/nodejs-poolController-master
> node src/index.js

19:34:08.260 INFO Starting up express server, http (port 3000)
19:34:08.386 INFO Starting up MDNS server
19:34:08.502 INFO initializing logger
19:34:09.118 INFO Configuration settings:
19:34:09.120 INFO Settings:

*******************************
 Version: 5.1.0
 
Configuration file name: config.json
*******************************{
    "version": "5.1.0",
    "equipment": {
        "controller": {
            "intellicom": {
                "installed": 0,
                "friendlyName": ""
            },
            "intellitouch": {
                "installed": 1,
                "friendlyName": ""
            },
            "virtual": {
                "pumpController": "default",
                "chlorinatorController": "default"
            }
        },
        "circuit": {
            "hideAux": false,
            "friendlyName": {
                "1": "",
                "2": "",
                "3": "",
                "4": "",
                "5": "",
                "6": "",
                "7": "",
                "8": "",
                "9": "",
                "10": "",
                "11": "",
                "12": "",
                "13": "",
                "14": "",
                "15": "",
                "16": "",
                "17": "",
                "18": "",
                "19": "",
                "20": ""
            }
        },
        "chlorinator": {
            "installed": 1,
            "desiredOutput": {
                "pool": -1,
                "spa": -1
            },
            "friendlyName": "",
            "id": {
                "productName": ""
            }
        },
        "pump": {
            "1": {
[... 277 lines cut ...]
19:47:04.366 INFO Received valve status packet.  
        Possible association with undefined. 
          valve data: 165,1,15,16,2,29,15,54,32,0,0,0,0,0,0,32,0,0,0,4,87,87,32,0,97,0,0,0,4,0,0,134,245,3,13,4,43
19:47:04.764 ERROR Aborting controller packet Set Heat/Temperature.  Tried 10 times to write 255,0,255,165,1,16,33,136,4,83,100,4,0,2,30
19:47:07.323 WARN Error writing controller packet 'Get Heat/Temperature' to serial bus.  Tried 5 times to write 255,0,255,165,1,16,33,200,1,0,1,160
19:47:09.707 ERROR Aborting controller packet Get Heat/Temperature.  Tried 10 times to write 255,0,255,165,1,16,33,200,1,0,1,160
19:47:11.097 WARN Setting logging level to info
19:47:37.430 INFO Received valve status packet.  
        Possible association with undefined. 
          valve data: 165,1,15,16,2,29,15,55,32,0,0,0,0,0,0,32,0,0,0,4,87,87,32,0,97,0,0,0,4,0,0,134,245,3,13,4,44
19:47:57.663 INFO Received valve status packet.  
        Possible association with undefined. 
          valve data: 165,1,15,16,2,29,15,55,32,0,0,0,0,0,0,32,0,0,0,4,87,87,32,0,96,0,0,0,4,0,0,134,245,3,13,4,43
19:48:05.758 INFO Received valve status packet.  
        Possible association with undefined. 
          valve data: 165,1,15,16,2,29,15,55,32,0,0,0,0,0,0,32,0,0,0,4,87,87,32,0,97,0,0,0,4,0,0,134,245,3,13,4,44
19:48:38.837 INFO Received valve status packet.  
        Possible association with undefined. 
          valve data: 165,1,15,16,2,29,15,56,32,0,0,0,0,0,0,32,0,0,0,4,87,87,32,0,97,0,0,0,4,0,0,134,245,3,13,4,45
19:49:38.225 INFO Received valve status packet.  
        Possible association with undefined. 
          valve data: 165,1,15,16,2,29,15,57,32,0,0,0,0,0,0,32,0,0,0,4,87,87,32,0,97,0,0,0,4,0,0,134,245,3,13,4,46
19:50:39.621 INFO Received valve status packet.  
        Possible association with undefined. 
          valve data: 165,1,15,16,2,29,15,58,32,0,0,0,0,0,0,32,0,0,0,4,87,87,32,0,97,0,0,0,4,0,0,134,245,3,13,4,47

tagyoureit, 2018-09-05:
You need to set `intellitouch: 0`.  Intellitouch has settings that the Easy Touch doesn't have and those are the messages that the app is trying to send.

csblank, 2018-09-06:
I made the change you suggested and now, I'm getting the following messages.

00:37:09.771 INFO Starting up SSDP server
00:37:10.389 WARN Update available!  Version 5.1.1 can be installed.  You have 5.1.0
00:37:11.021 INFO No pool controller (Intellitouch or IntelliComII) detected.  No configuration request messages sent.
00:37:13.889 INFO Received valve status packet.
        Possible association with undefined.
          valve data: 165,1,15,16,2,29,20,44,32,0,0,0,0,0,0,32,0,0,0,4,86,86,32,0,80,0,0,0,4,0,0,134,245,3,13,4,19
00:37:15.955 WARN Error writing pump packet 'Pump control panel' to serial bus.  Tried 5 times to write 255,0,255,165,0,96,33,4,1,255,2,42
00:37:18.343 ERROR Aborting pump packet Pump control panel.  Tried 10 times to write 255,0,255,165,0,96,33,4,1,255,2,42
00:37:18.347 WARN Setting logging level to Debug.  Will revert to previous level in 2 minutes.
00:37:20.913 WARN Error writing pump packet 'Pump Status' to serial bus.  Tried 5 times to write 255,0,255,165,0,96,33,7,0,1,45
00:37:23.302 ERROR Aborting pump packet Pump Status.  Tried 10 times to write 255,0,255,165,0,96,33,7,0,1,45
00:37:25.867 WARN Error writing chlorinator packet 'Set Salt %' to serial bus.  Tried 5 times to write 16,2,80,17,0,115,16,3
00:37:28.254 ERROR Aborting chlorinator packet Set Salt %.  Tried 10 times to write 16,2,80,17,0,115,16,3

tagyoureit, 2018-09-06:
Set both of these to `never`...
> "virtual": {
"pumpController": "default",
"chlorinatorController": "default"

csblank, 2018-09-06:
I set the pump controller to "never".  I left the chlorinator set to default, As I have a "Pentair IntelliChlor IC40".  Now, when I start the program I get the following. I was trying to change the temperature.  Seems like I'm having issue writing to the serial bus.  Its a 2-wire setup so perhaps its something with signaling the "transmit".  Wasn't sure if you had any knowledge on RS485 directly.  I'm using a Raspberry PI HAT.  I know that I have the DT+ and DT- right, else I couldn't ready any values.

14:05:17.217 INFO 
        {"text":"User request to update pool heat set point to 83: 165,1,16,33,136,4,83,100,4,0,2,30","status":"OFF","value":83}
14:05:19.608 WARN Error writing controller packet 'Set Heat/Temperature' to serial bus.  Tried 5 times to write 255,0,255,165,1,16,33,136,4,83,100,4,0,2,30
14:05:22.007 ERROR Aborting controller packet Set Heat/Temperature.  Tried 10 times to write 255,0,255,165,1,16,33,136,4,83,100,4,0,2,30

tagyoureit, 2018-09-06:
I'm not sure how the Easy Touch handles heat.  It could be a different command.  Do you have any remote panels/controllers attached to your Easy Touch?  If so, I would suggest downloading the 5.2-DEV branch and using the new [capture tool](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution).  When you are recording you would change the heat and capture what the remote controller sends to Easy Touch and we could add it to the system.  Otherwise, you'll most likely be using this app in a read-only mode for many aspects.

closed 2018-11-16 by tagyoureit

MaxVonEvil, 2019-04-27:
This is an old closed issue, yet I ran into the same issue. 
In my case it turned out it was the internal serial interface which was mucking things up. To fix:
1) sudo raspi-config
2) under interfacing options, serial, answer no to both questions to disable the serial interface.

@tagyoureit  - you might want to add this to the installation instructions?

tagyoureit, 2019-05-02:
Thanks... added to the [wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details#raspi).