tagyoureit/nodejs-poolController #104: Serial port recovering from lost connection

bsileo, 2018-11-04:
### Expected behavior and actual behavior

### What is actually happening (screenshots are helpful)
All control and updates to pool equipment are lost.

### Steps to reproduce the problem
Seems to happen at startup. After multiple reboots (often cold) I can get it to start working for some time.

### Specifications

1. What version of the code are you using?
5.2.0

2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-
[sileo-packetCapture.zip](https://github.com/tagyoureit/nodejs-poolController/files/2546393/sileo-packetCapture.zip)

3. Any errors that are in the console
22:45:35.456 INFO Serial port recovering from lost connection.
22:45:45.459 INFO Serial port recovering from lost connection.
22:45:55.464 INFO Serial port recovering from lost connection.
22:46:05.467 INFO Serial port recovering from lost connection.

4. Your config.json or in-use configuration file
5. Your pool equipment
Pentair

bsileo, 2018-11-05:
Here is a list of the installed packages as well in case this is a version issue of some sort?
[node_packages.txt](https://github.com/tagyoureit/nodejs-poolController/files/2546455/node_packages.txt)

tagyoureit, 2018-11-06:
Something happened with the URL and I can't download the packet capture zip file.  Can you please reattach?

bsileo, 2018-11-06:
[packetCapture.zip](https://github.com/tagyoureit/nodejs-poolController/files/2554116/packetCapture.zip)
Here you go.

bsileo, 2018-11-06:
Problem resolved. I found and fixed a wiring issue.

closed 2018-11-06 by bsileo

MaxVonEvil, 2019-04-23:
@bsileo - looks like I might be having the same. 
Can you perhaps elaborate on what your wiring issue was? Is it a cable length issue or just a bad connector at the pump?

I've tried with 2 separate JBtek RS485 interfaces and while I can (somewhat) talk to my stand-alone pump, the comms are spotty and I get the same "Serialport recovering from..." all over the place.

I'm kinda running out of ideas, short of trying another cable, so any advice would be appreciated.

bsileo, 2019-04-24:
This was a loose connection in an old splice I had on the physical wire coming from the pool panel. It was there as I was using it to connect the Pentair Protocol Adapter alongside my RS-485 device.  I ended up just pulling a new cable from the panel into the basement room where I have the PoolController and then this was resolved.

MaxVonEvil, 2019-04-24:
Okay, thanks @bsileo - I might end up having to run a new cable as well it seems :)

That said, your work on the pool controller suite of DHT's and smartapps is super impressive, yet as I only have a stand-alone pump, I'm looking to write a super-simple DHT for the pump, acting somewhat like a dimmer device, i.e. translating a 0-100% slider value to the proper RPM range, firing off the proper REST commands. That way I should be able to use everything from WebCore to even the SmartLighting app to run the regular pump schedules. Just wondering if you might have had similar thoughts at some point?

bsileo, 2019-04-26:
I like this idea I just never had the need as I do not spend much time
thinking about pump speed with my configuration. What would be ideal is to
add this as a child device in my App so you can just use the pump device as
well as the rest of the capabilities. The child device modeling Let's your
expose a single device while still having it fit into a bigger set of
capabilities. Do you want to collaborate on this approach?

johnny2678, 2019-04-26:
@bsileo @MaxVonEvil 

Great minds - I just opened #123 yesterday.  If @tagyoureit can create an API layer to change the RPMs on the Pump circuit, then that should be your mechanism for a Slider in ST, yes?