tagyoureit/nodejs-poolController #105: Pump "wattsnotset", "rpmnotset"

LavaGlass, 2018-11-13:
### Expected behavior and actual behavior
Opening the web UI expect to see the pump values but are just populated with "wattsnotset", "rpmnotset" and the same for all the others.

### What is actually happening (screenshots are helpful)
[image]

### Steps to reproduce the problem
No steps required to reproduce it just shows up in the web UI.

### Specifications

1. What version of the code are you using? 5.2.0
2. All output from the file log. https://drive.google.com/open id=1qI0EB45IdPKC_wv8HcUr5AMxWK7AVlZ5
3. Any errors that are in the console Not sure if this is an error but "INFO Serial port recovering from lost connection." comes up every now an then.
4. Your config.json or in-use configuration file
`{
    "version": "5.2.0",
    "equipment": {
        "controller": {
            "intellicom": {
                "installed": 1,
                "friendlyName": ""
            },
            "intellitouch": {
                "installed": 0,
                "friendlyName": "",
                "numberOfCircuits": 20,
                "numberOfPumps": 2,
                "numberOfCustomNames": 10,
                "numberOfSchedules": 12
            },
            "virtual": {
                "pumpController": "default",
                "chlorinatorController": "default"
            }
        },
        "circuit": {
            "hideAux": false,
            "friendlyName": {
                "1": "",
                "2": "",
                "3": "",
                "4": "",
                "5": "",
                "6": "",
                "7": "",
                "8": "",
                "9": "",
                "10": "",
                "11": "",
                "12": "",
                "13": "",
                "14": "",
                "15": "",
                "16": "",
                "17": "",
                "18": "",
                "19": "",
                "20": ""
            }
        },
        "chlorinator": {
            "installed": 0,
            "desiredOutput": {
                "pool": -1,
                "spa": -1
            },
            "friendlyName": "",
            "id": {
                "productName": ""
            }
        },
        "pump": {
            "1": {
                "type": "VF",
                "externalProgram": {
                    "1": 30,
                    "2": 30,
                    "3": 30,
                    "4": 30
                },
                "friendlyName": "Pump 1"
            },
            "2": {
                "type": "none",
                "externalProgram": {
                    "1": -1,
                    "2": -1,
                    "3": -1,
                    "4": -1
                },
                "friendlyName": ""
            }
        },
        "intellichem": {
            "installed": 0
        },
        "spa": {
            "installed": 0
        },
        "solar": {
            "installed": 0
        }
    },
    "poolController": {
        "appAddress": 33,
        "http": {
            "enabled": 1,
            "expressPort": 3000,
            "redirectToHttps": 0,
            "expressAuth": 0,
            "expressAuthFile": "/users.htpasswd"
        },
        "https": {
            "enabled": 0,
            "expressPort": 3001,
            "expressAuth": 0,
            "expressAuthFile": "/users.htpasswd",
            "expressKeyFile": "/data/server.key",
            "expressCertFile": "/data/server.crt"
        },
        "network": {
            "rs485Port": "/dev/ttyUSB0",
            "netConnect": 0,
            "netHost": "raspberrypi",
            "netPort": 9801,
            "inactivityRetry": 60
        },
        "log": {
            "logLevel": "info",
            "socketLogLevel": "info",
            "logPumpMessages": 0,
            "logDuplicateMessages": 0,
            "logConsoleNotDecoded": 0,
            "logConfigMessages": 0,
            "logMessageDecoding": 0,
            "logChlorinator": 0,
            "logIntellichem": 0,
            "logPacketWrites": 0,
            "logPumpTimers": 0,
            "logReload": 0,
            "logApi": 0,
            "logIntellibrite": 0,
            "fileLog": {
                "enable": 0,
                "fileLogLevel": "silly",
                "fileName": "output.log"
            }
        },
        "database": {
            "influx": {
                "enabled": 0,
                "host": "localhost",
                "port": 8086,
                "database": "pool"
            }
        }
    },
    "meta": {
        "notifications": {
            "version": {
                "remote": {
                    "version": "5.2.0",
                    "tag_name": "v5.2.0",
                    "dismissUntilNextRemoteVersionBump": false
                }
            }
        }
    },
    "integrations": {}
}`
5. Your pool equipment
Intelliflo VF and Intellicomm

tagyoureit, 2018-11-16:
Intellicom doesn't have the logic to tell this app what the watts, rpm, etc are.  You may want to try and set `"virtual": { "pumpController": "default"` to `"virtual": { "pumpController": "always"`.  This will have the app try to read the values directly from the pump.  Just make sure you don't have this app and the Intellicom both try to run the pump at the same time as the pump will get conflicting messages.  Enabling the `always` value will give this app the ability to control the pump directly, but don't use it unless you turn off all pump controls on the Intellicom.

tagyoureit, 2019-01-03:
Closing.  Please reopen if you still have issues.

closed 2019-01-03 by tagyoureit