tagyoureit/nodejs-poolController #107: Not all data written to Influx

arrmo, 2018-12-31:
### Expected behavior and actual behavior
Some temperature data seems to be showing up in Influx (e.g. airTemp, poolSetPoint, spaSetPoint) ... but the rest is missing? Thinking pool (water) temp in particular.

### What is actually happening (screenshots are helpful)
[image]

### Steps to reproduce the problem
Just run (enable) Influx. Is there a setting in config.json that allows logging for Influx (to help debug this)?

### Specifications

1. What version of the code are you using?
5.2-DEV.

2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))
Will add, once I know how to enable this integration. Thanks!

3. Any errors that are in the console
4. Your config.json or in-use configuration file
5. Your pool equipment

tagyoureit, 2019-01-03:
Not everything is included.  We can add more things but I tried to capture only the really important ones here.

arrmo, 2019-01-07:
Yep, that makes sense - but poolTemp seems pretty key to me ... LOL. It's odd, only the 3 noted above are stored - yet the code seems to call out several more (including poolTemp, not sure how / why that's missing).

Thoughts?

arrmo, 2019-01-07:
BTW, sorry for the slow reply! Seems my Github notifications were automatically being deleted (by Office 365). If you're waiting on me for something else, please poke!

arrmo, 2019-01-09:
Is there a way to control the outputs from this integration? Would like to help debug it, but not sure if there is a flag to get debug output.

Thanks!

tagyoureit, 2019-01-09:
Are you using the version in the integrations directory?  That one is waaay old.  There is a built-in version that you can enable just through the config file.  Not super well documented, but look for it in the [example config file](https://github.com/tagyoureit/nodejs-poolcontroller#only-output-changes-in-configjson).

arrmo, 2019-01-09:
Yep, my config file looks like the example - so I THINK I'm using the right one ... :-).

I do get some output to influx ... but for temperatures, only the 3 noted above (airTemp, poolSetPoint, spaSetPoint). Very odd, the others I see in the code are missing.

tagyoureit, 2019-01-09:
This is a view from Grafana, but it is all the fields I get in temperature.
[image]

arrmo, 2019-01-09:
Hmmm ... OK, that's very odd. What I captured above is all I see in influx (data-explorer). So I just stopped nodejs-poolController, removed the database, re-created the database, and restarted nodejs-poolController => I see exactly as above, just the three Fields (as influx calls it).

Not really sure what to say ... LMAO

arrmo, 2019-01-09:
Just tried Grafana also - the same, only the 3 Fields.

arrmo, 2019-01-09:
Also, FYI ... CLI query in influx,
```
> select * from temperature
name: temperature
time                airTemp poolSetPoint source                spaSetPoint
----                ------- ------------ ------                -----------
1546999985600668850 51      78           nodejs-poolcontroller 103
1546999989915658388 51      78           nodejs-poolcontroller 103
1546999993847962332 51      78           nodejs-poolcontroller 103
1546999997778317090 51      78           nodejs-poolcontroller 103
1547000007697095591 51      78           nodejs-poolcontroller 103
1547000037124462020 51      78           nodejs-poolcontroller 103
1547000046974114802 51      78           nodejs-poolcontroller 103
```

tagyoureit, 2019-01-09:
Oh, it's because your pump is off.  When your pool/pump turns on it will get the rest of the measurements.  I implemented it this way because when your pump is off it caches the last value and isn't really reflective of the real pool temp.  See the [code](https://github.com/tagyoureit/nodejs-poolController/blob/master/src/lib/comms/influx-connector.js#L147).

arrmo, 2019-01-09:
That makes sense! And I like that logic ... :-) But a slight hitch - just checked, pool is running - but no additional measurements exist ... :-(.

tagyoureit, 2019-01-09:
Do you have an older pump?  When you get back `/pump` what do you see?

```
"pump": {
"1": {
"pump": 1,
"name": "Pump 1",
"type": "VS",
"time": "10:09 PM",
"run": 10,
"mode": 0,
"drivestate": 0,
"watts": 160,
"rpm": 1300,
"gpm": 0,
"ppc": 0,
"err": 0,
"timer": 1,
"duration": "durationnotset",
"currentrunning": {
"mode": "off",
"value": 0,
"remainingduration": -1
},
"externalProgram": {
"1": -1,
"2": -1,
"3": -1,
"4": -1
},
"remotecontrol": 1,
"power": 1,
"friendlyName": "Pump 1",
"virtualController": "disabled"
}
}
```

the `power: 1` is what I'm looking for.  Maybe that doesn't work if you don't have a smart pump.

arrmo, 2019-01-09:
OK, that's it - glad I'm not (completely) crazy ... ;-). Yes, I have an old (dumb ... :-)) SS pump. Info, below (from /pump),
```
{  
   "pump":{  
      "1":{  
         "pump":1,
         "name":"Pump 1",
         "type":"SS",
         "time":"timenotset",
         "run":"runnotset",
         "mode":"modenotset",
         "drivestate":"drivestatenotset",
         "watts":"wattsnotset",
         "rpm":"rpmnotset",
         "gpm":"gpmnotset",
         "ppc":"ppcnotset",
         "err":"errnotset",
         "timer":"timernotset",
         "duration":"durationnotset",
         "currentrunning":{  
            "mode":"off",
            "value":0,
            "remainingduration":-1
         },
         "externalProgram":{  
            "1":-1,
            "2":-1,
            "3":-1,
            "4":-1
         },
         "remotecontrol":"remotecontrolnotset",
         "power":"powernotset",
         "friendlyName":"Pump 1",
         "virtualController":"disabled"
      }
   }
}
```

BTW, in influx the pump on/off does seem to be there - it's just this data not being written out.

arrmo, 2019-01-13:
BTW, any pointers of where this is (in the code)? I can play with it here, leave both options on the table - then send you an PR.

Thanks!

tagyoureit, 2019-01-13:
You could do something like this:

```
function poolOrSpaIsOn() {
    // return all non-light circuits
    const circuit = container.circuit.getAllLightCircuits()

    // map the obj to an array
    const circuitMap = Object.values(circuit)
    // loop through the circuits
    const results = circuitMap.filter((n) => {
        // find the Spa or Pool circuit
        if (n.circuitFunction === "Spa" || n.circuitFunction === 'Pool') {
            // if the circuit is on return the filtered value in the list
            if (n.status) {
                console.log(`${n.circuitFunction} is ${n.status}`)
                return true
            }
        }
    }
    )

    // if the list has 1+ "on" entry then the pool/spa pump is on
    return Object.keys(results).length>=1
}

if (poolOrSpaIsOn()) {
    console.log("good to go!")
}
else {
    console.log("nothing is on.  :()")
} 
```

It might even make sense to put it in the circuit code and then just make a single call.  I'm sure this would be useful elsewhere in the code, too.

arrmo, 2019-01-15:
Yes, this makes sense, and I can add it in - but as things have been reorganized lately ... where is the influx (data output) routine at now? Thanks!

tagyoureit, 2019-01-15:
Same place... this file didn't move.  https://github.com/tagyoureit/nodejs-poolController/tree/6.0-DEV/src/lib/comms

arrmo, 2019-01-17:
Thanks for the pointers! Code updated, PR#112 created. Let me know if it works for you - it does here.

tagyoureit, 2019-01-21:
Accepted!  [For public record - this is on the 6.0-DEV branch]

closed 2019-01-21 by tagyoureit