tagyoureit/nodejs-poolController #109: Tracked Files

arrmo, 2019-01-01:
A bit of a question, not necessarily an issue - but wondering about clean up here. I keep seeing the following files in my git status (i.e. showing up as modified) ... do they need to be marked as not tracked?
```
        modified:   package-lock.json
        modified:   package.json
        modified:   src/www/bootstrap/configClient.json
```

Thanks!

tagyoureit, 2019-01-03:
package.json is the file that holds all of the package information.  That one is needed in general but you can ignore it when you are committing.
package-lock.json tracks changes to the node_modules directory and this should not be changed.  We can ignore this one all together.
configClient.json needs to be included because the file isn't otherwise generated on it's own.  You can ignore it when publishing/committing but it needs to be included.

arrmo, 2019-01-04:
Yep, after playing and discussing - agreed! Do we tag package-lock.json as not tracked? Just trying to minimize the files to check / ignore each time.

Thanks!