tagyoureit/nodejs-poolController #113: Help Noob using Raspberry Pi 3 B+, Pentair EasyTouch 4, and ScreenLogic Wireless Kit

afgoody, 2019-01-29:
As the title says, I'm a noob with Raspberry Pi and json. I initially hit some snags that I posted here (https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=232293), but I think I was able to get back on track properly. However, when I try to start the NPM, I get the attached error.

[image]

When I check the USB traffic, I get a different result each time but only get the expected 'ffa5ff' occasionally. I'm certain the wires are connected to my RS485 USB adapter correctly. Do I need to put the adapter in a specific USB port to work?

[image]

I appreciate any help you can offer. My ultimate goal is to integrate my Pentair equipment with SmartThings using this DTH and SmartApp (https://github.com/bsileo/SmartThings_Pentair).

1. What version of the code are you using?

I thought I was using 5.3.0, but you can see the that I'm somehow only on 5.2.0. Do I need to update? How?

2. All output from the file log (See [packet capture](https://githu.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution))

I don't think I can since the NPM did not successfully start

3. Any errors that are in the console

See link posted above and attached screenshot

4. Your config.json or in-use configuration file

So far, I have not customized the config.json file

5. Your pool equipment

Pentair EasyTouch 4 and ScreenLogic Wireless Kit

tagyoureit, 2019-01-30:
If it is working, don't sweat the errors!  Good luck with the SmartThings integration.

GoodyGizmos, 2020-05-27:
@tagyoureit Sorry to resurrect this old thread, but I'm trying to get 6.0 going. I've attached the logs from the npm install, npm start (one for poolController and one for the website). As with 5.X, there are numerous errors, but I'm not sure if any of them are important. To test if the setup is working, I'm trying this URL (http://192.168.1.3/web/dashboard/index.html). Is that correct? I'm using nvm 0.34.0, node v10.15.3, and npm 6.4.1. 
[2020-05-27T04_18_29_243Z-debug.log](https://github.com/tagyoureit/nodejs-poolController/files/4686778/2020-05-27T04_18_29_243Z-debug.log)
[2020-05-27T04_23_46_748Z-debug.log](https://github.com/tagyoureit/nodejs-poolController/files/4686779/2020-05-27T04_23_46_748Z-debug.log)
[2020-05-27T06_09_46_477Z-debug.log](https://github.com/tagyoureit/nodejs-poolController/files/4686780/2020-05-27T06_09_46_477Z-debug.log)

tagyoureit, 2020-05-27:
So...
For the serialport errors with `npm i` run it as sudo `sudo npm i`.  Serialport needs admin privileges to install the bindings for the ports.

For the startup errors, you aren't running node v10:
`/home/pi/.nvm/versions/node/v14.3.0/lib/node_modules/npm/node_modules/npm-lifecycle/index.js`
See the v14.3.0?  I'd back down to Node v10.  Easiest way to do this is to use [n](https://github.com/tj/n).  It's as simple as `npm install -g n` and  `sudo n 10 latest`.

Ah, it looks like your 3rd log is using Node v10.  Somehow it looks like you are installing with Node v14 but then trying to run with Node v10.  That would cause issues for sure.  

Blow away the `./node_modules` directory and re-install with sudo privileges and making sure that it is v10.

GoodyGizmos, 2020-05-28:
I decided to just reflash the SD card w/ Raspbian. I use nvm to manage node.js and npm. If I shouldn't, please let me know. The current versions are nvm 0.34.0/node v10.20.1/npm v6.14.4.

I ran sudo npm install and received errors (pasted below), then I ran npm start and got more errors (two logs attached). I am not doing any additional configuration of any .json as I would have in v5.X since the Readme said it should autoconfigure. Am I missing a step?

[2020-05-28T04_12_53_538Z-debug.log](https://github.com/tagyoureit/nodejs-poolController/files/4693242/2020-05-28T04_12_53_538Z-debug.log)
[2020-05-28T04_12_53_628Z-debug.log](https://github.com/tagyoureit/nodejs-poolController/files/4693243/2020-05-28T04_12_53_628Z-debug.log)

pi@raspberrypi:~/pool $ sudo npm install
npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

> @serialport/bindings@2.0.8 install /home/pi/pool/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install EACCES: permission denied, access '/root/.npm'
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/pool/node_modules/@serialport/bindings/build'
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/pool/node_modules/@serialport/bindings
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN nodejs-poolController@6.0.0 No repository field.
npm WARN nodejs-poolController@6.0.0 No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@2.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-05-28T04_10_31_456Z-debug.log

tagyoureit, 2020-05-28:
I don't think it matters if you use nvm or n.  However, it looks like you still have some environmental errors.
If you run `node --version` and `npm --version` from the command line what do you see?

Here are some oddities from your logs:
1. `Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.`  => this message shouldn't be appearing for current versions of Node.  See https://github.com/nodejs/help/issues/1877.
1. `gyp ERR! node -v v10.15.2` doesn't match `3 info using node@v10.20.1` from your logs.
1. `prebuild-install WARN install EACCES: permission denied, access '/root/.npm'` => looks to be an access issue possibly with how nvm was installed?  Does nvm require sudo access?

GoodyGizmos, 2020-05-29:
v10.2.1 and 6.14.4

I did not use sudo to install nvm. I used this command that I found somewhere online and put in a help file I've maintained to document how I install your code.

wget -qO- https://raw.githubusercontent.com/nvm-sh/nvm/v0.34.0/install.sh | bash

tagyoureit, 2020-05-29:
Looks like there is a slightly newer version:
wget -qO- https://raw.githubusercontent.com/nvm-sh/nvm/v0.35.3/install.sh | bash

Did you make sure to run `nvm use node` _in a new shell_ after you installed per [instructions](https://github.com/nvm-sh/nvm#usage)?  Just grasping at straws here.

tagyoureit, 2020-07-04:
Closing this for now.  LMK if you need further help.

GoodyGizmos, 2020-07-05:
@tagyoureit, Honestly, I just gave up, but am willing to try again. Really, what I'd love to have is soup to nuts line by line copy/paste instructions starting from a freshly installed instance of Raspbian. I don't think the goal here is to make me a Linux expert. I've developed an ok troubleshooting level of knowledge for other integrations, but cannot seem to crest the hill on the learning curve for your integration. In any case, here are my current versions:
[image]

The code is unzipped into a folder called pool and here are the first lines of the error list from from an attempt at "sudo npm install" within that folder this morning:

npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

> @serialport/bindings@2.0.8 install /home/pi/pool/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install EACCES: permission denied, access '/root/.npm'
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/pool/node_modules/@serialport/bindings/build'
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/pool/node_modules/@serialport/bindings
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok

GoodyGizmos, 2020-07-05:
**I've read your instructions so many times. Perhaps a few things would be helpful to add in the meantime to reduce the learning curve for non Linux savvy noobs (i.e. me). I've put my thoughts for each line of your instructions with the hope that it might at least get a chuckle, but also consider adding more words.**

1. Install Nodejs. (https://nodejs.org/en/download/) **What version?**
2. Update NPM (https://docs.npmjs.com/getting-started/installing-node). **What version**
3. Download the latest code release OR clone with git clone git@github.com:tagyoureit/nodejs-poolController.git **Uhhh, ok, I know how to download and unzip, so I'll just do that (requires an understanding of how to transfer files from a Windows environment to an RPi...I use WinSCP)**
4. Unzip into nodejs-poolController. **I just make a folder called "pool" because it's a lot shorter and easier to remember/type**
5. Run 'npm install' in the new folder (where package.json exists). This will automatically install all the dependencies (serial-port, express, sockets.io, etc). **Uhhh, ok, I use Putty for command line instructions. I "cd pool," then type "npm install" and get a bunch of permission issues, so I try "sudo npm install" then get different errors)**
6. Run the app with 'npm start'* (again, in the root directory). It should now run properly. **I haven't gotten this far.**
7. to run with a specific configuration, run node index.js arg where arg is the name of your current config file. eg npm start configCustomized.json. By default, the app will load config.json. **I haven't gotten this far, but I thought this wasn't required in v6 since it autoconfigures, no?**

tagyoureit, 2020-07-05:
I don't have a working MicroSD card right now (I just ordered one).  I can give you a full walkthrough after that arrives.

Here is what I would do (and all of these on your Linux box / through PuTTy):
1 & 2.  Install `n` with `sudo npm install -g n`.  Then run `sudo n 10` to install the latest version of Node 10.   `node --version` should bring up version 10.  `npm --version` should bring up the latest NPM (6.14.x).
3.  You should use git.  Run `sudo apt-get install git` (or `sudo yum install git` if you are on RPM based linux).  `git clone git@github.com:tagyoureit/nodejs-poolController.git` will clone the repository.  To keep things simple here it will install into `nodejs-poolController`.  If this works you can figure out how to install into a different directory later.
4.  `cd nodejs-poolController` then `git checkout next`.  This will switch to the development branch.
5.  `npm i`.  You should not get any errors here if the version of node/npm is correct.
6.  Yes, `npm start`.
7.   There is no notion of running with a different configuration file.

GoodyGizmos, 2020-07-06:
Thanks. I imaged my SD card with the latest Raspbian using Etcher. Enabled WiFi and started with your instructions verbatim until Step 3:

pi@raspberrypi:~ $ git clone git@github.com:tagyoureit/nodejs-poolController.git
Cloning into 'nodejs-poolController'...
The authenticity of host 'github.com (192.30.255.113)' can't be established.
RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'github.com,192.30.255.113' (RSA) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

I tried to add a "sudo" to the front, but got the same errors.

GoodyGizmos, 2020-07-06:
I'm going on ~20+ minutes of the webClient's "npm start" and wondering your recommended method for running both at the same time.

tagyoureit, 2020-07-07:
Yes, sorry... `git clone https://github.com/tagyoureit/nodejs-poolController.git` is the right url to clone.  

You'll need two command prompts to run both apps.  You can either use `screen` or `tmux` for two virtual sessions or if you want the apps to start up automatically if there is a power failure you can use [pm2](https://github.com/tagyoureit/nodejs-poolController/wiki/PM2).  

Glad to see you are making progress.  If you hit the hamburger menu in the top right of the app you can switch from SSDP to manually inputting the ip address.  (You can enable SSDP on the server in the config.json file if it isn't enabled.). Also, currently the browser makes a direct connection to the nodejs-poolController app.  I have it on my to-do list to make it go through the -webClient server as a proxy.

GoodyGizmos, 2020-07-07:
It looks like turning on SSDP got everything going, so thank you. I was even able to get bsileo's SmartThings integration working. I must need to still fix something in the SmartThings section of config.json based on the errors I'm getting, but I can troubleshoot that and get pm2 working again (that's what I used with 5.3) later this week. I'll let you know if I encounter any issues with that. Thanks again for your help and let me know if you want me to type up some "instructions for dummies."

[image]

tagyoureit, 2020-07-07:
You'll need @bsileo's new version of the [SmartThings/Hubitat](https://github.com/bsileo/hubitat_poolcontroller) code.  None of the v5.3 integrations will work with 6.0 as everything is brand new.  But he already ported this over.

GoodyGizmos, 2020-07-08:
@tagyoureit I'm all set up with pm2 running (there's a typo in the spelling of the webclient path in the sample script).

I've tested all of the functionality of my EasyTouch 4. All seem to work well except one of my Features (Spillway) is missing and another one (Slide) is listed under Circuits instead of Features. I also have an extra 'Not Used' Circuit and a "Solar' virtual circuit. On the EasyTouch 4, I think the Solar virtual circuit is configured as the Spillway, so that may explain part of what's going on. Any ideas?

[image]

tagyoureit, 2020-07-08:
Alright, that should be easy to fix.  I just need to understand where the circuits/features begin and end.  Can you do 3 things for me:
1. a [packet capture](https://github.com/tagyoureit/nodejs-poolController/wiki/How-to-capture-all-packets-for-issue-resolution) "with reset" so I can see all the packets from the beginning.
2. Can you look in the `poolConfig.json` and tell me which circuit/feature ID's are wrong?  
3. [optional, but very helpful] If you can enable the LAST circuit & feature in your pool system on the panel and do a packet capture I can definitely confirm the end values.  If not, I may still be guessing at the end value for the circuits/features.

@rstrouse is separately working on making the virtual circuits sync with the actual heat states.  We can double check the solar virtual circuit once we nail down the equipment id ranges.

The ranges for the ET2-4 that I currently have are:
Circuits: 1-6 (1=spa, 6=pool, 2-5 = 4 physical circuits)
Features: 11-14 (4 feature circuits)

The virtual circuits get ID's in the 192 range and don't actually exist on *Touch.

GoodyGizmos, 2020-07-09:
Here are the 3 things you requested (I didn't know how long to run Steps 1 and 3, so I went about 5 minutes on each):

[Step1.zip](https://github.com/tagyoureit/nodejs-poolController/files/4894452/Step1.zip)
[Step2.zip](https://github.com/tagyoureit/nodejs-poolController/files/4894458/Step2.zip)
[Step3.zip](https://github.com/tagyoureit/nodejs-poolController/files/4894459/Step3.zip)

Here are some more files and videos that could be helpful:

[image]

https://youtu.be/8MlgwxQdL14
https://youtu.be/P9AWvC5u33w

Spillways is assigned to the AuxX circuit. I also notice in the config.json that the autoconfig set IntelliChem to true, but I DO NOT have an IntelliChem.

tagyoureit, 2020-07-09:
Ok, I pushed up some changes to make this work.  I didn't really see any new features/circuits added in the Step3 and I also couldn't watch your videos (marked private) but was able to replicate on my ET8 system.  Pull the latest and try now.

I also didn't see the intellichem get added to the system.  Maybe that was from old code.  You can blow away your data/*.json files and start fresh.  Everything should be picked up correctly.

GoodyGizmos, 2020-07-10:
Here's what I just did:

pm2 stop all
deleted all .json from the data subfolder
(in the nodejs-poolController folder) npm update
(in the home folder) pm2 start ecosystem.config.js

I waited a few minutes and refreshed the WebClient on my browser. The Chem Controllers sections now says "No Chem Controllers" (good thing), but I still do not see the AuxX Spillway circuit.

Did I do the update correctly?

The videos are now Unlisted (not Private), so you should be able to view them using the links above.

tagyoureit, 2020-07-10:
Do you have screenlogic by any chance?

GoodyGizmos, 2020-07-10:
Yes

tagyoureit, 2020-07-10:
Oh, good... can you do a packet capture when screenlogic is asking/receiving it's configuration?  You can usually set the temperature once or twice and that will trigger it.  If you monitor the packets you should see ~80 or so packets going back/forth across the wire. 

I didn't see the Aux Extra in your prev captures... Hmmm, actually I wonder if I need to re-ask for it.  But regardless this will definitely help in case there is anything else I missed.

GoodyGizmos, 2020-07-10:
Just to confirm, did I did the update correctly (the commands I used above)?

Here is a packet capture:
[20200709.zip](https://github.com/tagyoureit/nodejs-poolController/files/4901009/20200709.zip)

While this was running, I used ScreenLogic to turn the pool on, turn the gas heater on, change the temp a few degrees 4-5 times, turned off the heater, turned on the Spillway (AuxX circuit), turned off the Spillway, and turned off the pool. Then I stopped the packet capture. I don't know where to look for the packet traffic using ScreenLogic.

tagyoureit, 2020-07-10:
Oh... good point. 

pm2 stop all
deleted all .json from the data subfolder

(in the nodejs-poolController folder) 
     **git pull** <-- This is the one you need to get the latest code.
     npm update. <-- This is a good step.  Not necessary unless we update dependencies but can't hurt.
(in the home folder) pm2 start ecosystem.config.js

tagyoureit, 2020-07-10:
I have a feeling I still may miss the aux extra the first go round, but if you trigger SL to push the config I think it should pick up your spillway. 

The packet capture also didn't get ScreenLogic doing it's thing.  A fail-safe to do that is unplug it and plug it back in.  But if you go into any of the settings and change stuff there it definitely requests the entire config.

GoodyGizmos, 2020-07-10:
It looks like you got it! Thanks. Does the Freeze VC just indicate if the panel is sensing a freeze condition (i.e. <35F)?

[image]

GoodyGizmos, 2020-07-10:
Also, what are the numbers in parenthesis next to each of the Schedules?

tagyoureit, 2020-07-10:
Great!

> Does the Freeze VC just indicate if the panel is sensing a freeze condition (i.e. <35F)?

Yup, exactly.

> Also, what are the numbers in parenthesis next to each of the Schedules?

Internal ID of the schedules.  Probably can be hidden from the view.

marc-leonard, 2020-07-10:
> Thanks. I imaged my SD card with the latest Raspbian using Etcher. I enabled WiFi and SSH and started with your instructions verbatim...
> 
> Steps 1 & 2:
> [image]
> 
> Step 3:
> 
> pi@raspberrypi:~ $ git clone [git@github.com](mailto:git@github.com):tagyoureit/nodejs-poolController.git
> Cloning into 'nodejs-poolController'...
> The authenticity of host 'github.com (192.30.255.113)' can't be established.
> RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
> Are you sure you want to continue connecting (yes/no)? yes
> Warning: Permanently added 'github.com,192.30.255.113' (RSA) to the list of known hosts.
> [git@github.com](mailto:git@github.com): Permission denied (publickey).
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> 
> I tried to add a "sudo" to the front, but got the same errors, so I took to Google and found that I should try substituting "https://github.com/" instead of "[git@github.com](mailto:git@github.com):". I did that and it worked.
> 
> Step 4:
> Seemed to go well: "Switched to a new branch 'next'"
> 
> Step 5:
> Seemed to go well. At the end of all the random text was this...
> 
> npm WARN nodejs-poolController@6.0.0 No repository field.
> npm WARN nodejs-poolController@6.0.0 No license field.
> 
> added 579 packages from 412 contributors and audited 580 packages in 123.198s
> 
> 40 packages are looking for funding
> run `npm fund` for details
> 
> found 9 low severity vulnerabilities
> run `npm audit fix` to fix them, or `npm audit` for details
> 
> I assumed this was all OK, so I continued.
> 
> Step 6:
> Looks like winning, except it just sat there after 5 minutes. I need to try and get the webserver going too, so I hit Ctrl-C to get back to a command prompt.
> [image]

If GoodyGizmos is a noob I'm not sure how to classify myself, but, I am also newer user trying to set this up on a Raspberry Pi to control my Pentair equipment, linking to the ISY through the polisy. I've attempted the instructions on the main wiki page, but the information in this conversation seems to have guided me closer to functionality.

I started with a fresh raspbian installation, installed node and npm, and confirmed the versions:
[image]

I cloned the git repository using git clone git clone https://github.com/tagyoureit/nodejs-poolController.git, moved into the nodejs-poolController directory, and entered git checkout next to move to the newer branch

npm i
There were a LOT of messages, but at the end I received the same WARN messages as GoodyGizmos

npm start
The response to this varies slightly each time, but after a few minutes it generally looks like:
[image]

The quantity and details of the error messages change each time I try to start it.  However, when I was following the main instructions on the wiki page I got VERY different messages.  This is seems to be MUCH closer to functioning.  Unfortunately, it doesn't seem to be picking up anything, here's what I see at port 3000
[image]

I suspected that I was supposed to also follow some of the original installation instructions, so I've also attempted this with Mongodb and polyglot v2 (with an override to port 5000) installed on the Raspberry Pi, but land at the same result. 

I would really appreciate some advice about how to get this running.  I've been using an Autelis for more than a year, but it is unreliable, occasionally reporting the incorrect status to the ISY.

In my most recent attempt I used the following almost unchanged config.json (with logging enabled):

{
  "controller": {
    "comms": {
[... 127 lines cut ...]
        "excludeDest": []
      },
      "unknown": {
        "enabled": true,
        "includeSource": [],
        "includeDest": [],
        "excludeSource": [],
        "excludeDest": []
      }
    },
    "app": {
      "enabled": true,
      "level": "info",
      "captureForReplay": false
    }
  },
  "appVersion": "6.0.0"
}

And this is the most recent packetlog:
[packetLog(2020-07-10_14-17-04).log](https://github.com/tagyoureit/nodejs-poolController/files/4905145/packetLog.2020-07-10_14-17-04.log)

I'd really appreciate any advice on what I'm missing here.  I can see that many other people have this working, so I'm clearly missing something to which I am completely oblivious.  And thank you to tagyoureit for your work on this, and to you and GoodyGizmos for explaining it so thoroughly.

Marc

tagyoureit, 2020-07-10:
Hi @marc-leonard,
everything in the packet log looks good.  I think you might just be going to the wrong client.  The screenshot you posted (with port 3000) is for the 5.3 version.  That shouldn't even be available/loading if you have the 6.0 version (`next` branch) running.  

You should separately install one of:
1. [nodejs-poolController-dashPanel](https://github.com/rstrouse/nodejs-poolController-dashPanel).  This is built primarily around the IntelliCenter but will work with *Touch.
1. [nodejs-poolController-webClient](http://github.com/tagyoureit/nodejs-poolController-webClient).  Built primarily around EasyTouch/IntelliTouch but will work with other systems.

and these are separately packaged front ends that connect to 6.0.  6.0 has fundamentally different data structures than 5.3.  But even before you try to connect the new UI's look it the `./data` directory.  There will be 2 json files in there that contain the configuration and state information for the pool.  If those are populating correctly, then move on to getting the web clients up and running.  

One hint - if you load up the `-webClient` set `"ssdp": {"enabled": false}}` to true and it will automatically connect to the pool controller.

tagyoureit, 2020-07-10:
Oh, I just noticed something else that I thought you were quoting Goody:
> Step 6:
Looks like winning, except it just sat there after 5 minutes. I need to try and get the webserver going too, so I hit Ctrl-C to get back to a command prompt.

You need to have the web server up and running along with the web client.  If you stop the main app then of course nothing can connect to it.  You can either open up two SSH sessions or use screen or TMUX to open multiple sessions.  Or you can setup [pm2](https://github.com/tagyoureit/nodejs-poolController/wiki/PM2) to launch both applications upon reboot.

tagyoureit, 2020-07-10:
And @GoodyGizmos, one minor correction on my end:

> (in the nodejs-poolController folder)
git pull <-- This is the one you need to get the latest code.
npm update. <-- This is a good step. Not necessary unless we update dependencies but can't hurt.
(in the home folder) pm2 start ecosystem.config.js

`npm update` will only update existing modules.  If you need to install new packages (eg if I add them to either project) then you should run `npm i` (or `npm install`).  I actually would not recommend upgrading your packages beyond the versions that we ship which is what `npm update` will do.

marc-leonard, 2020-07-11:
> Hi @marc-leonard,
> everything in the packet log looks good. I think you might just be going to the wrong client. The screenshot you posted (with port 3000) is for the 5.3 version. That shouldn't even be available/loading if you have the 6.0 version (`next` branch) running.
> 
> You should separately install one of:
> 
> 1. [nodejs-poolController-dashPanel](https://github.com/rstrouse/nodejs-poolController-dashPanel).  This is built primarily around the IntelliCenter but will work with *Touch.
> 2. [nodejs-poolController-webClient](https://github.com/tagyoureit/nodejs-poolController-webClient).  Built primarily around EasyTouch/IntelliTouch but will work with other systems.
> 
> and these are separately packaged front ends that connect to 6.0. 6.0 has fundamentally different data structures than 5.3. But even before you try to connect the new UI's look it the `./data` directory. There will be 2 json files in there that contain the configuration and state information for the pool. If those are populating correctly, then move on to getting the web clients up and running.
> 
> One hint - if you load up the `-webClient` set `"ssdp": {"enabled": false}}` to true and it will automatically connect to the pool controller.
> 
> Here's a picture of your pool in rstrouse's web client:
> [image]

Thank you, as I suspected you identified something I was completely ignoring!  The two jsons do appear to be populating correctly in ./data.   

As an FTI: I installed the webclient, but get a few (I believe) minor errors on installation
[image]

after npm i, setting ssdp to true in config.json, and npm start, it tries to launch, but eventually stalls after the parcel build
[image]

But, my real preference is to connect the polisy nodeserver so I can skip the web interface.  What is the appropriate api key for 6.0?  I.e., instead of api_url: http://192.168.1.23:3000, what port should I be using if I've left the config.json unchanged (except for enabling logging)?  

I really appreciate both your help and responsiveness!!
Marc

tagyoureit, 2020-07-11:
Parcel takes a loooooooong time to build on a Pi.  When you see "Building index.js..." you were maybe 90% of the way there.

Unfortunately, no on has updated the connector for ISY to 6.0 yet.  And this is the first I'm hearing of Polisy.  If the Polisy integration is anything like the ISY integration [link to 5.3 integration file](https://github.com/tagyoureit/nodejs-poolController/blob/master/src/integrations/socketISY.js) and there were some funky [mappings](https://github.com/tagyoureit/nodejs-poolController#only-output-changes-in-configjson) in the older `config.json`.  Another user created a [Polygot ISY](https://github.com/brianmtreese/nodejs-pool-controller-polyglotv2) interface but that as well would need to be updated.  (And I'm not sure if that would help you in your Polisy setup, anyway.). 

Someone with some light JS skills (and an ISY) would need to work on this to get it up and running.  I could help walk you through some of it but I have a lot of stuff on my backlog right now I'm trying to clear out.

marc-leonard, 2020-07-11:
Thanks, that polyglot isy integration is what I was trying to use. The polisy is the universal devices box that runs the polyglot nodeservers and plays nicely with the isy. 

Sounds like I actually did have the individual pieces working, but the pieces I selected don’t work together. 

Perhaps downgrading to 5.3 might be the solution that will work with the existing isy integration. 

Thanks again for your work, this is great!!
Marc

kurtifier, 2020-07-27:
Thank you @tagyoureit for creating this huge project! I am currently just starting and I am running into some issues. I am getting errors (failed to send/receive) bytes when I start npm.
node version:  10.22.0
npm version: 6.14.6
[logs.zip](https://github.com/tagyoureit/nodejs-poolController/files/4979391/logs.zip)

[image]

When I try to access the web client, all I see is can't get. I am not sure what I did wrong and I apologize I am fairly new to this. 
[image]

I am also trying to integrate with home kit and after I installed the plugin with homebridge, it auto added what you see below. None work or are being updated. My end goal is to automate with home kit and I would really appreciate any help! Thank you so much!
[image]

tagyoureit, 2020-07-27:
> I am getting errors (failed to send/receive) bytes when I start npm.

Failed packets are to be expected as part of the RS485 bus.  You can read about it in the [wiki](https://github.com/tagyoureit/nodejs-poolController/wiki/RS-485-Adapter-Details#failed-packets).

> When I try to access the web client, all I see is can't get. I am not sure what I did wrong and I apologize I am fairly new to this.

The main server runs on port 4200.  If you navigate to `http://192.168.1.149/state/all` you should see all of your pool equipment in JSON format.  In order to get a UI that you can use, you need to download one of the two current [client](https://github.com/tagyoureit/nodejs-poolController#web-clients).  Depending on which one you install you will either access it on port 5150 (dashPanel) or 8080 (webClient).  

If you are successful with the above two items, then you can proceed to the next step for the Homebridge installation.  gadget-monk has created [code](https://github.com/tagyoureit/nodejs-poolController#home-automation-bindings-previously-integrations) that you can use with the bindings (bindings send data to other places, like Homebridge).

kurtifier, 2020-07-28:
Thank you! I totally forgot to download the web client and everything is now working including home kit! Thank you for your fast reply @tagyoureit and all your help!

closed 2020-08-11 by tagyoureit